dresden-elektronik/deconz-rest-plugin #6190: Device name

thijsmans, 2022-07-11:
## Device

- Product name: S-Mitter MultiWhite Basic Smart Remote Control
- Manufacturer: Halemeier GmbH
- Model identifier: HA-ZBM-MW2
- Device type : Please remove all unrelated device types. 
  - Remote

## Screenshots
[image]

### Basic
[image]

### Power Configuration
![Power Configuration cluster ( 0001 )https://user-images.githubusercontent.com/25135711/178335908-8b3cab38-b213-4f72-a5e3-486f7a58c679.png)

### Identify
![Identify cluster ( 0003 )https://user-images.githubusercontent.com/25135711/178335995-f3f6e1c8-33de-42c8-9837-6cb19c2c41f8.png)

### Diagnostics
![Diagnostics cluster ( 0B05 )https://user-images.githubusercontent.com/25135711/178336256-e9a54337-ce2e-4505-87b2-b98972e7026e.png)

### Other clusters that are not mentioned above
[image]

Smanar, 2022-07-12:
Hello to start, can you try with adding DDF ?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Halemeier GmbH",
  "modelid": "HA-ZBM-MW2",
  "product": "MultiWhite Basic Smart Remote Control",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },        
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },    
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

and add to the "button_maps.json" file (you have it in the "devices" folder too like for DDF) thoses lines

```
        "GmbHremote": {
            "vendor": "GmbH",
            "doc": "MultiWhite Basic Smart Remote Control",
            "modelids": ["HA-ZBM-MW2"],
            "map": [
                [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"]
            ]
        },
```

- I have make bind only on cluster 0x0006, can need 0x0008 0x0300 or 0x0005
- have used ransom value for button map, not sure thoses one will work

To find how to fill the button_maps.json I need log when using the device.
- Add the DDF and edit the button_maps.json
- Re-include the device
- If I m right now you will be able to see it in phoscon and the node name in deconz will be other thing than a hexa number like 0xXXXX
- go on deconz/help/debug view, enable the flag "info" and "info_l2"
- press a button.
- you will see error about button map

I just need the used button and the according error line.
If not all buttons are working, you probably need to add missing binding, remember only the on/off is able to work at 100%

thijsmans, 2022-07-26:
Sorry for the delay; I run deCONZ as Homeassistant add-on, so I first had to get shell access to the filesystem of the docker... Turns out this is a PITA. And unfortunately, this also means that all changes in the DDF/json-files are gone after each restart of the add-on.

What I did:

- delete the node from the network;
- create a dir called "/usr/share/deCONZ/devices/halemeier";
- in this dir, create a file called "multiwhiterc.schema.json";
- in this file, paste the json you supplied (the long bit);
- in the file "/usr/share/deCONZ/devices/butt_maps.json" and added this bit:

```
         "HalemeierGmbHremote": {
            "vendor": "Halemeier GmbH",
            "doc": "MultiWhite Basic Smart Remote Control",
            "modelids": ["HA-ZBM-MW2"],
            "map": [
                [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"]
            ]
        }
```
This is a slightly altered version of your json (naming of the object (GmbHremote to HalemeierGmbHremote, and vendor GmbH --> Halemeier GmbH). To be exact, in the first try I did not change a thing, but it did not work so I renamed the vendor to the exact spelling used by the remote itself. Turns out, this doesn't matter ;-)

- I enabled joining mode through Phoscon (devices > switches > add new switch). 

**Result:**

- Phoscon throws an error about not finding new devices. It doesn't display it in the list of switches either.
- VNC shows the remote by it's hex label (0xCD6F)

After the first attempt, I thought that maybe deCONZ needed to restart to scan for new json-files. But as I said: every time the deCONZ docker container restarts, the filesystem resets. There is a Homeassistant [feature request pending](https://community.home-assistant.io/t/deconz-support-management-of-ddf-files-through-ha-config/370331), but nothing is happening anytime soon. So I hope the restart is not needed.

All the more reason to get the official support up and running :) Any idea why the remote wasn't recognized?

Smanar, 2022-07-27:
Yeah the HA OS is not easy to use, you have some trick here https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Docker are not persistant, so you need to mount a folder (and HA don't work like classic dockers)

Deconz use thoses line to detect devices
```
  "manufacturername": "Halemeier GmbH",
  "modelid": "HA-ZBM-MW2",
```

And only `"modelids": ["HA-ZBM-MW2"],` for the button_map file.

The status is "Gold" so I think the problem was from the used path for the DDF.
You can to load the file in the editor and make "hot relaod" on the menu, it will not survice at reboot, but can be used to test.

>create a dir called "/usr/share/deCONZ/devices/halemeier";

Not normal, all folder need to already exist.

thijsmans, 2022-07-27:
In VNC, I opened the file "devices/halemeier/multiwhiterc.json" and used Hot Reload. The remote control showed up as "Switch 8" in Phoscon (and VNC). So this helped!

These are the log lines for the buttons. There are four numerical buttons (for scenes / presets); two buttons for dimming (up and down), two buttons for the color temperature, and one on/off-button. I made screenshots, because I could not copy the text out of VNC:

**#1**
[image]

 **#2**
[image]

**#3**
[image]

**#4**
[image]

**Dimming up**
[image]

**Dimming down**
[image]

**Color-button yellow**
[image]

**Color-button white**
[image]

**On/off**
[image]

Smanar, 2022-07-27:
>the switch is not available for inclusion in a group

Yep, need to add this feature in the DDF too, can take exemple for this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/dresden_elektronik/kobold.json
Need to add 
```

        {
          "name": "config/group",
          "default": "auto"
        },
```

and, if needed, for all used clusters (here for the 0x0006)

```
    {
      "bind": "groupcast",
      "src.ep": 1,
      "cl": "0x0006",
      "config.group": 0
    },
```

But will see later.

Something funny, I m making bind only for the on/off cluster but all other are working ...
I will ask if it s not because of the legacy code.

According to your log, the json edition for button map can be

```
        {
         "HalemeierGmbHremote": {
            "vendor": "Halemeier GmbH",
            "doc": "MultiWhite Basic Smart Remote Control",
            "modelids": ["HA-ZBM-MW2"],
            "map": [
                [1, "0x01", "SCENES", "RECALL_SCENE", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 1"],
                [1, "0x01", "SCENES", "RECALL_SCENE", "2", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 2"],
                [1, "0x01", "SCENES", "RECALL_SCENE", "3", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 3"],
                [1, "0x01", "SCENES", "RECALL_SCENE", "4", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 4"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "0", "S_BUTTON_5", "S_BUTTON_ACTION_HOLD", "Dimm up"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "1", "S_BUTTON_6", "S_BUTTON_ACTION_HOLD", "Dimm down"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "1", "S_BUTTON_7", "S_BUTTON_ACTION_SHORT_RELEASED", "Color yellow  press"],
                [1, "0x01", "COLOR_CONTROL", "STEP_COLOR_TEMPERATURE", "3", "S_BUTTON_8", "S_BUTTON_ACTION_SHORT_RELEASED", "Color white press"],
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_9", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_10", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"]
            ]
        }
```

But as you need to restart deconz to test this file, idk if you can test it, you don't need to restart the docker, only the app.

thijsmans, 2022-07-28:
Okay, it definately did NOT work with the Homeassistant add-on :) All changes were lost upon restarting the add-on. I disabled the add-on and created a seperate docker outside of HA, using the deconz-community/deconz-docker image and with a persistent mount to /opt/deCONZ. The directory "/opt/deCONZ" gets populated with:

- config.ini
- devices (directory; empty)
- otau (directory; empty)
- vnc (directory with log, .pid, .pem and passwd-file)
- zcldb.txt
- zll.db

The devices directory is empty. All devices seem to be loaded to /usr/share/deCONZ/devices. Is this normal behavior? Will files in /opt/deCONZ be used? Or should I change the mount to /usr/share/deCONZ/devices instead of /opt/deCONZ?

Anyway, as to the remote:

- I added this item to the device json-file:
```
        {
          "name": "config/group",
          "default": "auto"
        },
```
and added this to button_maps.json:
```
    {
      "bind": "groupcast",
      "src.ep": 1,
      "cl": "0x0006",
      "config.group": 0
    },
```

Then I restarted the docker container, with these results:

- the on/off button works. Took me a while to realise that the remote uses one button as two, alternating between on and off (button 9 -> on, and 10 -> off);
- the four scene-buttons work;
- changing color temperature gets recognized (key assignment lights up on press), but has no actual effect;
- dimming up and down does not work; I can't even get labels for the key assignment:

[image]

Please note that the key binding for dimming up is the white space between the beige header (Key assignment [+]) and "On / off".

Also, is it possible that deCONZ is not letting the remote go to sleep? The battery seems to be draining.

Mimiix, 2022-07-28:
> Okay, it definately did NOT work with the Homeassistant add-on :) All changes were lost upon restarting the add-on. I disabled the add-on and created a seperate docker outside of HA, using the [...]

Please note for them working in Phoscon, you need Phoscon to add support for it.

This repository only includes support with deCONZ and events in the websocket :)

Smanar, 2022-07-28:
Lol, yep, deconz is not something easy to use ^^, so much projects.

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

So `/usr/share/deCONZ/devices/` will be reseted on next deconz installation, better to mount the second one, it will be never updated.

>the on/off button works. Took me a while to realise that the remote uses one button as two, alternating between on and off (button 9 -> on, and 10 -> off);

Yep, IDK how to do, you have only 1 physical button, but 2 requests, I can mix them, but you will be not able to differentiate them, you will just receive the same state/buttonevent for the 2 requests (but the only 1 button)

>dimming up and down does not work; I can't even get labels for the key assignment

Like said mimix, this is managed by phoscon, you haven't different websocket return for them ?

>Also, is it possible that deCONZ is not letting the remote go to sleep? The battery seems to be draining.

This is a big issue.
The device haven't poll system, so the problem is probably from binding/reporting.

first thing to do, set sleeper to true

` "sleeper": true,`

Then you have only 1 report configured

```
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
```

You can increase thoses timer, but they are standard, you haven't something in logs spamming ? in deconz / help / debug view, with "info" and "info_l2" ?
Try remove all binds in the DDF (for group too)

closed 2022-08-26 by github-actions[bot]