dresden-elektronik/deconz-rest-plugin #6192: Hue ZHA lights

ebaauw, 2022-07-12:
## Device
- Product name: Hue white and color ambiance
- Manufacturer: Signify Netherlands B.V.
- Model identifier: LCG002 (but also many others)
- Device type: Extended color light 

As discussed on Discord, this likely concerns all recent Hue lights, that use the ZHA profile on endpoint 0x0B instead of ZLL.

These lights support attribute reporting for _On/Off_, _Current Level_, _Current X_, _Current Y_, _Color Temperature_ (but not: _Color Mode_, _Enhanced Color Mode_, and _Color Loop Active_).  This should be leveraged to reduce polling.

With the latest firmware, these lights support dynamic scenes, _Candle_ and _Fireplace_, through the 0xFC03 cluster, similar to the Hue gradient light strip, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5891.  0xFC03/0x0002 supports attribute reporting as well.  This should be supported by additional values for `state.effect`, similar to the special scenes in the Müller lights or LIDL Xmas light strip.

I've hacked together a definition of 0xFC03 for `general.xml`, allowing me to set the dynamic scene, and query 0xFC03/0x0002.  Unfortunately, the GUI doesn't show data type `ostring` in a useful way, but at least you can trigger a _Read Attributes_ and see the response in the sniffer.

```xml
    <cluster id="0xfc03" name="Hue" mfcode="0x100b">
      <description>Hue-specific cluster.</description>
      <server>
        <command id="0x00" dir="recv" name="Set Dynamic Scene" vendor="0x100b" required="m">
          <payload>
            <attribute id="0x0000" type="u16" name="Command" showas="hex" default="0x0020" required="m"/>
            <attribute id="0x0001" type="bool" name="Enable" required="m"/>
            <attribute id="0x0002" type="enum8" name="Scene" required="m">
              <value name="None" value="0x00"/>
              <value name="Candle" value="0x01"/>
              <value name="Fireplace" value="0x02"/>
            </attribute>
          </payload>
        </command>
        <attribute id="0x0002" name="Status" type="ostring" mfcode="0x100b" access="r" required="m"/>
      </server>
    </cluster>
```

## Screenshots
[image]
[image]

### Basic
[image]

### FC03
[image]

ebaauw, 2022-07-13:
Here's a first go at a DDF and accompanying switch.  Should create a unicast binding for 0xFC03 and attribute reporting for 0x0002, with 300s interval.  Attribute is linked to `on`, with a backup refresh interval of 330.  Read on `bri`, `ct`, `x` and `y` should be disabled - I see no more polling in the sniffer. 
Issues:
- No support for `hue` nor `sat`; when setting (Enhanced) Hue and/or Saturation, `colormode` reports `xy`;
- No support for setting `effect`, but it does report `candle` or `fireplace` correctly;  It won't report `colorloop` correctly, though.

Here's the DDF, `white_and_color_ambiance_ble.json`:
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["$MF_SIGNIFY", "$MF_SIGNIFY"],
  "modelid": ["LCA006", "LCG002"],
  "product": "Hue white and color ambiance (BLE)",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_EXTENDED_COLOR_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x0b"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/colorcapabilities"
        },
        {
          "name": "config/ctmax"
        },
        {
          "name": "config/ctmin"
        },
        {
         "name": "state/alert"
       },
       {
         "name": "state/bri",
         "read": {"fn": "none"}
       },
       {
         "name": "state/colormode"
       },
       {
         "name": "state/ct",
         "read": {"fn": "none"}
       },
       {
         "name": "state/effect",
         "default": "none",
         "values": [
           ["\"none\"", "no effect is active"],
           ["\"colorloop\"", "colorloop through hue values"],
           ["\"candle\"", "candle dynamic effect"],
           ["\"fireplace\"", "fireplace dynamic effect"]
         ]
       },
       {
         "name": "state/on",
         "parse": {
             "ep": "0x0b", "cl": "0xfc03", "mf": "0x100b", "at": "0x0002",
             "script": "fc03_state.js"
         },
         "read": {
           "ep": "0x0b", "cl": "0xfc03", "mf": "0x100b", "at": "0x0002"
         },
         "refresh.interval": 330
       },
       {
         "name": "state/reachable"
       },
       {
         "name": "state/x",
         "read": {"fn": "none"}
       },
       {
         "name": "state/y",
         "read": {"fn": "none"}
       }
     ],
     "bindings": [
       {
         "bind": "unicast",
         "src.ep": "0x0b",
         "cl": "0xfc03",
         "report": [
           { "mf": "0x100b", "at": "0x0002", "dt": "0x41", "min": 1, "max": 300 }
         ]
       }
     ]
   }
 ]
}
```
And the `fc03_state.js` script to parse 0x0002:
```js
/* global R, ZclFrame */

const attrid = ZclFrame.at(1) << 8 | ZclFrame.at(0)
const dt = ZclFrame.at(2)
if (attrid === 0x0002 && dt === 0x41) {
  const len = ZclFrame.at(3)
  if (len >= 4) {
    const mode = ZclFrame.at(5) << 8 | ZclFrame.at(4)
    R.item('state/on').val = ZclFrame.at(6) !== 0
    R.item('state/bri').val = ZclFrame.at(7)
    if (mode === 0x000b && len === 8) {
      R.item('state/x').val = ZclFrame.at(9) << 8 | ZclFrame.at(8)
      R.item('state/y').val = ZclFrame.at(11) << 8 | ZclFrame.at(10)
      R.item('state/colormode').val = 'xy'
      R.item('state/effect').val = 'none'
    } else if (mode === 0x000f && len === 10) {
      R.item('state/ct').val = ZclFrame.at(9) << 8 | ZclFrame.at(8)
      R.item('state/x').val = ZclFrame.at(11) << 8 | ZclFrame.at(10)
      R.item('state/y').val = ZclFrame.at(13) << 8 | ZclFrame.at(12)
      R.item('state/colormode').val = 'ct'
      R.item('state/effect').val = 'none'
    } else if (mode === 0x00ab && len === 10) {
      R.item('state/x').val = ZclFrame.at(9) << 8 | ZclFrame.at(8)
      R.item('state/y').val = ZclFrame.at(11) << 8 | ZclFrame.at(10)
      R.item('state/colormode').val = 'xy'
      const effect = ZclFrame.at(13) << 8 | ZclFrame.at(12)
      switch (effect) {
        case 0x8001:
          R.item('state/effect').val = 'candle'
          break
        case 0x8002:
          R.item('state/effect').val = 'fireplace'
          break
        default:
          R.item('state/effect').val = 'none'
          break
      }
    }
  }
}
```

ebaauw, 2022-08-04:
Ping

theCheek, 2022-08-20:
Just got the Hue Signe Floor lamp and looking to get the three independent color zones working. Is this still an ongoing issue?

ebaauw, 2022-08-20:
I don't have a Signe, but I assume it's similar to the gradiant lightstrip, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5891.

ebaauw, 2022-09-11:
See PR for latest code.

## Currrent State
- For now, use regular attributes for light state using `Item/val`, because`R.item('state/...').val` doesn't generate websocket notifications.  By using 0xFC03/0x0002, `on`, `bri`, `ct`, and `xy` could be polled using a single `read` (combining three requests into one), and `on`, `bri`, `ct`, `xy`, and `dynamic_effect` could be reported in a single report (combining four reports into one);
- ZLL light state is polled with 5 second refresh;
- ZB3 light state is reported every 5 minutes (and on changes) with fallback polling.  Note the `colormode` and `effect` aren't reported, so these are polled every 5 seconds.

## TODO
- Make `startup` updatable from API;
- Make `state/dynamic_effect` updatable from API.

## ISSUES
- How to select lights based on manufacturer, profile (ZLL vs ZB3), device type, and presence of 0xFC03 cluster?  I would prefer not to have to whitelist all model identifiers.
- How to specify selective polling of _Level Control_, _Color Control_: only when light is on?  Note that you'd need to specify somehow the order of the clusters for this; currently clusters are polled in reverse order: first _Color Control_, then _Level Control_, then _On/Off_.  Note that this would still apply when using 0xFC03/0x0002, as `colormode` and `effect` aren't in there.
- `startup/xy` incorrectly reports the regular `state/xy` values: DDF doesn't seem to honour `mf` and uses the regular attributes instead.  Note Hue lights use manufacturer-specific attributes 0x0003 and 0x0004 for startup colour.
- `state/dynamic_effect`: How to specify command payload/parameters from DDF?  Or do we still need C++ code for that?
- Combine more than four attributes in a single `read` for polling - more than four doesn't seem to work.  Currently poll _Current X_, _Current Y_, and _Color Temperature_ separately from _Enhanced Color Mode_ and _Color Loop Active_.

ebaauw, 2022-10-04:
Ping

ebaauw, 2022-10-15:
Doing some more tests, now that setting multiple items from a script will trigger the corresponding event notifications.

Look like the light firmware no longer reports _Color Mode_ nor _Enhanced Color Mode_ correctly.  It seems to be updated only when explicitly setting a colour.

When activating colour loop, they report `hs` alright, but when deactivating colour loop, they continue to report `hs`, even though the light switched back to the previous colour mode, `xy` or `ct`.  Worse, the FC03 attribute reports an `xy` state, even when the mode before the colour loop was `ct` and the light is actually driving the white channels (instead of the RGB channels).  Same happens on a power cycle with power-on settings to previous state.  The light comes back in `ct` mode alright, but reports `xy`.

ebaauw, 2022-10-22:
I think PR is good to go.  See PR for description of new attributes.

Here's a ZB3 light:
```
$ ph get /lights/8
{
  "capabilities": {
    "color": {
      "gamut": {
        "blue": [
          0.1532,
          0.0475
        ],
        "green": [
          0.17,
          0.7
        ],
        "red": [
          0.6915,
          0.3083
        ]
      },
      "gamut_type": "C",
      "modes": [
        "ct",
        "effect",
        "hs",
        "xy"
      ]
    },
    "ct": {
      "max": 500,
      "min": 153
    },
    "effects": [
      "none",
      "colorloop",
      "candle",
      "fireplace",
      "loop"
    ]
  },
  "colorcapabilities": 31,
  "config": {
    "startup": {
      "bri": 254,
      "ct": "previous",
      "on": true,
      "xy": [
        0.9961,
        0
      ]
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "99ca432dc96688b849c3e1004ae39b55",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-10-22T12:21Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCA006",
  "name": "Entrance 1",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "xy",
    "ct": 500,
    "effect": "none",
    "hue": 0,
    "on": false,
    "reachable": true,
    "sat": 254,
    "xy": [
      0.6915,
      0.3083
    ]
  },
  "swversion": "1.93.11",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:0b:1c:ff:75-0b"
}
```

And here's a ZLL light:
```
$ ph get /lights/4
{
  "capabilities": {
    "color": {
      "gamut": {
        "blue": [
          0.167,
          0.04
        ],
        "green": [
          0.409,
          0.518
        ],
        "red": [
          0.675,
          0.322
        ]
      },
      "gamut_type": "B",
      "modes": [
        "ct",
        "effect",
        "hs",
        "xy"
      ]
    },
    "ct": {
      "max": 500,
      "min": 153
    },
    "effects": [
      "none",
      "colorloop"
    ]
  },
  "colorcapabilities": 31,
  "config": {
    "startup": {
      "bri": 254,
      "ct": "previous",
      "on": false,
      "xy": "previous"
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "78dc6494e79ecd895c2b0495f906c12a",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-10-22T12:21Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCT007",
  "name": "Extended color light 4",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "ct",
    "ct": 399,
    "effect": "none",
    "hue": 0,
    "on": false,
    "reachable": true,
    "sat": 0,
    "xy": [
      0.4759,
      0.4135
    ]
  },
  "swversion": "67.88.1",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:10:41:c0:e6-0b"
}
```

ebaauw, 2022-11-18:
After some discussion with @Thomas-Vos if we could get the `archetype` as exposed by the Hue bridge from Hue lights, I sniffed the Hue bridge pairing an LCA006 bulb.  This revealed the following manufacturer-specific attributes carried by the bulb:

Cluster | Attribute | Type | Value
-- | -- | -- | --
_Basic_ | 0x0000 | _Unsupported_
_Basic_ | 0x0001 | uint32 | 0
_Basic_ | 0x0003 | _Unsupported_
_Basic_ | 0x0040 | cstring | Philips-LCA006-1-A60HECLv1
_Basic_ | 0x0041 | map32 | 0xa7e a4812
_Basic_ | 0x0050 | map32 | 0x0000 0001
_Basic_ | 0xf109 | boolean | true
_Level Control_ | 0x0003 | uint16 | 200
0xfc01 | 0x0000 | map8 | 0x0b
0xfc01 | 0x0002 | _Unsupported_
0xfc03 | 0x0001 | map32 |  0x0000 0007
0xfc03 | 0x0002 | ostring | 0f:00:01:fe:6e:01:0f:75:f5:68
0xfc03 | 0x0011 | map16 | 0x0001
0xfc03 | 0x0012 | _Unsupported_
0xfc03 | 0x0013 | _Unsupported_
0xfc03 | 0x0030 | _Unsupported_

There's also a dialogue between the bridge and the bulb using manufacturer-specific commands on the _Basic_ cluster.  The bridge sends 0xc0 and the bulb responds with 0xc1.  The payload for 0xc0 seems to be the starting point and max length; the payload for 0xc1 seems (at least in part) TLV encoded.

0xc0 | 0xc1
-- | --
000000000040 | 0000000000004c000000350a4a0801158d667a8b1a410a064c434130303612185369676e696679204e65746865726c616e647320422e562e420e48756520636f
003500000040 | 0000350000004c000000176c6f72206c616d704802520b080b1207089f0810031801

Trying to decode the 0xc1 payload:
```
 0                    1                      2                      3                    4                   5                     6
 0 1 2 3 4 5 6 7 8 9  0  1  2 3  4 5 6 7 8 9 0 1  2 3  4 5 6 7 8 9  0 1  2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5  6 7  8 9 0 1 2 3
0000000000004c000000 35 0a 4a08 01158d667a8b1a41 0a06 4c4341303036 1218 5369676e696679204e65746865726c616e647320422e562e 420e 48756520636f
                                                       L C A 0 0 6       S i g n i f y   N e t h e r l a n d s   B . V .       H u e   c o

 0                    1                        2                     3        
 0 1 2 3 4 5 6 7 8 9  0  1  2 3  4 5 6 7  8 9  0 1  2 3  4 5 6 7 8 9 0 1 2
0000350000004c000000 17 6c6f72206c616d70 4802 520b 080b 1207089f0810031801
                         l o r   l a m p
```

ebaauw, 2022-11-18:
Added these newly discovered attributes to `general.xml` and queried all Hue lights I've currently online (LST001, LWB006, LCT007, LCT015, LCA006, LCE002, LCG002, LCL003, LCX004, LCX012, Hue Ensis, latest Hue Bloom).

As for the _Basic_ cluster attributes:
- 0x0000, 0x0002, and 0x0003 are only supported on older bulbs, but even then their values are empty string or 0.
- 0x0001 seems supported by all but the gamut-A lights, but only shows value 0.
- 0x0040 is indeed the `productid` from the Hue API v1;
- 0x0041 seems to be a product identifier, but doesn't match the 10-digit code from https://www.assets.signify.com/is/content/Signify/Assets/hue/united-states/20220124-effects-compatible-lights.pdf.
- 0x0050 is 0x0000 0001 for all lights.  0xf109 is not supported on older lights, but is `true` for the others.

0x0003 on the _Level Control_ cluster matches the `min_dim_level` as exposed by the Hue API v2 times 1000.  So a value of 1000 is exposed as 1.0 %.

ebaauw, 2022-11-19:
Here's the ZB3 light:
```
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect",
      "blink",
      "breathe",
      "okay",
      "channelchange",
      "finish",
      "stop"
    ],
    "bri": {
      "min_dim_level": 0.2
    },
    "color": {
      "ct": {
        "computes_xy": true,
        "max": 500,
        "min": 153
      },
      "effects": [
        "none",
        "colorloop",
        "candle",
        "fireplace",
        "loop"
      ],
      "gamut_type": "C",
      "modes": [
        "ct",
        "effect",
        "hs",
        "xy"
      ],
      "xy": {
        "blue": [
          0.1532,
          0.0475
        ],
        "green": [
          0.17,
          0.7
        ],
        "red": [
          0.6915,
          0.3083
        ]
      }
    }
  },
  "colorcapabilities": 31,
  "config": {
    "bri": {
      "execute_if_off": true,
      "startup": "previous"
    },
    "color": {
      "ct": {
        "startup": "previous"
      },
      "execute_if_off": true,
      "xy": {
        "startup": "previous"
      }
    },
    "groups": [
      "0",
      "256"
    ],
    "on": {
      "startup": "previous"
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "e1254c27ff8fa25592babfeb8e3110d0",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-11-19T21:10Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCA006",
  "name": "Hue Gamut-C",
  "productid": "Philips-LCA006-1-A60HECLv1",
  "state": {
    "alert": "none",
    "bri": 57,
    "colormode": "xy",
    "ct": 366,
    "effect": "candle",
    "hue": 7846,
    "on": false,
    "reachable": true,
    "sat": 188,
    "xy": [
      0.4923,
      0.4149
    ]
  },
  "swversion": "1.93.11",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:0b:1c:ff:75-0b"
}
```

ebaauw, 2022-11-19:
Here's a gamut-C ZLL light:
```
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect",
      "blink",
      "breathe",
      "okay",
      "channelchange",
      "finish",
      "stop"
    ],
    "bri": {
      "min_dim_level": 1
    },
    "color": {
      "ct": {
        "computes_xy": true,
        "max": 500,
        "min": 153
      },
      "effects": [
        "none",
        "colorloop"
      ],
      "gamut_type": "C",
      "modes": [
        "ct",
        "effect",
        "hs",
        "xy"
      ],
      "xy": {
        "blue": [
          0.1532,
          0.0475
        ],
        "green": [
          0.17,
          0.7
        ],
        "red": [
          0.6915,
          0.3083
        ]
      }
    }
  },
  "colorcapabilities": 31,
  "config": {
    "bri": {
      "startup": "previous"
    },
    "color": {
      "ct": {
        "startup": "previous"
      },
      "xy": {
        "startup": "previous"
      }
    },
    "groups": [
      "0",
      "256"
    ],
    "on": {
      "startup": false
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "e47c4cecb378c65544a5f7d7b6a76e75",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-11-19T21:12Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCT015",
  "name": "Hue Gamut-C ZLL",
  "productid": "Philips-LCT015-2-A19ECLv5",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "ct",
    "ct": 366,
    "effect": "none",
    "hue": 8417,
    "on": false,
    "reachable": true,
    "sat": 140,
    "xy": [
      0.4573,
      0.41
    ]
  },
  "swversion": "1.88.1",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:04:34:2c:ff-0b"
}
```

ebaauw, 2022-11-19:
Here's a gamut-B ZLL light:
```
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect",
      "blink",
      "breathe",
      "okay",
      "channelchange",
      "finish",
      "stop"
    ],
    "bri": {
      "min_dim_level": 2
    },
    "color": {
      "ct": {
        "computes_xy": true,
        "max": 500,
        "min": 153
      },
      "effects": [
        "none",
        "colorloop"
      ],
      "gamut_type": "B",
      "modes": [
        "ct",
        "effect",
        "hs",
        "xy"
      ],
      "xy": {
        "blue": [
          0.167,
          0.04
        ],
        "green": [
          0.409,
          0.518
        ],
        "red": [
          0.675,
          0.322
        ]
      }
    }
  },
  "colorcapabilities": 31,
  "config": {
    "bri": {
      "startup": "previous"
    },
    "color": {
      "ct": {
        "startup": "previous"
      },
      "xy": {
        "startup": "previous"
      }
    },
    "groups": [
      "0",
      "256"
    ],
    "on": {
      "startup": false
    }
  },
  "ctmax": 500,
  "ctmin": 153,
  "etag": "438febf7ab2cd835fc0be299aec9751a",
  "hascolor": true,
  "lastannounced": null,
  "lastseen": "2022-11-19T21:12Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LCT007",
  "name": "Hue Gamut-B ZLL",
  "state": {
    "alert": "none",
    "bri": 10,
    "colormode": "xy",
    "ct": 500,
    "effect": "none",
    "hue": 0,
    "on": false,
    "reachable": true,
    "sat": 254,
    "xy": [
      0.675,
      0.322
    ]
  },
  "swversion": "67.88.1",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:10:41:c0:e6-0b"
}
```

ebaauw, 2022-11-19:
I added the 0xC0 / 0xC1 commands to the _Basic_ cluster, for querying the Hue light capabilities.
If the payload length is larger than 0x35, you need to issue multiple 0xC0 commands to get the full payload.  Note that the requested length of 0x40 (64) is including the 10 bytes before the payload and the length byte, so the first 0xC1 returns 0x35 (53) bytes.  Hence the next 0xC0 uses an offset of 0x35.

[image]

ebaauw, 2022-11-20:
Also did the Hue smart plug:
```
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect",
      "blink",
      "breathe",
      "okay",
      "channelchange",
      "finish",
      "stop"
    ]
  },
  "config": {
    "groups": [
      "0"
    ],
    "on": {
      "startup": false
    }
  },
  "etag": "edbf69445ba97934cc35ddd94e8f7a79",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2022-11-20T15:24Z",
  "manufacturername": "Signify Netherlands B.V.",
  "modelid": "LOM007",
  "name": "Hue smart plug",
  "productid": "Philips-LOM007-1-SPPIUv2",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swconfigid": "493EC159",
  "swversion": "1.93.6",
  "type": "On/Off plug-in unit",
  "uniqueid": "00:17:88:01:0b:df:e7:98-0b"
}
```

closed 2023-01-07 by manup