dresden-elektronik/deconz-rest-plugin #6194: Unable to connect to any device and paring not working after upgrading to 2.17.01 / 26780700

faxioman, 2022-07-12:
## Describe the bug
After upgrading to 2.17.01 / 26780700 (my previous version was 2.12.06 / 26720700) I lost all my lights (are greyed out) and sensors (are unreachable).
Many times when executing an upgrade I need to pair again all devices, like for a lot of deconz users, but this time I'm unable to pair again any device

## Steps to reproduce the behavior
Upgrade to 2.17.01 / 26780700, all lost. After a reboot, no change and all device are still unreachable.

## Expected behavior
Upgrade procedure works and stops to loose my devices ... like a lot of other times ;)

## Screenshots
Screenshot not applicable

## Environment
 - Host system: Raspberry Pi (Zero W)
 - Running method: Raspbian
 - Firmware version: 26780700
 - deCONZ version: 2.17.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes, microUSB to USB
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
```
Jul 12 21:42:49 raspberrypi deCONZ[5298]: 21:42:41:227 DB saved in 135 ms
Jul 12 21:42:49 raspberrypi deCONZ[5298]: 21:42:42:130 Daylight now: nauticalDusk, status: 220, daylight: 0, dark: 1
Jul 12 21:42:49 raspberrypi deCONZ[5298]: 21:42:42:130 apsUseExtPanid is 0x0000000000000001 but should be 0, start reconfiguration
Jul 12 21:42:49 raspberrypi deCONZ[5298]: 21:42:42:130 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF078E07, start reconfiguration
Jul 12 21:42:49 raspberrypi deCONZ[5298]: 21:42:42:130 Skip automatic channel change, TODO warn user
```

## Additional context
I have many Ikea bulbs and lights and many Aqara sensor

Mimiix, 2022-07-12:
Hi,

As this is not a bug report as described in #5113 I am closing this issue. Feel free to open an topic on the forum.

Kind regards,
Dennis

closed 2022-07-12 by Mimiix

faxioman, 2022-07-12:
Hi @Mimiix, I see...
I'm a long time user of deConz but I have always less time to invest on this "hobby".
I recently bought a ConBeeII and this time I prefer to send back the device to Amazon a switch to another and more reliable zigbee solution. Unfortunately I have really no time to invest in forum trying to find a solution ... afterall I was just upgrading my ConBee from web interface 😅
Thank for your support!

Bye.

Mimiix, 2022-07-12:
You can litterally copy paste this post on the forums to get help.

Either way, please do as you wish. It's a free world :)

faxioman, 2022-07-12:
Ok, let's see what happens, thanks!