dresden-elektronik/deconz-rest-plugin #6195: Battery life no longer reported for Xiaomi environment sensor (WSDCGQ11LM)

jamieshaw, 2022-07-13:
## Describe the bug
Xiaomi Environmental sensor (WSDCGQ11LM) no longer reports battery after upgrading to deCONZ 2.15.  Was working fine in 2.14, potentially attributable to #5990. Other readings including temperature, humidity, and pressure seem fine.

## Steps to reproduce the behavior
Behaviour is consistent, and reproducible without steps. My home has 4 of the aforementioned sensors, all 4 report 0% battery.

## Expected behavior
Battery capacity should be reported correctly.

## Screenshots
N/A

## Environment
 - Host system: PC
 - Running method: Ubuntu 18.04 Virtual Machine under ESXi
 - Firmware version: (26720700)
 - deCONZ version: (2.15.3)
 - Device: ConBee II
 - Do you use an USB extension cable: Yes — mains-powered USB 2.0 hub.
 - Is there any other USB or serial devices connected to the host system? Yes — Aeotec Z-Wave 7 stick, bluetooth adapter.

## deCONZ Logs
N/A

## Additional context
Reopening closed but as issue is still relevant.

jamieshaw, 2022-07-13:
I can confirm I am on the latest, I just copy and pasted my previous ticket as it was closed without any input.

I have four of these sensors, and since updating to the aforementioned version, all four report 0% battery but still output data.

I attempted to re-pair one of the devices this morning to see if it would spur something to life, alas, it did not.

jamieshaw, 2022-07-14:
> Mine are all doing fine. How do they look like from an API perspective?

As in what is returned via the REST API? Same deal.  Home Assistant sees 0% battery, but temperature, pressure, and humidity are returned just fine.

jamieshaw, 2022-07-14:
Have configured the GUI (I'm running headless) and captured the recommended logs.  These have been attached to the original post, if they are of any help.

Feel free to ask away for further debug data if it helps.

Also worth noting that these _aren't_ the latest version of the sensors following the rebranding of Aqara and it's v3 hub.  These were purchased several years ago alongside a (then) Mijia Mi Home hub.

[image]

KingTomaHawk, 2022-07-15:
I received the same sensor today and I can confirm that the battery shows 0% on phoscon after pairing. The other values appear to be correct.

yamaha1983, 2022-07-19:
I updated to the latest version 2.17.1 and I also do not get a battery percentage. Always reporting 0%.
I also repaired the device entirely (delete sensor and created again with 3 new IDs [pressure, temperature,humidity]
Nothing could fix the problem, so I downgraded back to 2.14.1. That only could fixed the issue and battery is reporting fine again.

It seems that due to the structureal change to the DDF format, new errors are coming in. Same with motion sensor and lux values. Not working in the newest version.

Mimiix, 2022-07-19:
> Not working in the newest version

Not true, here it is working:
[image]

jamieshaw, 2022-07-20:
> > Not working in the newest version
> 
> 
> 
> Not true, here it is working:
> 
> [image]
> 
> 

Either way, it _clearly_ is and _isn't_ working for some users, myself included, stemming from a reproducible change in a particular version.

An investigation and fix is **definitely** required.

ebaauw, 2022-07-20:
Yesterday, I reset the `lumi.weather` sensor connected to my Aqara hub and paired it with my dev/test deCONZ, v2.17.1-beta.  This morning, the API shows a value in `config.battery`.  The API plugin is using the DDF to expose the sensor.  I have set _DDF Mode_ in the _Control_ panel to _Basic_, with only _Gold_ DDF status enabled.

If you enable _DDF_ in the _Debug View_, you should see messages like:
```
Jul 20 07:36:57 pi5 deCONZ[7310]: 07:36:43:123 00:15:8d:00:07:6d:68:12-01-0405/config/battery expression: const vmin = 2700;
Jul 20 07:36:57 pi5 deCONZ[7310]: 07:36:43:154 00:15:8d:00:07:6d:68:12-01-0402/config/battery expression: const vmin = 2700;
Jul 20 07:36:57 pi5 deCONZ[7310]: 07:36:43:184 00:15:8d:00:07:6d:68:12-01-0403/config/battery expression: const vmin = 2700;
```
every 55-60 minutes (as deCONZ receives the special attribute report form the sensor).

jamieshaw, 2022-07-20:
> Yesterday, I reset the `lumi.weather` sensor connected to my Aqara hub and paired it with my dev/test deCONZ, v2.17.1-beta. This morning, the API shows a value in `config.battery` alright. The API plugin is using the DDF to expose the sensor. I have set _DDF Mode_ in the _Control_ panel to _Basic_, with only _Gold_ DDF status enabled.
> 
> If you enable _DDF_ in the _Debug View_, you should see messages like:
> 
> ```
> Jul 20 07:36:57 pi5 deCONZ[7310]: 07:36:43:123 00:15:8d:00:07:6d:68:12-01-0405/config/battery expression: const vmin = 2700;
> Jul 20 07:36:57 pi5 deCONZ[7310]: 07:36:43:154 00:15:8d:00:07:6d:68:12-01-0402/config/battery expression: const vmin = 2700;
> Jul 20 07:36:57 pi5 deCONZ[7310]: 07:36:43:184 00:15:8d:00:07:6d:68:12-01-0403/config/battery expression: const vmin = 2700;
> ```
> 
> every 55-60 minutes (as deCONZ receives the special attribute report form the sensor).

I've done just that, and left it running in the background.  The following was returned.

```
16:25:32:258 failed to evaluate expression for 00:15:8d:00:02:d7:7f:c2-01-0405/config/battery: const vmin = 2700;
const vmax = 3000;
let bat = Attr.val;

if      (bat > vmax) { bat = vmax; }
else if (bat < vmin) { bat = vmin; }

bat = ((bat - vmin) /(vmax - vmin)) * 100;

if      (bat > 100) { bat = 100; }
else if (bat <= 0)  { bat = 1; } // ?

Item.val = bat;
, err: SyntaxError: Syntax error
16:25:32:260 failed to evaluate expression for 00:15:8d:00:02:d7:7f:c2-01-0402/config/battery: const vmin = 2700;
const vmax = 3000;
let bat = Attr.val;

if      (bat > vmax) { bat = vmax; }
else if (bat < vmin) { bat = vmin; }

bat = ((bat - vmin) /(vmax - vmin)) * 100;

if      (bat > 100) { bat = 100; }
else if (bat <= 0)  { bat = 1; } // ?

Item.val = bat;
, err: SyntaxError: Syntax error
16:25:32:262 failed to evaluate expression for 00:15:8d:00:02:d7:7f:c2-01-0403/config/battery: const vmin = 2700;
const vmax = 3000;
let bat = Attr.val;

if      (bat > vmax) { bat = vmax; }
else if (bat < vmin) { bat = vmin; }

bat = ((bat - vmin) /(vmax - vmin)) * 100;

if      (bat > 100) { bat = 100; }
else if (bat <= 0)  { bat = 1; } // ?

Item.val = bat;
, err: SyntaxError: Syntax error
```

ebaauw, 2022-07-20:
That looks to be an error in the DDF file, or in the script to translate the battery voltage reported by the sensor to a percentage.  What version are you on?  What DDF file are you using for the sensor?

jamieshaw, 2022-07-21:
> That looks to be an error in the DDF file, or in the script to translate the battery voltage reported by the sensor to a percentage. What version are you on? What DDF file are you using for the sensor?

2.17.01 / 01/07/2022

As for the file, not sure about the question, haven't changed anything so _should_ be the default?

jamieshaw, 2022-07-24:
If you can point me in the right direction of the DDF agile, I'm more than happy to help.

The installation was done through APT on Ubuntu, but haven't touched anything regarding the installation files other than updates through APT.

bee-con, 2022-07-24:
> If you can point me in the right direction of the DDF agile, I'm more than happy to help.

You could try this: Open DeCONZ, right-click on the sensor, choose "Edit DDF".
This will open the DDF editor in a new window, the title bar will display the name of the DDF.
Then click "File" --> "Save as", and a dialog window will show you where the file is located.
Then click "Cancel" to avoid changing the DDF.

On my Raspberry Pi, the DDF for your sensor is located here:
/usr/share/deCONZ/devices/xiaomi/xiaomi_wsdcgq11lm_temp_hum_press_sensor.json

```
root@raspberrypi:/# cd /usr/share/deCONZ/devices/xiaomi

root@raspberrypi:/usr/share/deCONZ/devices/xiaomi# grep WSDCGQ11LM *
xiaomi_wsdcgq11lm_temp_hum_press_sensor.json:  "product": "Aqara temp./hum./press. sensor WSDCGQ11LM",

root@raspberrypi:/usr/share/deCONZ/devices/xiaomi# grep "xiaomi_battery.js" xiaomi_wsdcgq11lm_temp_hum_press_sensor.json
          "parse": {"fn": "xiaomi:special", "at": "0xff01", "idx": "0x01", "script": "xiaomi_battery.js"}
          "parse": {"fn": "xiaomi:special", "at": "0xff01", "idx": "0x01", "script": "xiaomi_battery.js"}
          "parse": {"fn": "xiaomi:special", "at": "0xff01", "idx": "0x01", "script": "xiaomi_battery.js"}

root@raspberrypi:/usr/share/deCONZ/devices/xiaomi# cat "xiaomi_battery.js"
const vmin = 2700;
const vmax = 3000;
let bat = Attr.val;

if      (bat > vmax) { bat = vmax; }
else if (bat < vmin) { bat = vmin; }

bat = ((bat - vmin) /(vmax - vmin)) * 100;

if      (bat > 100) { bat = 100; }
else if (bat <= 0)  { bat = 1; } // ?

Item.val = bat;
root@raspberrypi:/usr/share/deCONZ/devices/xiaomi#

```

jamieshaw, 2022-07-25:
> You could try this: Open DeCONZ, right-click on the sensor, choose "Edit DDF".
>
> This will open the DDF editor in a new window, the title bar will display the name of the DDF.
>
> Then click "File" --> [...]

Thanks. Followed this and seeing the same, pretty sure I'm seeing the same file at `xiaomi_battery.js`.

```
const vmin = 2700;
const vmax = 3000;
let bat = Attr.val;

if      (bat > vmax) { bat = vmax; }
else if (bat < vmin) { bat = vmin; }

bat = ((bat - vmin) / (vmax - vmin)) * 100;

if      (bat > 100) { bat = 100; }
else if (bat <= 0)  { bat = 1; } // ?

Item.val = bat;
```

hellow554, 2022-07-28:
Same issue for me, 0% battery shown. What's different is, that I have a newer aqara version.

[image]
[image]

jamieshaw, 2022-07-31:
Awaiting an update. 🙂

jamieshaw, 2022-08-13:
Still waiting. Any ETA on a potential fix?

jamieshaw, 2022-08-17:
So, interestingly, this issue has been resolved… somehow.

I'm currently in the middle of rebuilding the VM that I use for hosting all my home automation stuff, and today, reinstalled deCONZ (same method, via APT), and simply migrated over the `zll.db` file—Battery is now being reported.

Guessing something went funky during the upgrade process and that the clean install on a new machine remedied this.
Just had another dig into the following files, and can confirm **both** are identical between working and non-working instances.
`/usr/share/deCONZ/devices/xiaomi/xiaomi_wsdcgq11lm_temp_hum_press_sensor.json`
`/usr/share/deCONZ/devices/xiaomi/xiaomi_battery.js`

🤷‍♂️

closed 2022-08-17 by jamieshaw

SwoopX, 2022-08-18:
Hm, as good as it is that your issue is resolved through the migration, as unsatisfying it is to still tap in the dark regarding the root cause.

Do you still have the old VM and could compare software packages, qt to be specific?

jamieshaw, 2022-08-19:
> Hm, as good as it is that your issue is resolved through the migration, as unsatisfying it is to still tap in the dark regarding the root cause.
> 
> 
> 
> Do you still have the old VM and could compare software packages, qt to be specific?

I do – just let me know what you wish me to capture and I can provide. 🙂

SwoopX, 2022-09-10:
@jamieshaw Riddle's been solved, see the linked issue. Maybe you can share your Qt version there...