dresden-elektronik/deconz-rest-plugin #6198: Connecte Smart Thermostat

joernnilsson, 2022-07-16:
## Device

- Product name: Connecte Smart Thermostat
- Manufacturer: _TZE200_4hbx5cvx
- Model identifier: TS0601
- Device type:
  - Thermostat

Manufacturer web page: https://en.connecte.no/termostat

I do have a contact to the manufacturer, and can probably request any missing/useful documentation.

## Screenshots

### Basic
[image]

### Identify
[image]

### Thermostat
Multiple screenshots to see whole attribute list
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Tuya Specific
[image]

joernnilsson, 2022-07-25:
I have received documentation/app note from the manufacturer, please see attached pdf

It seems that most functions are reachable through the Tuya specific cluster
[App note.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9179958/App.note.pdf)
. Is this outside the scope of deconz?

Smanar, 2022-07-25:
For the moment tuya TRV are not possible usinf DDF waiting for this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054

joernnilsson, 2022-07-25:
Ok, thanks :+1:

joernnilsson, 2022-09-13:
Please keep this issue open. It is waiting for [#6054 ](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054) to be merged.

joernnilsson, 2022-10-06:
Bumping, as we're still waiting for https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054 to be merged

joernnilsson, 2022-10-30:
Bumping to avoid auto close, as we're still waiting for https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054 to be merged.

joernnilsson, 2022-11-22:
Bumping to avoid auto close. Still waiting for https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054 to be merged.

closed 2022-12-29 by github-actions[bot]