dresden-elektronik/deconz-rest-plugin #6199: MOES 2 Gang Switch _TZE200_g1ib5ldv

sn0wf4ke, 2022-07-16:
## Device

- Product name: MOES 2 Gang Switch (https://www.moeshouse.com/collections/smart-switch/products/zigbee-smart-wall-light-switch-with-neutral-wire-or-no-neutral-wire-wiring-no-capacitor-needed-smart-life-tuya-2-3-way-muilti-control-association-works-with-alexa-google-home-hub-required?variant=39357460578385)
- Manufacturer: _TZE200_g1ib5ldv
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
   - Switch

## Screenshots
[image]
[image]

### Basic
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### Tuya specific
[image]

### OTAU
[image]

### Time
[image]

sn0wf4ke, 2022-07-17:
ok I figured out how to bring one gang up inspired by #5869. 
My DDF now looks this way:
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_g1ib5ldv",
  "modelid": "TS0601",
  "vendor": "$MF_TUYA",
  "product": "Smart Switch Module",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "public": false,
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 84000,
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 1,
            "dt": "0x10",
            "eval": "Item.val == 1 ? 1 : 0;",
            "fn": "tuya"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```
But how to control the second gang?

closed 2022-08-15 by github-actions[bot]

forast3r, 2022-08-19:
> ok I figured out how to bring one gang up inspired by #5869. My DDF now looks this way:
>
> ```json
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZE200_g1ib5ldv",
>   "modelid": [...]

Did you manage to get the 2nd gang working? (I have received a similar one but 3-Gang)

mweedon, 2022-11-28:
Any news on the 2 or 3-gang DDF?

I'm thinking it's time to migrate to zigbee2mqtt. :-(