dresden-elektronik/deconz-rest-plugin #6201: ConBee II - Hue light doesn't work individually - but they work in group

dawiinci, 2022-07-16:
## Describe the bug
This seems to be a never-ending issue as I've seen other issues in 2019:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1883

The lights are in plain sight, direct connection to the coordinator. Aqara sensors work instantly.

However, the lights don't work randomly. All of them are at some point of time not possible to switch on or off. I have multiple lights (first gen Hue bulb, latest gen Hue bulb and other lights from Philips Hue).

Dongle is on an extension cable - connected to M1 Mac (USB-A). I set serial baud rate to 38400 - no rts/cts. Tested with ZHA and ZigBee2MQTT.

The stick is brand new and on version 0x26720700 - not updated. I tried both zigbee2mqtt and zha, the issue remains the same.

I already replaced the stick because it started to fail after a firmware upgrade. The new stick has a newer firmware already which is not working out of the box.

## Steps to reproduce the behavior
connect hue light
switch it on and off for some time
it will fail eventually

If I switch on and off the group that contains all lights they DO work.

## Expected behavior
work

## Screenshots
no

## Environment

 - Host system: Mac M1
 - Running method: Zigbee2MQTT, ZHA doesn't matter
 - Firmware version: (0x26720700)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? 
    z-wave on separate extension cable, hdd and rfxcom on separate usb hub

## deCONZ Logs
no

## Additional context
hue, aqara

Please do not buy this stick if you plan using hue lights - unless they finally fix this!
Very disappointed!

Mimiix, 2022-07-16:
> Please do not buy this stick if you plan using hue lights - unless they finally fix this!
Very disappointed!

I have multiple in use without any issues. 

Can you please provide some logs with log levels APS, APS_L2, error, ERROR_L2, info, INFO_L2.

I am tempted to close this as it isn't a bug at first sight(#5113), but I'm happy to give it a go.

dawiinci, 2022-07-16:
I can give you this log of an attempt to switch on the light. 

```
Info 2022-07-16 17:46:56Succesfully changed options
Info 2022-07-16 17:46:57MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"591ad-1"}'
Debug 2022-07-16 17:47:05Received MQTT message on 'zigbee2mqtt/Licht Stehleuchte/set' with data '{"state":"ON"}'
Debug 2022-07-16 17:47:05Publishing 'set' 'state' to 'Licht Stehleuchte'
Error 2022-07-16 17:48:06Publish 'set' 'state' to 'Licht Stehleuchte' failed: 'Error: Command 0x0017880100e6c6af/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Debug 2022-07-16 17:48:06Error: Command 0x0017880100e6c6af/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.sendZclFrameToEndpoint (/Users/server/Steuerung/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:654:23) at runMicrotasks (<anonymous>) at runNextTicks (node:internal/process/task_queues:61:5) at processTimers (node:internal/timers:499:9) at Endpoint.sendRequest (/Users/server/Steuerung/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20) at Endpoint.command (/Users/server/Steuerung/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:772:28) at Object.convertSet (/Users/server/Steuerung/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:283:17) at Object.convertSet (/Users/server/Steuerung/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:906:32) at Publish.onMQTTMessage (/Users/server/Steuerung/zigbee2mqtt/lib/extension/publish.ts:246:36)
Info 2022-07-16 17:48:06MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Licht Stehleuchte' failed: 'Error: Command 0x0017880100e6c6af/11 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Licht Stehleuchte"},"type":"zigbee_publish_error"}'
```

ebaauw, 2022-07-16:
This is not an issue with the deCONZ REST plugin.

In my experience devices reacting to group commands, but not to unicast commands indicates an issue with the Zigbee network.  Either interference or routing issues.  With deCONZ I'd check for Zigbee error codes in the deCONZ log (see Wiki) indicating radio issues.  Routing is harder to diagnose, I'd try turning on _Source Routing_ in the deCONZ GUI.  Or running a Zigbee sniffer to analyse route discovery.  I have no clue how to diagnose this using ZHA or Z2M.

I'm frequently facing USB3 interference from my Intel mac mini for the Bluetooth connection of my magic keyboard and mouse and Zigbee is more vulnerable to interference than Bluetooth.

Mimiix, 2022-07-16:
In that case I'm closing this issue.

closed 2022-07-16 by Mimiix

dawiinci, 2022-07-17:
Well I don't have any active bluetooth and I even disabled it, wifi is on the other side of the spectrum and I switched off all possible devices nearby. There is no apparent reason why it fails, except the dongle. Many people have this issue. For me it was rock solid until I made a firmware update. 

Seems like there is a combination that causes this bug. But I guess I will switch to another dongle and avoid the hassle find/explain the bug.

dawiinci, 2022-07-27:
While waiting for a new controller...

All my devices have an LQI of 255. That is weird, right.

Also, why is a group reacting instantly, but the single device does not? Also why are my sensors working?

dawiinci, 2022-08-05:
Just for you information. I replaced with a Sonoff Dongle and it works as expected. 

The LQI is lower but more realistic and all lights work. I guess you still have a big bug in your firmware.

I hope you can fix it, but for me the new dongle fixes all issues.