dresden-elektronik/deconz-rest-plugin #6210: MOCREO ST-4 temperature sensor with external probe

srd424, 2022-07-20:
This one may not be possible, as the manufacturer & model ID are not populated, but there is an identifier in the "Date Code" attribute in the first "Basic" cluster that could be used (I appreciate this probably requires a device specific hack, which may be undesirable.) The actual probe temperature seems to be read just fine, it appears in the first temperature cluster (under "01".) The humidity measures just seem to return 0. The second temperature cluster (under "08") doesn't appear to return sensible data.

If this is too much of a disaster zone then feel free to close :)

## Device

- Product name: MOCREO Temperature Sensor - ST4
- Manufacturer: MOCREO (not shown in basic cluster)
- Model identifier: ST-4 (not shown in basic cluster)
- Device type : Please remove all unrelated device types. 
  - Sensor

https://www.amazon.co.uk/MOCREO-Temperature-Waterproof-Thermometer-Refrigerator/dp/B09G9MG19D

## Screenshots

[image]
[image]

### Basic

[image]

### Temperature measurement

[image]

srd424, 2022-07-20:
I don't know - this is the only device I've got by them so far! I'm not a Zigbee expert, but it smacks of "we cloned the example in the SDK and filled in the bare minimum to get it working with our own hub" :( Annoying though because it's the only temperature sensor with an external probe I can get hold of here in the UK..

Smanar, 2022-07-20:
This one is not available in UK ? https://www.domadoo.fr/fr/peripheriques/5998-owon-sonde-de-temperature-deportee-sur-cable-zigbee.html

srd424, 2022-07-20:
No - and they won't ship to the UK either now because brexit has made the costs/paperwork too ludicrous :( I may be able to get some sent to a friend's parents in France and then bought over..

srd424, 2022-08-11:
FWIW, I've switched zigbee2mqtt, because I can at least make this sensor sort-of work with that codebase (its "interview" of the device fails, but its fingerprint concept allows me to match the device and declare what it exposes, and it emits recognizable temperature readings OK.)

closed 2022-08-11 by srd424