dresden-elektronik/deconz-rest-plugin #6214: Sensor missing upon pairing light

Preto, 2022-07-23:
## Describe the bug
### In short
In the C++ code there is a `ROB_200-011-0` Sunricher dimmer (#2829) that exposes both a `/light` and a `/sensor` resource. This worked flawlessly. Unfortunately adding new modules, only results in a `/light` resource. No `/sensor` is ever created. Pre-DDF releases like 2.12.6 worked like a charm, yet the latest stable version (2.17.01) with DDF support shows this issue.

The fix is to open deCONZ, then (right-)click on an older, working dimmer. Edit its DDF and promote it to Gold. Then store it at the correct location and reload deCONZ. Start resetting & pairing the dimmer module again and you will receive both the `/light` and two `/sensor` in the API just like before.

It seems the legacy C++ code doesn't pick it up and also the on-the-fly DDF's don't get used. Please make legacy work, or move it to DDF completely?

---

### Long story
In issue #2829 both the light source and the sensor of the `ROB_200-011-0` dimmer where added. Recently I've bought new ones and I've noticed only the `/light` resource is exposed in the API. Both deCONZ and the API show this resource. The Phoscon App does not show any result, but after a page refresh you'll notice its there. There might be an issue here too, although not relevant to this repo. Searching for a new device with "Add new lights" or "Add new sensor" makes no difference and will never result in a `/sensor`.

In the past the API developers did there magic to add both the light and the sensor to the API as this dimmer is a light source but also contains sensor data like consumption and power. All current dimming modules in my environment work perfectly. Also re-pairing an existing sensor will result in both `/light` and `/sensor` again.

Yet, adding a complete new dimmer does not expose a sensor. I will explain thoroughly in the next chapters why I placed the bug report in this repository.

## Steps to reproduce the behavior
1. Place the dimming module behind a light or behind a wall-swich
2. Pair it, either as new module or reset the module a couple of times.
3. Adding via _Light_ or _Sensor_ in Phoscon App results in only a `/light` resource

## Expected behavior
Like in the past and pre-DDF releases, pair and expose both the `/light` and its `/sensor` (2x) resources in the API.

## Screenshots
The screenshots below still apply.

[image]

## Environment
deCONZ and the API are not in consensus, but my deCONZ version is 2.17.01. 

```
$ date
Sat Jul 23 09:36:07 CEST 2022

$ ll /home/pi/.local/share/dresden-elektronik/deCONZ/zll.db 
-rw-r--r-- 1 pi pi 253952 Jul 23 09:36 /home/pi/.local/share/dresden-elektronik/deCONZ/zll.db

$ dpkg -s deconz | grep '^Version:'
Version: 2.17.01-debian-buster-stable

$ sqlite3 /home/pi/.local/share/dresden-elektronik/deCONZ/zll.db 'select * from config2 where key in ("swversion","apiversion")'
apiversion|2.16.01
swversion|2.16.01

$ curl -s GET http://ip/api/key/config | python -mjson.tool | egrep 'swversion|apiversion'
    "apiversion": "1.16.0",
    "swversion": "2.17.1",
```
Note, I did restart deCONZ after updating this time :-)

```
$ ps auxf | grep [d]eCONZ
pi         76637  0.0  0.0   8196  2784 ?        S    Jul22   0:00 /bin/bash /usr/bin/deCONZ-autostart.sh
pi         76668  3.8  2.5 472636 98848 ?        Sl   Jul22  37:22  \_ deCONZ --auto-connect=1 --dbg-error=1 --http-port=8080 --pid-file=/run/user/1000/deconz/deconz.pid
```

 - Host system: Raspberry Pi
 - Running method: Ubuntu 20.04.4 LTS (Focal)
 - Firmware version: 26720700
 - deCONZ version: 2.17.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

deCONZ is installed as a Debian package on Ubuntu Focal as it seems that the APT repository of DE does not have a `arm64` package in its Ubuntu repository!

```
[... 89 lines cut ...]
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0101",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
```
To get this issue fixed, I've promoted this DDF on disk from Draft to Gold. After that the sensor will get exposed and I think everything is fine.

It took me some time to figure out. As it is unexpected behavior and worked in the past, it will probably help others to get this fixed. I really hope someone can find out what happened here and fix it.

## Additional context
I am fairly certain that in Crossfade (v2.12.6) the device paired successful with all light and sensors attached.

_Note that I have anonymized the unique identifiers of the device in question._

_Note to Github moderator(s): I see a lot of closed issues regarding not using a template. It seems when you are not logged in and click on the New Issue button and login there, you will never see the template selection and will go straight to the input fields and never notice the Issue Templates at all._

SwoopX, 2022-07-23:
As this is not a bug, but a matter of configuration, I'm closing the issue. If you feel an available DDF is adequate and sufficiently adequate to deserve any promotion to Gold status, please feel free to request that in the forum or raise a corresponding PR. Thanks.

closed 2022-07-23 by SwoopX

Preto, 2022-07-23:
Hi SwoopX,

I think I understand what you mean. The implementation you did in the past for this dimmer is no longer present in the C++ code, right? So that's why it's not considered a bug anymore, but simply missing configuration c.q. a new device request? It would have been practical/nice to swap Labels then though...

Anyway, hope I've helped someone searching for this device, or also having issues with arm64 ;-)

I'll drop the DDF that deCONZ made based upon the previous dimmer module that was added through the C++ code.
Below DDF worked for me to pair the ROB_200-011-0 dimmer, advertised as _Robb Smells Zigbee Dimmer 2-Wire 400w_:

```
$ cat /home/pi/.local/share/dresden-elektronik/deCONZ/devices/rob_200-011-0.json 
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ROBB smarrt",
  "modelid": "ROB_200-011-0",
  "product": "ROB_200-011-0",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/rob_200-011-0.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
[... 473 lines cut ...]
      "dst.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x00000E10"
        },
        {
          "at": "0x0400",
          "dt": "0x2A",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

</p>
</details>