dresden-elektronik/deconz-rest-plugin #6215: Paulmann ZB Spot Adapter 968.93, 955.23, 955.24

bwatconbee, 2022-07-23:
## Device

- Product name: Urail ZB Spot Adapter
- Manufacturer: Paulmann
- Model identifier: 968.93 (silber), 955.23 (weiss), 955.24 (schwarz)
- Device type : 
  - Light

Den Adapter gibt es in 3 verschiedenen Farben mit jeweils anderen Identifikationsnummern!
Adapter wird gefunden aber nicht schalten. 
Es wird nur der Schieber zum Dimmen angezeigt. (siehe Screenshot 3!)

## Screenshots

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-07-23:
Can you please post om English and provide all screens?

Thanks!

bwatconbee, 2022-07-23:
Is it better that way? Which screens are missing exactly?

Mimiix, 2022-07-23:
Of all clusters. Either way, can you see if the rest api works correctly? I want to know if it's a phoscon or deconz issue.

bwatconbee, 2022-07-23:
What do you mean by all clusters? The screenshot with the cluster info is there! My setup consists only of the Conbee Stick and an adapter, which I got for testing! The Rest-Api only makes sense when a device is found or is responsive! What exactly can I deliver now?

bwatconbee, 2022-07-23:
OK! I think I found that with the API! The adapter is also visible via the API! See screenshot 5!

Mimiix, 2022-07-23:
If you use the api, can you control using that?

The documentation on the REST API can be found here : https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

bwatconbee, 2022-07-24:
The way I see it, there is the same result with the API! Please see the last Screenshots! 
The put commands are successfully issued but there is no reaction on the Paulmann adapter!
What else can I do?

Mimiix, 2022-07-24:
So for whatever reason this device doesn't play by the rules of Zigbee. In that case we need all. Screenshots from each cluster in the device ( see the wiki, how to read clusters) if you don't know how.

Once you've done that, a Dev needs to look into it..

bwatconbee, 2022-07-24:
Sorry, but it seems to work with the API! My mistake was that switching on is immediate (parameter true), but switching off takes about 2 seconds". The API feedback is immediate, the reaction on the adapter is greatly delayed!

Mimiix, 2022-07-24:
> Sorry, but it seems to work with the API! My mistake was that switching on is immediate (parameter true), but switching off takes about 2 seconds". The API feedback is immediate, the reaction on the adapter is greatly delayed!

In that case, the bug is with phoscon.

You need to open an topic on the forums. Https://forum.phoscon.de

closed 2022-07-24 by Mimiix