dresden-elektronik/deconz-rest-plugin #6222: Spruce Soil Sensor

rohrsh, 2022-07-26:
## Device

- Product name: Spruce Soil sensor
- Manufacturer: PLAID SYSTEMS
- Model identifier: PS-SPRZMS-SLP3
- Device type : Please remove all unrelated device types. 
  - Sensor

Expecting to read temperature and relative humidity (for soil moisture)

See https://zigbee.blakadder.com/Plaid_PS-SPRZMS-SLP3.html

## Screenshots

### Basic
[image]

### Identify

[image]

### Alarms

### Device Temperature

[image]

### Relative Humidity

[image]

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

### Power

[image]

### Overall
[image]

Smanar, 2022-07-26:
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "PLAID SYSTEMS",
  "modelid": "PS-SPRZMS-SLP3",
  "vendor": "Plaid Systems",
  "product": "Spruce Soil sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

For the battery I m using the attribute 0x0020, but both attribute are at 0 on your capture (0x0020 and 0x0021, cluster 0x0001)

rohrsh, 2022-07-27:
Nice one!

The device appears in my web interface: 

[image]

Humidity has been reporting either 0% (when on my desk) or 45% (when in a glass of water). I'll put it in some soil now to see if it functions differently when in proper use.

rohrsh, 2022-07-27:
Another source https://www.zigbee2mqtt.io/devices/PS-SPRZMS-SLP3.html

Smanar, 2022-07-27:
Z2m use too the cluster 0x0001 for battery.
The one I have used 0x0021 is still not working ? (You gave 0% battery on your capture), can you try to read again the attributes on this cluster to check wich one attribute is working ?

rohrsh, 2022-07-28:
[image]

Smanar, 2022-07-28:
Haaa, it use the attribute 0x0000, not classic.
Try with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "PLAID SYSTEMS",
  "modelid": "PS-SPRZMS-SLP3",
  "vendor": "Plaid Systems",
  "product": "Spruce Soil sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0000",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 2000; const vmax = 3000; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0001", "at": "0x0000"},
          "refresh.interval": 3700,
          "default": 0
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

I m using a formulea to convert the value 3075 to a percent.

WhistleMaster, 2022-08-06:
It's working fine with your latest DDF @Smanar ! Thanks !
[image]

Smanar, 2022-08-06:
Nice I submit the PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

closed 2022-09-04 by github-actions[bot]