dresden-elektronik/deconz-rest-plugin #6224: GCFFlasher to list current firmware version

mmattel, 2022-07-26:
## Describe the bug
Got my new ConBee II an flashed the firmware with the W10 command line tool - worked.
What I missed was to get the current firmware release when running the -l list devices command.
As you can write to it, you can read too and it would not be a big thing adding this to the current output.
I do only want to flash and I am not using the deconz sw.

## Steps to reproduce the behavior

Run `GCFFlasher -l`

## Expected behavior
Output also writes current firmware

## Screenshots
none

## Environment
 - Host system: W10x64
 - Running method: plain
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Not relevant because only using the flash command.

## Additional context
Note, I am using a headless system where the ConBee II will be connected to without running deconz. W10 is just for flashing as there is no pure GCFFlasher alone and I do not install a full blown package that needs to get stopped just for using the flasher - maybe to think about providing the flasher alone...

Smanar, 2022-07-26:
>W10 is just for flashing as there is no pure GCFFlasher alone

Just by curiosity, have you tried this flasher https://github.com/dresden-elektronik/gcfflasher ?

Mimiix, 2022-07-26:
Hi,

As this is not a bug report as described in #5113 I am closing this issue.

Please use this github https://github.com/dresden-elektronik/gcfflasher

Kind regards,

closed 2022-07-26 by Mimiix

mmattel, 2022-07-27:
Note, would be nice if you would point readers of your documentation to the [gcfflasher](https://github.com/dresden-elektronik/gcfflasher) repo.
There is no mentioning of it or link provided.

Mimiix, 2022-07-27:
> Note, would be nice if you would point readers of your documentation to the [gcfflasher](https://github.com/dresden-elektronik/gcfflasher) repo. There is no mentioning of it or link provided.

Hi!

Feel free to edit the wiki documentation   :)!

mmattel, 2022-07-28:
Done, see wiki