dresden-elektronik/deconz-rest-plugin #6227: deconz_event not triggered with Xiaomi button WXKG01LM

Adriien-M, 2022-07-27:
## Describe the bug
I currently have Deconz version 6.9.0 and everything is working fine

As soon as I update Deconz to 6.10.0 or higher, Xiaomi round button (WXKG01LM / lumi.sensor_switch) no longer trigger `deconz_event`
However, the last version of the button (WXKG11LM / lumi.sensor_switch.aq2) still works correclty

## Steps to reproduce the behavior
Update Deconz from 6.9.0 to the last version,  `deconz_event` will not be triggered when you press the button

## Expected behavior
Update Deconz from 6.9.0 to a newer version shouldn't break Xiaomi WXKG01LM support

## Screenshots
N/A

## Environment
 - Host system: NAS Synology
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26580700
 - deCONZ version: 6.10.0+
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Dongle bluetooth

## deCONZ Logs
N/A

## Additional context
N/A

Mimiix, 2022-07-27:
Hi,

Can you please use the deCONZ version? I am not sure what you mean with these versions as this is the Addon version. You can find it on the gateway information page.  

I am wondering what the API provides when you click it. In PHoscon, click "Help" and select "api helper". Here you can see the events the button produces.

Adriien-M, 2022-07-27:
Oh sorry, in 6.9.0, the Deconz version is 2.11.05
Then in 6.1.0 it bumps deCONZ to 2.12.6

Mimiix, 2022-07-27:
Those are fairly old versions. I'd recommend going to the latest, which is 2.17.1.

Adriien-M, 2022-07-27:
I tried yesterday to update the plugin to the latest version which use 2.17.1 but it doesn't work
Then, I tried to update to 2.12.6, but it doesn't work either
So, I'm stuck with the version 2.11.05...

Mimiix, 2022-07-27:
Can you share the requested information when you use 2.17.1?

Hi,

Can you please use the deCONZ version? I am not sure what you mean with these versions as this is the Addon version. You can find it on the gateway information page.

I am wondering what the API provides when you click it. In PHoscon, click "Help" and select "api helper". Here you can see the events the button produces.

Adriien-M, 2022-07-27:
After 2.17.1, Events list remains empty for this kind of switch
[image]

In Deconz logs I have
```
11:10:43:425 unknown node 0x00158D0001E18F41 (0x2F2B), lqi: 255
11:10:43:425 APS-DATA.indication from unknown node 0x00158D0001E18F41
```

Mimiix, 2022-07-27:
What happens if you re-pair the switch?

Adriien-M, 2022-07-27:
Thank you, I had already tried this manipulation without success but with the latest version it works...
That's weird because on 3 switches, only one was still working after upgrade. Anyway, after re-pair, my two other switches are now working!

closed 2022-07-27 by Adriien-M