dresden-elektronik/deconz-rest-plugin #6230: IKEA Nymåne pendant lamp

cklit, 2022-07-28:
## Device

- Product name: Ikea Nymåne (pendant)
- Manufacturer: IKEA of Sweden
- Model identifier: NYMANE PENDANT
- Device type : 
  - Light

## Screenshots

[image]

[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control
[image]

### Color Control
[image]

### IKEA specific
[image]

### Green power
[image]

Mimiix, 2022-07-28:
Hi,

What is the issue with it? Isn't it already working?

cklit, 2022-07-28:
Not for me :/ Unable to control it from Home Assistant, and does not show up in Phoscon either. I have 2 lamps, its the same with both, and I've tried to reset multiple times.

cacedo, 2022-08-08:
It's working for me. I reset it by switching on/off 6 times, then add device in Home Assistant

cklit, 2022-08-10:
It might have been a local issue. I tried once again to reset the bulb (6x on/off) and search for the lights. Usually new lights show up in the bottom of the list in Phoscon, but these did not for whatever reason. But I could see in the deconz vnc panel that it found the lamps and named them "Color temperature light <x>". Once I refreshed the page in Phoscon I could see "Color temperature light" mixed together with all my other lights.

They behaved strange for about 2 mins (verrrrrry slow response, but only those 2 lights), which leads me to think that it may be something locally with my network, but now they both work perfectly. 

Closing.

closed 2022-08-10 by cklit