dresden-elektronik/deconz-rest-plugin #6236: Smoke Detector TS0601 (_TZE200_ntcy3xu1)

Rett1, 2022-07-30:
Could you please add support for the following Tuya Smoke detector?

## Device

- Product name: Zigbee Smoke detector
- Manufacturer: _TZE200_ntcy3xu1
- Model identifier: TS0601
- Device type : 
  - Other: Smoke detector

## Screenshots
[image]

### Basic
[image]

### Node Info
[image]

### OTAU
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### Tuya specific
[image]

Rett1, 2022-07-30:
https://forum.phoscon.de/t/smoke-detector-ts0601-tze200-ntcy3xu1/2283

Smanar, 2022-07-30:
There is not yet DDF for device using tuya cluster for smoke detection ?

Can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_ntcy3xu1",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Smoke sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x23",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    }
  ]
}
```

Have tried to enable fire and battery.
But this device have a siren too ?

Rett1, 2022-07-30:
Thanks for the quick response!

I haven't used DDF so far but tried to add your recommended items. 
The battery remains with "0"
```
"config": {
  "battery": 0,
```
and the state does not seem to work.
```
"state": {
  "fire": null,
  "lastupdated": "none",
  "lowbattery": null,
  "test": null
```

Maybe I'm missing something basic related to DDFs?

Smanar, 2022-07-31:
No if the device have been reconized, it probably mean you are fine.
Can you enable log (in deconz / help / debug view) with flag "info" "info_l2" and "ddf" then trigger an alarm, you will see some logs with "tuya" inside.

Rett1, 2022-08-07:
Thanks for your quick response. 

The device has been detected but I am still having trouble getting it to work properly. The battery stays at "0", the fire state at "true" and I have no idea how to add a siren.

```
{
  "config": {
    "battery": 0,
    "on": true,
    "reachable": true
  },
  "etag": "50569ae34f9865500c9b30d45c8531aa",
  "lastannounced": "2022-08-07T13:34:23Z",
  "lastseen": "2022-08-07T13:43Z",
  "manufacturername": "_TZE200_ntcy3xu1",
  "modelid": "TS0601",
  "name": "Fire 160",
  "state": {
    "fire": true,
    "lastupdated": "2022-08-07T13:37:46.330",
    "lowbattery": null,
    "test": null
  },
  "type": "ZHAFire",
  "uniqueid": "a4:c1:38:73:68:ab:81:17-01-0500"
}
```

Also the log is not very helpful
```
16:00:25:446 Websocket 172.1.2.3:45678 send message: {"attr":{"id":"160","lastannounced":null,"lastseen":"2022-08-07T14:00Z","manufacturername":"_TZE200_ntcy3xu1","modelid":"TS0601","name":"Fire 160","swversion":null,"type":"ZHAFire","uniqueid":"a4:c1:38:73:68:ab:81:17-01-0500"},"e":"changed","id":"160","r":"sensors","t":"event","uniqueid":"a4:c1:38:73:68:ab:81:17-01-0500"} (ret = 323)
16:00:25:503 TY_DATA_REPORT: seq 80, dpid: 0x0E, type: 0x04, length: 1, val: 2
16:00:25:505 TY_DATA_REPORT: seq 80, dpid: 0x0E, type: 0x04, length: 1, val: 2
16:00:25:822 TY_DATA_REPORT: seq 81, dpid: 0x04, type: 0x01, length: 1, val: 1
16:00:25:823 TY_DATA_REPORT: seq 81, dpid: 0x04, type: 0x01, length: 1, val: 1
16:00:25:893 rule event /config/localtime: 16:00:24.890 -> 16:00:25.890 (1000ms)
16:00:25:950 TY_DATA_REPORT: seq 82, dpid: 0x0E, type: 0x04, length: 1, val: 2
16:00:25:952 TY_DATA_REPORT: seq 82, dpid: 0x0E, type: 0x04, length: 1, val: 2
```

Could you please give me some additional hints?

Smanar, 2022-08-07:
>16:00:25:503 TY_DATA_REPORT: seq 80, dpid: 0x0E, type: 0x04, length: 1, val: 2

This is the battery level: 0 = low , 1 = middle, 2 = full

>16:00:25:822 TY_DATA_REPORT: seq 81, dpid: 0x04, type: 0x01, length: 1, val: 1

This one is a mystery ....

The DP 8 is the bool to test alarm. 

So in the next DDF, have corrected the battery and added state/test, it' s a bool value, it permit test the smoke detection (If I have correclty understand)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_ntcy3xu1",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Smoke sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xEF00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 14, "eval": "if (Attr.val == 0) { Item.val = 0 } else if (Attr.val == 1) { Item.val = 50 } else { Item.val = 100 }" },
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = (Attr.val == 0);" },
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/test",
          "parse": {"fn": "tuya", "dpid": 8, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    }
  ]
}
```

Rett1, 2022-08-30:
Thanks a lot for following up and the very helpful additional information!

- The battery is working as described above
`"battery": 100,`

- fire shows now `"fire": false,` as expected

- for test I'm not sure what is the expected result.
It's still  `"test": null`

```23:04:34:727 TY_DATA_REPORT: seq 21, dpid: 0x04, type: 0x01, length: 1, val: 0
23:04:34:730 TY_DATA_REPORT: seq 21, dpid: 0x04, type: 0x01, length: 1, val: 0
23:04:34:731 TY_DATA_REPORT: seq 21, dpid: 0x04, type: 0x01, length: 1, val: 0
23:04:34:892 a4:c1:38:73:68:ab:81:17-01-ef00/config/battery expression: if (Attr.val == 0) { Item.val = 20 } else if (Attr.val == 1) { Item.val = 50 } else { Item.val =100 } --> 100
23:04:34:893 TY_DATA_REPORT: seq 22, dpid: 0x0E, type: 0x04, length: 1, val: 2
23:04:34:894 TY_DATA_REPORT: seq 22, dpid: 0x0E, type: 0x04, length: 1, val: 2
23:04:34:895 TY_DATA_REPORT: seq 22, dpid: 0x0E, type: 0x04, length: 1, val: 2
23:04:35:167 TY_DATA_REPORT: seq 23, dpid: 0x04, type: 0x01, length: 1, val: 1
23:04:35:168 TY_DATA_REPORT: seq 23, dpid: 0x04, type: 0x01, length: 1, val: 1
23:04:35:170 TY_DATA_REPORT: seq 23, dpid: 0x04, type: 0x01, length: 1, val: 1
23:04:35:319 a4:c1:38:73:68:ab:81:17-01-ef00/config/battery expression: if (Attr.val == 0) { Item.val = 20 } else if (Attr.val == 1) { Item.val = 50 } else { Item.val =100 } --> 100
23:04:35:320 TY_DATA_REPORT: seq 24, dpid: 0x0E, type: 0x04, length: 1, val: 2
23:04:35:320 TY_DATA_REPORT: seq 24, dpid: 0x0E, type: 0x04, length: 1, val: 2
23:04:35:321 TY_DATA_REPORT: seq 24, dpid: 0x0E, type: 0x04, length: 1, val: 2

Smanar, 2022-08-31:
>23:04:34:730 TY_DATA_REPORT: seq 21, dpid: 0x04, type: 0x01, length: 1, val: 0

I have found this one, it s the tamper information (if you try to open it)
For the test feature, perhaps your device don't have it, you have not a button to test the working mode ? (without using gaz or smoke) ?

So can add the tamper

```
        {
          "name": "state/tampered",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = (Attr.val != 0);" },
          "read": {"fn": "none"}
        },
```

lons81, 2022-09-18:
Hello,

I have followed this Guide and imported the DDF into my VNC.
The smoke sensor is visible in VNC but it doesn't show up in Phosonz...
So what can I do?

Thanks, Lons

Smanar, 2022-09-18:
Hello , you have the same device ?

```
  "manufacturername": "_TZE200_ntcy3xu1",
  "modelid": "TS0601",
```

It's visible on deconz, on "node information" or on attribute on basic cluster"

Have you included the device before or after adding the DDF ?

lons81, 2022-09-18:
Yes same device, i see it in node infos & basic cluster...
I added one over VNC Deconz and edit the DFF from this device... doesn't appere in Phoscon, so i delete the device and copy the "ts0601.json" to
"/home/pi/.local/share/dresden-elektronik/deCONZ/devices" and search for it in Deconz VNC.
In Deconz i see 5 Dire Devices but not in Phoscon...

[image]
[image]

Smanar, 2022-09-18:
>In Deconz i see 5 Dire Devices but not in Phoscon...

Perhaps phoscon have problem, but your device have name "Fire 72" instead of  0xXXXX, it mean the API have reconized it, can you take a look direclty in the API
In phoscon / help / API Information / sensors

lons81, 2022-09-18:
API looks like this:
[image]

Phoscon Sensor:
[image]

Smanar, 2022-09-19:
? all seem fine, perhpas Phoscon don't support fire detector ...
You don't use other third app ?

lons81, 2022-09-20:
Third app is only Hue Essential on my Android Phone.
I have checked the App, the name "Fire 72" comes from the Android App...

Smanar, 2022-09-27:
You are sure Hue essential support fire dectector ?

Mimiix, 2022-12-16:
Re-opened as we had 2 duplicates getting opened. @Smanar  What do we need to get this show on the road :)?

Smanar, 2022-12-16:
He :( like usualy, a better support for Tuya cluster with DDF.
I can have return state for this device, but not possible to use the siren.

Smanar, 2022-12-17:
I have a doubt, this device haven't siren ?
So the DDF is finished ?

btk999, 2022-12-18:
Hi @Smanar - for me with the DDF the device works fine. I see battery state and test and real alarm is triggered in my 3rd party app (homebridge with homebridge-hue connected to deconz). 

Problem 1: The device is just not shown in phoscon at the moment.

The device has a siren (it beeps loud if alarm or test mode is triggered).
Problem 2: I'd like to trigger the siren/alarm of all other devices so that if an alarm occurs in room 1 i also get noticed in all other rooms. Do you think that this is possible?

Smanar, 2022-12-18:
IDK, it's for that I m asking.

I can't found something saying this device can trigger the siren.
There is some test here https://github.com/Koenkk/zigbee2mqtt/issues/12622 it seem there is a muffling but I don't see what it is, and it seem it's not on all device.

And for the moment, not possible to trigger siren using DDF.

It's for that I don't make a PR yet for this device.

ThijsCogas, 2022-12-30:
Hi,

I'm also struggling of adding the smoke detectors. In Deconz, I see this device now:
 
[image]

But when I'm trying to edit the DDF I can't change anything. I'm using Home Assistant and the Deconz add on. 
[image]

In Phoscon I saw sometimes when adding the sensor ready, but can't see it in the sensors list. Please help!

Smanar, 2022-12-30:
Faster way is putting the DDF in the HA not persistent folder /data/.local/share/dresden-elektronik/deCONZ/devices/
https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/20

Rett1, 2022-12-31:
You won't see this device in Phoscon.

By the way, mine broke after some weeks. It simply stopped working. Will have to look for another smoke sensor now.

ThijsCogas, 2023-01-03:
> 

Unfortunately the link is broken and I don't have the .local folder. I can create the folder structure. But what should be the filename? I've checked with file editor, Studio code server and SSH addons.

Smanar, 2023-01-03:
The link is working again, the server was OOT.

>I can create the folder structure

Not sure it's normal, the folder need to already exist.

>But what should be the filename?

That you want, just need to be a .json file and in the "devices" folder.