dresden-elektronik/deconz-rest-plugin #6241: Device name

km8742, 2022-08-05:
## Device
- NOTE: I'm running deCONZ headless so I can't give GUI values. Is there any way to read those datasets from the zll.db or similar?
- Product name: R7048 Temperature & Humidity Sensor
- Manufacturer: Woox
- Model identifier: Unknown
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots
How can I get this data in headless installation?

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Sorry for the little amount of data, It would be great if there was a possibility to supply this data with the headless version of deCONZ.

Thank you for your work!

Mimiix, 2022-08-06:
We really need the screenshots. Zll.db does not provide.

km8742, 2022-08-06:
I'm currently running ConBee II on a headless raspberry pi (w/o graphical interface installed).
Can i temporarily create a Ubuntu Box, install deCONZ and the ConBee II to this machine to read out the missing data or will this mess with the configuration of my raspberry pi in any way?

Mimiix, 2022-08-06:
You need to restore a back up in that case and afterwards restore it on the pi.

km8742, 2022-08-06:
Simply the 'Backup options' in the ConBee II Homepage?

Mimiix, 2022-08-06:
Yes

km8742, 2022-08-06:
Updated the main post - is this data enough?

Smanar, 2022-08-06:
Missed ^^.

There is a "read" button, use it to read all attributes to fill the table
As this device is a sleeper you probably need to wake it in same time (with pressing shortly the button generaly)

km8742, 2022-08-06:
How do I see if it worked?? :))
There is a message 'not in network' above - is that a problem?

km8742, 2022-08-06:
Deleted the node for it to be readded under the GUI setup... currently nothing happened. I'll wait until it is readded so I can try to read the data again.

Smanar, 2022-08-06:
When it work the attribute is populated with values (we need model id and manufacture name to reconize the device)
But you just need to press the "read button" and make short press on the device to wake it up.

Mimiix, 2022-08-06:
It need to be in network.

km8742, 2022-08-07:
Had some trouble connecting to it trough the test network. I've updated the images on top - to you need any more data?

Smanar, 2022-08-07:
Hello, you miss the node picture with all clusters, but I presume you have only 1 endpoint the "01".
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_amqudjr0",
  "modelid": "TS0201",
  "product": "R7048 Temperature & Humidity Sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

Do you know how to install them ?
If you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
Else https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

km8742, 2022-08-07:
Created a file /usr/share/deCONZ/devices/woox/woox_r7048_ts0201.json containing your data and it seems it works. Being recognized and using DDF ZHAHumidity and ZHATemperature. I did not get any readings trough the websocket but it's listed in the API under /sensors twice (once humidity and once temperature). How do I have to calculate those values to get percent and Celsius values? :-)

Smanar, 2022-08-07:
@km8742 you have the battery level too ?

km8742, 2022-08-07:
It doesn't show in the phoscon APP overview (can't click it there - not even rename it).
In the deCONZ App I see a red battery icon (strange cause the product is new).

Screenshots:
[image]
[image]

Smanar, 2022-08-07:
And nothing if you select it in phoscon ?

km8742, 2022-08-07:
You mean the phoscon IP http://localhost:8080/pwa - right?
Nothing happens if I click it.

Smanar, 2022-08-07:
Just clic on "humidity sensor" to have something like that https://community.jeedom.com/uploads/default/original/2X/4/407693233e72cb88c26b8720476d15257c6a1aa7.png

km8742, 2022-08-07:
Nothing happens. In the Firefox Console I get the following: 
Uncaught TypeError: sg is null

Smanar, 2022-08-07:
Just clic on that

[image]

to have the same capture than on my previous link

km8742, 2022-08-07:
The other sensors work as intended (click to get details) so I don't think its a browser error.

Smanar, 2022-08-07:
Ha ?

I will give this capture on the phoscon forum, I think it's because of the battery value is prbably "null", but first time I see that.

You can see the complete json for this device in phoscon / help / API information / sensors / select sensor / ...
You will found something like that

```
{
    "config": {
        "battery": 100,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "844ae80c0d4f2610d8503cec18fc1857",
    "lastannounced": null,
    "lastseen": "2022-08-07T17:52Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.weather",
    "name": "Multi Sensor",
    "state": {
        "lastupdated": "2022-08-07T17:52:53.879",
        "temperature": 2454
    },
    "swversion": "20161129",
    "type": "ZHATemperature",
    "uniqueid": "00:15:8d:00:20:3f:88:56-01-0402"
}
```

Except on this one the battery is 100

km8742, 2022-08-07:
Strangely it shows as two sensors - #18 does not show up in the main listing.

Output:
```
"17": {
        "config": {
            "battery": 0,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "4edd3ce8464c89d3bc8d024e42b49894",
        "lastannounced": null,
        "lastseen": "2022-08-07T18:33Z",
        "manufacturername": "_TZ3000_amqudjr0",
        "modelid": "TS0201",
        "name": "Humidity 17",
        "state": {
            "humidity": 6220,
            "lastupdated": "2022-08-07T18:33:14.364"
        },
        "type": "ZHAHumidity",
        "uniqueid": "a4:c1:38:34:f5:45:81:f8-01-0405"
    },
    "18": {
        "config": {
            "battery": 0,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "etag": "fe6870690d8190e312bc2acad9baa150",
        "lastannounced": null,
        "lastseen": "2022-08-07T18:33Z",
        "manufacturername": "_TZ3000_amqudjr0",
        "modelid": "TS0201",
        "name": "Temperature 18",
        "state": {
            "lastupdated": "2022-08-07T18:33:14.240",
            "temperature": 2258
        },
        "type": "ZHATemperature",
        "uniqueid": "a4:c1:38:34:f5:45:81:f8-01-0402"
    }
}
```

I also have three Smoke alarms from Woox which do not show up in the listing under Devices > Sensors.

Smanar, 2022-08-08:
It's normal, when you go in "API information" you have the full API, phoscon is just a third app that use this API.
You can have device in the API but not visible in phoscon.
Deconz create 1 entry by cluster, so as you have 1 cluster for temperature and one for the humidity (used cluster), you have 2 entry in the API, phoscon just mix them.

>three Smoke alarms from Woox which do not show up in the listing under Devices > Sensors.

You mean not visible in phoscon, but visible in "API Information" ?

BTW what is your deconz version ? (To know your phoscon version)
Because the json seem fine I don't explain the error message, perhaps if you clear browser cache ? (or try with another browser)

BTW the battery not working, can you try this DDF instead (can need some hour for first battery report)
On this DDF I m using the attribute 0x0020 instead of the attribute 0x0021 for battery bind
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_amqudjr0",
  "modelid": "TS0201",
  "product": "R7048 Temperature & Humidity Sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 300,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]

```}

km8742, 2022-08-08:
deCONZ is a fresh install: v.2.17.1
ConBee II Firmware: 26720700
I currently can only use Firefox - but I've tried two different PCs, both can't access the Humidity 17 Sensor (no action on click)
Smoke Alarms: Yes those are showing up in the deCONZ app and are populated over the WebSocket - but do not show up in the Phoscon App (http://IP:8080/pwa) in the "Sensors" tab - they show up in the API.
DEF: I've updated the file and did a hot-reload, currently waiting for a readout so I can see if anything changes.

Smanar, 2022-08-09:
>also did a read while pressing the sensor button

This need to update the value too, so mean the problem is not from the bind (it update the value in the GUI and deconz update the api in same time)

I have created a report on the phoscon forum for your issue https://forum.phoscon.de/t/browser-error-when-selecting-device/2310 Perhaps someone will ask more information later, I can't reproduce something on my side.

New DDF to test, I m using again the attribute 0x0021 but I use more stuff in config/battery

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"_TZ3000_amqudjr0",
   "modelid":"TS0201",
   "product":"R7048 Temperature & Humidity Sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_HUMIDITY_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0405"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/offset",
               "default":0
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/humidity"
            },
            {
               "name":"config/battery",
               "awake":true,
               "refresh.interval":7300,
               "parse":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "eval":"Item.val = Attr.val / 2",
                  "fn":"zcl"
               },
               "read":{
                  "ep":1,
                  "cl":"0x0001",
                  "at":"0x0021"
               }
            },
            {
               "name":"state/lastupdated"
            }
         ]
      },
      {
         "type":"$TYPE_TEMPERATURE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0402"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/offset",
               "default":0
            },
            {
               "name":"config/battery",
               "awake":true,
               "refresh.interval":7300,
               "parse":{
                  "at":"0x0021",
                  "cl":"0x0001",
                  "ep":1,
                  "eval":"Item.val = Attr.val / 2",
                  "fn":"zcl"
               },
               "read":{
                  "ep":1,
                  "cl":"0x0001",
                  "at":"0x0021"
               }
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/temperature"
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":300,
               "max":43200,
               "change":"0x00000001"
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0402",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x29",
               "min":60,
               "max":300,
               "change":"0x00000064"
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0405",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x21",
               "min":60,
               "max":300,
               "change":"0x00000064"
            }
         ]
      }
   ]
}
```

This one is a little more "aggressive", we will probably need to disable somes thing later.

km8742, 2022-08-09:
DDF works now, shows 91 percent in API - Thanks a lot!
I'll keep an eye on the forum post - would be great if I could help

Smanar, 2022-08-10:
Nice I will post the DDF as it.

But if you have time to try the DDF withoutt hoses lines

`"refresh.interval":7300,`

It's a poll that is enabled when the reporting fail, realy aggressive for battery, if we can't better to disable it.

km8742, 2022-08-11:
Readings over the websocket work like a charm with the removed refresh.interval. I'm getting a reading all 30 Minutes.

km8742, 2022-08-11:
Thanks a lot for the great support!

Smanar, 2022-08-12:
Thx to you, I remove it too in the PR.