dresden-elektronik/deconz-rest-plugin #6248: Tuya TS0601 device not adding properly to deConz & not showing in Phoscon at all.

Ltek, 2022-08-08:
## Describe the bug
 Sensor not pairing fully

## Steps to reproduce the behavior
pair device

## Expected behavior
proper identification in deConz and shows up in Phoscon

## Screenshots
attached
[image]

## Environment
 - Host system: PC
 - Running method: Windows
 - Firmware version: 26400500
 - deCONZ version: 2.18.0
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
attached
[Tuya_deConzDebug_TS0601.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9277862/Tuya_deConzDebug_TS0601.txt)

## Additional context

Mimiix, 2022-08-08:
Is this device supported?

Mimiix, 2022-08-08:
Hi there,

It seems the device is not fully supported, working yet as your other device request is still open:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6063

Opening this issue as a bug, is not needed at this point as the device is not supported yet.

Closing.

Kind regards,

closed 2022-08-08 by Mimiix

Ltek, 2022-08-08:
I opened the bug because the device is recognized but doesn't work.

 I spent a lot of time working with manup on this issue and I'd like to
have it resolved. Its been ignored  over a month since any he remoted into
my system to work on it

Mimiix, 2022-08-08:
I pinged him to check the original issue :)