dresden-elektronik/deconz-rest-plugin #6251: Lumi Plug maeu01 - Wrong metering

basty2, 2022-08-08:
## Describe the bug
Since the update to 18.0, my aqara plug delivers the wrong active power. Instead of 135W it delivers 1350W.
 
## Steps to reproduce the behavior
not applicable
 
## Expected behavior
the value should be 1/10 of that
 
## Screenshots
not necessary
 
## Environment
model: lumi.plug.maeu01

## deCONZ Logs
not necessary
 
## Additional context
up to v.2.17 it worked fine.

Mimiix, 2022-08-08:
@basty2 Can you please fill in the report as intended?

basty2, 2022-08-09:
@Mimiix Is it so ok or do you need more information?

Mimiix, 2022-08-09:
Is good :)

Thanks :)

Smanar, 2022-08-11:
Hello, can you edit the file `devices\xiaomi\xiaomi_sp-euc01_smart_plug.json`

At this part

```
          "name": "state/power",
          "refresh.interval": 10,
          "read": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 21,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 21,
            "eval": "Item.val = Math.round(Attr.val);"
          }
```

try 

` "eval": "Item.val = Math.round(Attr.val / 10);"`

basty2, 2022-08-11:
@Smanar  I changed the data, restarted deconz, but it is not working. Values are 10x too high.

Smanar, 2022-08-12:
The bad value is state/power ?
Can you check if you have something in logs with the flag "DDF" ?

Can you check in the file if the modification is "persistent" ?

basty2, 2022-08-15:
Hi @Smanar,

the bad value is state/power. I have activated the debug mode, there are some entries at DDF, but none from the affected device. The modification should be persistent, if i open the file again, the new entry still exists.

Smanar, 2022-08-15:
>there are some entries at DDF, but none from the affected device

Ha ? The DDF status is "Gold" so I don't see why the DDF is not used for this device ...

Edit :
Ok I have found a "check line" in the DDF

```
  "comment": "DDF for device firmwares at least above 0.0.0_0022, paired with xBee I fw 0x26400500 / xBee II fw 0x266f0700 and above",
  "matchexpr": "R.endpoints.includes(0x15) && R.endpoints.includes(0x1F)",
```

I think your device don't have Endpoint 0x15 and 0x1F ?

You can remove the line "matchexpr" to remove the check, but idk what will happen an other plug with different firmware (but same model)

basty2, 2022-08-15:
I used "//" before the line matchexpr. But it doesn't work.
the ID of the active power is 0x050B as you can see in the screenshot.

basty2, 2022-08-15:
[image]
maybe this helps. 
[image]

Smanar, 2022-08-15:
You can direclty remove the line to test.
From the log I think it explain why the DDF is not used for you.

basty2, 2022-08-15:
ok, now it is shown at ddf. But the value is still to high. 
[image]

Smanar, 2022-08-16:
Nice.
Change the line

`"eval": "Item.val = Math.round(Attr.val / 10);"`

And you will see difference on the same debug line.
If it work can you show all the device endpint ? It's the number on the node title (can have more than one part)

To make a better "matchexpr"

basty2, 2022-08-17:
Hi @Smanar , 
it is still not working. here ist my data 
[image]

Is this the node title? 
[image]

[image]

Smanar, 2022-08-17:
Ok so you have the endpoint 0x01 and 0xF2 on yours.
If you take a look in your log

[image]

Edit:
Ok so I have look better your log, and we can see `"eval": "if (Attr.val != -32768 && Attr.val != 32768) { Item.val = Attr.val; }",` but in the DDF we have `"Item.val = Math.round(Attr.val);"` it mean the device is using another DDF

The only DDF where I can see `"eval": "if (Attr.val != -32768 && Attr.val != 32768) { Item.val = Attr.val; }",` is a Sinope device.

You are sure there isn't a mistake somewhere ? with comparing the mac adress.

basty2, 2022-08-17:
@Smanar I renamed the devices folder and copied a completly new downloaded device from the github. But also this doesnt work. What kind of mistake could it be?
I renamed the th1124zb.json to th1124zb.json.old but also this doesn't work. I dont have a sinope device. How could I campare the Mac adress? the mac adress ends with 46-58.

Smanar, 2022-08-17:
If I m right with the flag "DDF" on debug you will see files used by devices, but as this one have a "matchexpr" field idk how it will react.

You can too on the GUI select the device (the node title) and choose "edit DDF" you will see the file used by deconz for the device.

Can be too a generic DDF (it's the defaut value), but how you have created it .... You haven't tried to edit the DDF using the editor ? https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/generic/items/state_power_item.json

basty2, 2022-08-17:
Hi @Smanar  . I tried edit ddf. it should be correct. 
[image]

Here are all entries of ddf debug: 
`17:52:18:574 04:cf:8c:df:3c:76:46:58-01/attr/manufacturername expression: Item.val = Attr.val --> LUMI
17:52:18:575 04:cf:8c:df:3c:76:46:58-01-0b04/attr/manufacturername expression: Item.val = Attr.val --> LUMI
17:52:18:576 04:cf:8c:df:3c:76:46:58-15-000c/attr/manufacturername expression: Item.val = Attr.val --> LUMI
17:52:18:576 04:cf:8c:df:3c:76:46:58-01-0702/attr/manufacturername expression: Item.val = Attr.val --> LUMI
17:52:18:577 04:cf:8c:df:3c:76:46:58-1f-000c/attr/manufacturername expression: Item.val = Attr.val --> LUMI
17:52:18:578 04:cf:8c:df:3c:76:46:58/attr/manufacturername expression: Item.val = Attr.val --> LUMI
17:52:18:608 04:cf:8c:df:3c:76:46:58-01/attr/modelid expression: Item.val = Attr.val --> lumi.plug.maeu01
17:52:18:608 04:cf:8c:df:3c:76:46:58-01-0b04/attr/modelid expression: Item.val = Attr.val --> lumi.plug.maeu01
17:52:18:609 04:cf:8c:df:3c:76:46:58-15-000c/attr/modelid expression: Item.val = Attr.val --> lumi.plug.maeu01
17:52:18:609 04:cf:8c:df:3c:76:46:58-01-0702/attr/modelid expression: Item.val = Attr.val --> lumi.plug.maeu01
17:52:18:609 04:cf:8c:df:3c:76:46:58-1f-000c/attr/modelid expression: Item.val = Attr.val --> lumi.plug.maeu01
17:52:18:609 04:cf:8c:df:3c:76:46:58/attr/modelid expression: Item.val = Attr.val --> lumi.plug.maeu01
17:52:21:264 04:cf:8c:df:3c:76:46:58-01/state/on expression: Item.val = Attr.val --> true
17:52:21:563 04:cf:8c:df:3c:76:46:58-01-0b04/state/power expression: if (Attr.val != -32768 && Attr.val != 32768) { Item.val = Attr.val; } --> 1467
17:52:44:402 04:cf:8c:df:3c:76:46:58-01-0702/state/consumption expression: Item.val = Attr.val --> 1448266
`

basty2, 2022-08-17:
there is also no change, if i change to (Attr.val / 100). 
Is this a problem only for me or is it a problem for several people?

Smanar, 2022-08-18:
Ok, so there is something strange.
I will ask for tips to others devs.

If you make a "hot reload" with same log enabled "DDF" (or perhaps "info" "info_l2"), you will see the file used used by deconz for this device, not possible it use the same ....

SwoopX, 2022-08-18:
Please delete any modified DDF, restart deconz and re-pair the device. The device exposes incorrect clusters when a response is not properly received during pairing.

basty2, 2022-08-18:
Hi @SwoopX. I deleted and reinstalled the complete devices folder. I tried to repair the device more than 20 times. I have now a new problem. The power value is not shown in electric metering, but In a new area. 
[image]
7 Watt should be the right Power Value at Analog Input Cluster: 
[image]

SwoopX, 2022-08-18:
The "problem" is a works as designed. Regardless, the device still doesn't look right and my gut feeling says it runs on a rather old firmware. It should look like this:

[image]

Please activate the OTA update plugin under plugins (in the menu bar), select the device and then select OTA update pane (bottom left). Then press the query button until you get some result.

SwoopX, 2022-08-18:
Firmware can be found here and must be placed in the otau folder to update
https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20211209165104_OTA_lumi.plug.maeu01_0.0.0_0041_20211206_0C22EC.ota

basty2, 2022-08-19:
@SwoopX Very strange. Because everythings worked until 2.17. But this was the solution. I got all the values. Thanks.

closed 2022-08-19 by basty2

softwerker64, 2022-10-25:
Looks like I'm vacing the same problem :-(
I updated the "lumi.plug.maeu01" with the [new firmware referenced above](https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20211209165104_OTA_lumi.plug.maeu01_0.0.0_0041_20211206_0C22EC.ota) and the device reports "09-10-2019" as SW-Version in Hue-Essentials.
However, the value "state.power" still seems to be ten times too high. In Hue-Essentials and via REST.

Any ideas? Thanks in advance!

softwerker64, 2022-12-03:
Installed newest update of deconz on my Raspberry with no luck, REST still reports these values (where power is ten times too high):
current: 0
power: 10
voltage: 0

"apt" reports this package: "deconz/unknown,now 2.19.03-raspbian-buster-stable armhf"

grimmaldus, 2024-09-20:
I have been experiencing the same issue for several years and have been stuck using version 2.17.1 since then. Could someone please explain what has changed in the newer versions? Is there any specific reason for this issue? Are there any known solutions or workarounds available?

Any help would be greatly appreciated. Thank you!

Smanar, 2024-09-20:
Hello, are you using the new firmware ? (for the device, not for the Gateway)

grimmaldus, 2024-09-20:
I’m using the file OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946.ota in deCONZ, but it shows as 0x00000020. I couldn’t find a newer version. When I want to use the latest version, 2.28.1 deCONZ, the only way to get the correct values is by changing the DDF.

Smanar, 2024-09-21:
Have checked the OTA file and yes this version is the 0x20.
Do you have access to the GUI to compare how look your device in deconz ?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-smart-plugs-not-working-correctly

Perhaps need to delete it and re-include it (as the old one is managed by legacy code, and the new one by DDF, perhaps a conflict)

grimmaldus, 2024-09-24:
> Have checked the OTA file and yes this version is the 0x20. Do you have access to the GUI to compare how look your device in deconz ?

[image] 

> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-smart-plugs-not-working-correctly

Oh nice I'm gonna try this file I couldn't get to it before. And let you know. 

> 
> Perhaps need to delete it and re-include it.

Yea I did it before after I updated.

Smanar, 2024-09-24:
Nope, you need to compare the clusters, the first image on the wiki page.
And from that I  have checked you have the last one, the 0x20.

You have "DDF" written on the node corner ?

grimmaldus, 2024-09-24:
> Nope, you need to compare the clusters, the first image on the wiki page. And from that I have checked you have the last one, the 0x20.
Thanks for your response. I’ve checked the clusters, and unfortunately, they don’t match. Regarding the DDF, I’m not entirely sure where I should be looking for it in the GUI since I couldn’t find any reference to it in the wiki. If by any chance you are referring to the "Edit DDF" option in the right-click node menu, then yes, I do have that, and that’s actually how I managed to edit the DDF and fix the initial issue.
> You have "DDF" written on the node corner ?
Additionally, I noticed that the file that led me to the 0x00000020 cluster has a different hash compared to the one on the wiki page. I wasn’t able to locate it earlier due to the broken link, but I will now try updating the firmware for the lumi.plug.maeu01 following the wiki page's firmware guide.

Smanar, 2024-09-25:
[image]

On this picture you can see Endpoint 01/15/1F/F2 and on the Endpoint 0x15 and 0x1F, you can see the cluster 0x000C
You need them, for the DDF can work.

`  "matchexpr": "(R.endpoints.indexOf(0x15) !== -1) && (R.endpoints.indexOf(0x1F) !== -1);",`

Else can you show how look your device ?

grimmaldus, 2024-09-25:
> Else can you show how look your device ?

[image]

Smanar, 2024-09-26:
So yes I can confirm, the DDF will never work, you have an old version.
But how you can have a version = 0x20 ? Or there is a OTA version above, but I don't see it on the github.

`DDF for device firmwares at least above 0.0.0_0022, paired with xBee I fw 0x26400500 / xBee II fw 0x266f0700 and above`

Can you check the swbuild on the cluster 0x0000 'basic" ?

grimmaldus, 2024-09-27:
> Can you check the swbuild on the cluster 0x0000 'basic" ?

[image]

> And BTW why you have "DDF" writted on the node corner ? Can you make "edit DDF" to see wich one DDF is used by the editor ? (It's written on the titlebar")

I'm not very familiar with the deCONZ GUI, to be honest, so I'm not sure why it shows up there, but I have edited the DDF for these devices.

[image]

bluemoehre, 2024-09-28:
I just upgraded some plugs from v41 to the new v45 of Aqara / Xiaomi and ran again into this annoying problem. I don't remember how I solved that in the past =/

Would be absolutely happy if we can fix this once for the future.

Smanar, 2024-09-28:
>I'm not very familiar with the deCONZ GUI, to be honest, so I'm not sure why it shows up there, but I have edited the DDF for these devices.

Yes, I m seing, idk how is the rest of the DDF, but your device can't work at all like the existing DDF (he don't use same cluster), so if you remove some line to force deconz use the DDF, it will not work.
Need to create a full DDF, but ATM, users just update their device with more recent firmware to use the last DDF, Idk why on your device you have a version 0x20 but an old firmware running.

>Deconz always should respect the multiplier properties (eg. 0x0605 - AC Power Divisor) to render the value.

If it can be respected by brand, and for information this device use now the cluster 0x000C, so no multiplier/diviser.

bluemoehre, 2024-09-28:
> If it can be respected by brand, and for information this device use now the cluster 0x000C, so no multiplier/diviser.

Based on what I see, that's not true. I already deleted and reconnected the plug from the network including a full device reset. Still I can confirm my product `lumi.plug.maeu01` @ firmware `v45` still uses divisors / multipliers. 

Do I miss something?

Smanar, 2024-09-28:
Last version is this one

[image]

bluemoehre, 2024-09-28:
> Last version is this one

this looks like < v43 to me.

v45:
[image]

Smanar, 2024-09-28:
To compare with others, can you share the device swbuild (attribute 0x4000 on cluster 0x0000) and the firmware version from ota cluster if you can ?

bluemoehre, 2024-09-28:
[image]

(Don't see any 0x4000 in 0x0000)

bluemoehre, 2024-09-28:
Tried to go down the rabbit hole and looks like the DDF here:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/xiaomi_sp-euc01_smart_plug.json

is causing issues with v45 and needs to be modified to match < v45 only. 

A new DDF is required for v45(+), also respecting Divisors, Multipliers, and fixing SW Version field.

Smanar, 2024-09-28:
Ok, so idk what happen and what is the last version we need to use but can try this DDF

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "$MF_LUMI",
    "modelid": "lumi.plug.mmeu01",
    "vendor": "Xiaomi",
    "product": "Smart plug",
    "sleeper": false,
    "status": "Gold",
    "matchexpr": "(R.endpoints.indexOf(0x15) == -1) && (R.endpoints.indexOf(0x16) == -1);",
    "subdevices": [
        {
            "type": "$TYPE_SMART_PLUG",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "at": "0x00f7",
                        "ep": 1,
                        "fn": "xiaomi:special",
                        "idx": "0x08",
                        "script": "xiaomi_swversion.js"
                    },
                    "read": {
                        "fn": "none"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/alert",
                    "description": "The currently active alert effect.",
                    "default": "none"
                },
                {
                    "name": "state/on"
                },
                {
                    "name": "state/reachable"
                }
            ]
[... 171 lines cut ...]
                    "max": 300,
                    "change": "0x00000001"
                },
                {
                    "at": "0x0505",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000064"
                },
                {
                    "at": "0x0508",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000064"
                }
            ]
        }
    ]
}
```

It will work with old cluster, all is by defaut because I realy don't have idea of what is working or not on this version.
So vlaues will probably need to be adapted.

bluemoehre, 2024-09-28:
For some reason I cannot get the DDF working properly. Already build my own one, and can see in deCONZ it's there and assigned. But in API output is no change at all. Already tried to set static values to power attributes etc. But again, no effect. 

So, first I have to find out what's wrong with the DDF / JSON files.

Smanar, 2024-09-28:
Perhaps you miss the bind/report ?

bluemoehre, 2024-09-28:
What exactly do you mean with "bind/report"? I change the JSON and then restart the service - which I assume should load the newest version. 
I can see changes in the DDF Editor but not in the API response.

Smanar, 2024-09-29:
The last part on DDF

```
                {
                    "at": "0x0508",
                    "dt": "0x21",
                    "min": 1,
                    "max": 300,
                    "change": "0x00000064"
                }
```

Or you mean it's not an update issue, but they are not present at all in the API ?

Deconz create new entries only if the gateway is in permit join, so can need to delete and re-include the device if there is too much difference beetween the 2 versions.

bluemoehre, 2024-09-29:
Just for safety I already re-added the device after upgrading the firmware. So I guess these prominent changes shown in my screenshots are by firmware. 
Still the issue is, that changing the DDF expressions to calculate values correctly are not reflected in API output, while at the same time I can see DDF changes in deCONZ UI / editor. I have no clue what's wrong this time. Neither restarting the service or hot-reloading the DDF does help.

Smanar, 2024-09-29:
Have you tried the logs ? In deconz/help/debug view ? with flag "DDF", and if needed "info" + "info_l2"

bluemoehre, 2024-09-29:
Yes, the correct file is assigned, as far as I see it in the logs. 

To make the problem more interesting: 
I upgraded a second Plug to v45 and can clearly see some bug in Phoscon / API, because the UI still shows old firmware version for that one, also some Ampere + Voltage values, which are not available in deCONZ itself. Tried to clear the cache of the browser, but no effect as well. Using Postman these values including wrong version number are still provided by the API. 

Without being able to rely on the API output, it will be hard to fix a value, since I do not know atm where data got stucked in the whole chain.

Smanar, 2024-09-30:
> in Phoscon / API, because the UI still shows old firmware version for that one

Yes deconz don't ask for swbuildID every time, so it can have memorised an old one.

>EDIT2: seems like deCONZ assigned different DDFs to each of the plugs. My one in /opt/deconz/devices to the first, and another one from /usr/share/deconz/devices/xiaomi to the second.

And it's same plugs ? Same clusters used ? there is no differences ? And yes there is a problem in deconz, you need to have only 1 DDF for a couple manufacturename/Modelid else deconz can pick a random one, even with a flag status=draft.

bluemoehre, 2024-10-01:
I removed my custom file, then used grep to search everywhere in deCONZ for a conflict, which was not the case. Re-adding all Plugs, same HW, same SW... but now it's like this:

[image]

I don't see how this can not be an internal bug in deCONZ. Debug log says that single plug does NOT match any DDF, while the other two do match.

Smanar, 2024-10-01:
It mean they are not same device.
One have endpoints 0x15 / 0x1F and not the other (false = not present DDF not used).
If they have them, can be a timing issue, deconz check for endpoint before having asked for them.

bluemoehre, 2024-10-01:
@Smanar in the meanwhile I found this is actually some (cahcing) bug in deCONZ.

Smanar, 2024-10-02:
Ha yes, and this issue is realy not good for yours problems ....