dresden-elektronik/deconz-rest-plugin #6256: Motion rules not triggered after update from 2.15.02

Helox, 2022-08-10:
## Describe the bug
Tried updating from 2.15.02 to 2.16.00, motion rules were not triggered so I decided to revert to 2.15.02 since I was running deconz in Docker so the wife would not be mad ;)
Today I decided to try updating again since there another new version, however the motion rules were still not working and reverting to 2.15.02 did not resolve the problem like the last time.

## Steps to reproduce the behavior
Setup using Docker image deconzcommunity/deconz:2.15.02
Adding motion sensor
Adding lamp to group
Adding motion rules to motion sensor using Hue Essentials to trigger group
Update docker image to deconzcommunity/deconz:2.18.00
Update firmware Conbee II

## Expected behavior
Motion rules being triggered

## Screenshots
Not applicable, logs included

## Environment
 - Docker image deconzcommunity/deconz:2.18.00
 - Host system: QNAP nas Linux 5.10.60 x86_64
 - Running method: Marthoc Docker container
 - Firmware version: 26720700
 - deCONZ version: 2.18.00
 - Device: Combee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
```
deconz_1  | [deconzcommunity/deconz] Starting deCONZ...
deconz_1  | [deconzcommunity/deconz] Current deCONZ version: 2.18.00
deconz_1  | [deconzcommunity/deconz] Web UI port: 8083
deconz_1  | [deconzcommunity/deconz] Websockets port: 8084
deconz_1  | [deconzcommunity/deconz] Using options --auto-connect=1 --appdata=/opt/deCONZ --dbg-info=1 --dbg-aps=0 --dbg-zcl=0 --dbg-ddf=0 --dbg-dev=0 --dbg-zdp=0 --dbg-ota=0 --dbg-error=0 --dbg-http=0 --http-port=8083 --ws-port=8084
deconz_1  | [deconzcommunity/deconz] Modifying user and group ID
deconz_1  | [deconzcommunity/deconz] Checking device group ID
deconz_1  | [deconzcommunity/deconz] VNC port: 5900
deconz_1  | tigervncserver: No VNC server running for this user!
deconz_1  | Cleaning stale pidfile '/home/deconz/.vnc/filestore:5900.pid'!
deconz_1  | 
deconz_1  | TigerVNC server sessions:
deconz_1  | 
deconz_1  | X DISPLAY #	RFB PORT #	PROCESS ID	SERVER
deconz_1  | /usr/bin/xauth:  file /home/deconz/.Xauthority does not exist
deconz_1  | 
deconz_1  | New Xtigervnc server 'filestore:0 (deconz)' on port 5900 for display :0.
deconz_1  | Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /home/deconz/.vnc/passwd filestore:0 to connect to the VNC server.
deconz_1  | 
deconz_1  | WebSocket server settings:
deconz_1  |   - Listen on :6080
deconz_1  |   - Web server. Web root: /usr/share/novnc
deconz_1  |   - SSL/TLS support
deconz_1  |   - Backgrounding (daemon)
deconz_1  | [deconzcommunity/deconz] NOVNC port: 6080
deconz_1  | QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-deconz'
deconz_1  | libpng warning: iCCP: known incorrect sRGB profile
deconz_1  | libpng warning: iCCP: known incorrect sRGB profile
deconz_1  | 20:12:42:832 HTTP Server listen on address 0.0.0.0, port: 8083, root: /usr/share/deCONZ/webapp/
deconz_1  | 20:12:43:587 /opt/deCONZ/config.ini exists and is writeable
deconz_1  | 20:12:43:587 /opt/deCONZ/zll.db exists and is writeable
deconz_1  | 20:12:43:894 COM: /dev/ttyACM0 / serialno: , ConBee II
deconz_1  | 20:12:43:894 ZCLDB init file /opt/deCONZ/zcldb.txt
deconz_1  | 20:12:44:023 DDF enabled for Gold status
deconz_1  | 20:12:44:147 gw run mode: docker
deconz_1  | 20:12:44:147 GW sd-card image version file does not exist: /opt/deCONZ/gw-version
deconz_1  | 20:12:44:148 DB sqlite version 3.34.1
deconz_1  | 20:12:44:149 DB PRAGMA page_count: 151
deconz_1  | 20:12:44:149 DB PRAGMA page_size: 4096
deconz_1  | 20:12:44:149 DB PRAGMA freelist_count: 4
[... 536 lines cut ...]
deconz_1  | 20:14:00:395 SC state change failed: 00:17:88:01:02:03:03:89-02-0402
deconz_1  | 20:14:00:666 rule event /groups/1/state/any_on: 0 -> 1
deconz_1  | 20:14:01:079 rule event /groups/9/state/any_on: 0 -> 9
deconz_1  | 20:14:01:079 trigger rule 51 - Motion sensor 13 arm
deconz_1  | 20:14:01:696 No presence sensor found for 0x0017880102030389, endpoint: 0x02
deconz_1  | 20:14:01:698 SC config/delay: synced
deconz_1  | 20:14:01:699 SC --> StateFinished
deconz_1  | 20:14:01:699 SC state change finished: 00:17:88:01:02:03:03:89-02-0406
deconz_1  | 20:14:01:778 SC config/ledindication: synced
deconz_1  | 20:14:01:778 SC --> StateFinished
deconz_1  | 20:14:01:778 SC state change finished: 00:17:88:01:02:03:03:89-02-0406
deconz_1  | 20:14:01:874 No presence sensor found for 0x0017880102030389, endpoint: 0x02
deconz_1  | 20:14:01:875 SC config/sensitivity: synced
deconz_1  | 20:14:01:875 SC --> StateFinished
deconz_1  | 20:14:01:876 SC state change finished: 00:17:88:01:02:03:03:89-02-0406
deconz_1  | 20:14:02:089 ZCL attribute report 0x001788010201E6FB for cluster: 0x0400, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
deconz_1  | 20:14:03:912 No presence sensor found for 0x0017880102030389, endpoint: 0x02

```
## Additional context
Philips Hue motion sensors
Philips Hue lamp as zigbee routers
Controlling groups in Hue Essentials or Homekit is still working
Also Homekit detects the motion being triggered 
Homekit integration using oznu/homebridge docker image

Mimiix, 2022-08-10:
Hi,

If I'm correct, this is a duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6169

Do you agree? If so, I propose to close this issue. 

Kind regards, 
Dennis

Helox, 2022-08-10:
Yes seems like the same issue, do note that it was working in 2.15.02, after my upgrade to 2.17.00 I tried reverting to each minor version until everything was working again. However I cannot confirm if this error is also caused by the "dx" condition, I do also have the same issue as mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6169#issuecomment-1193196363_

Mimiix, 2022-08-10:
In that case, I'll close this issue. Perhaps you can add your findings to that issue?

SwoopX, 2022-08-10:
This is indeed a duplicate. Interestingly, dx rules do not seem to be impacted, as described in the referenced issue.

Helox, 2022-08-15:
Duplicate of #6169, comments added to previous issue

closed 2022-08-15 by Helox