dresden-elektronik/deconz-rest-plugin #6258: Wall switch - zigfred uno (ZFU-1D-CH)

WhistleMaster, 2022-08-11:
## Device

- Product name: zigfred uno
- Manufacturer: Siglis
- Model identifier: zigfred uno
- Device type : Switch

## Screenshots

[image]
[image]

## 05

### Basic
[image]

### Identify
[image]

### Alarms
N/A

### Device Temperature
N/A

### Groups
[image]

### Scenes
[image]

### On/Off
[image]
[image]

### Level Control
[image]

### Color Control
[image]
[image]
[image]
[image]
[image]
[image]

### Simple Metering
N/A

### Diagnostics
N/A

### Other clusters that are not mentioned above
0345 Unknown
[image]

## 06

### Basic
[image]

### Identify
[image]

### Alarms
N/A

### Device Temperature
N/A

### Groups
[image]

### Scenes
[image]

### On/Off
[image]
[image]

### Level Control
N/A

### Color Control
N/A

### Simple Metering
N/A

### Diagnostics
N/A

### Other clusters that are not mentioned above
N/A

## 07

### Basic
[image]

### Identify
[image]

### Alarms
N/A

### Device Temperature
N/A

### Groups
[image]

### Scenes
[image]

### On/Off
[image]
[image]

### Level Control
[image]

### Color Control
N/A

### Simple Metering
N/A

### Diagnostics
N/A

### Other clusters that are not mentioned above
N/A

--- 

Device documentation (in German only): https://docs.zigfred.ch/uno/

According to https://www.zigbee2mqtt.io/devices/ZFU-1D-CH.html, it should be able to get additional information, such as:
- button_1_single
- button_1_double
- button_1_hold
- button_1_release
- button_2_single
- button_2_double
- button_2_hold
- button_2_release
- button_3_single
- button_3_double
- button_3_hold
- button_3_release
- button_4_single
- button_4_double
- button_4_hold
- button_4_release

The device developer, @Atokulus, has provided what's needed for Z2M:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/siglis.js
https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts

WhistleMaster, 2022-08-14:
Seems like the button event are in this unknown cluster **0xFC42**. According to Z2M:

```
    manuSpecificSiglisZigfred: {
        ID: 0xFC42,
        manufacturerCode: 0x129C,
        attributes: {
            buttonEvent: {ID: 0x0008, type: DataType.uint32},
        },
        commands: {},
        commandsResponse: {},
    },
```

Is that something that could be done through DDF ?

Smanar, 2022-08-16:
So from that I m reading :
- It use the endpoint 0x06 for the realy
- It use the endpoint 0x07 for the dimmer
- It use the endpoint 0x05 for the buttonevent

The problem is wich one cluster are used for the buttonevent ?

On Z2M they are binding the onoff, the setlevel and the manufacture one, but from here it seem all event are using only the manufacture cluster https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/siglis.js#L11

To make short, for me, for buttonevent you need to use the legacy code OR a DDF script.
The legacy code use the xml file, for exemple

`[1, "0x01", "SENGLED", "COMMAND_0", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On (single press)"],`

And the code check for mode, endpoint, cluster and ZCL command
So if you have a different ZCL command for each buttonpress, you can use the legacy code.
Else it s possible to make a short JS code (can take the same used in z2m project with some adapation) to make the state/buttonevent managed by DDF for exemple like this switch https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/legrand/wireless_scenes_command.json

To check if it possible the easier way for me, is starting a basic DDF, enable log in deconz with "info" and "info_l2' make all action on button and see the result in logs.
You will see error message but with usefull informations.

WhistleMaster, 2022-08-16:
Thanks for your jumping here ! 

Just to add a point: the endpoint 0x05 is controlling a little LED on the device itself:
[image]

> The problem is wich one cluster are used for the buttonevent ?

From what I understand, the unknown cluster **0xFC42** is the one used for the events. See [from Z2M:](https://github.com/Koenkk/zigbee-herdsman/blob/6d43a67eefc5dd97641014c11a5295c3b6e518be/src/zcl/definition/cluster.ts#L4780):

```
    manuSpecificSiglisZigfred: {
        ID: 0xFC42,
        manufacturerCode: 0x129C,
        attributes: {
            buttonEvent: {ID: 0x0008, type: DataType.uint32},
        },
        commands: {},
        commandsResponse: {},
    },
```

Seems that there is an attribute (0x0008) which give then:

```
const button = buttonEvent & 0xff;
const state = (buttonEvent >> 8) & 0xff; 
```

WhistleMaster, 2022-08-16:
Could it be something like that, with DDF:

```
        {
          "name": "state/buttonevent",
          "parse": {
            "at": "0x0008",
            "cl": "0xfc43",
            "ep": 5,
            "eval": "Item.eval",
            "fn": "zcl",
            "mf": "0x129c"
          }
        }
```

And the eval part would be something like Z2M ?

Smanar, 2022-08-17:
Will be better to use an external script

`"script": "zigfrede.js"`

And yes Z2m use JS in their code, so we can mimic it, I can help you for that, but not sure we need it.
Perhaps it s not usefull and can use the legacy code with just editing the XML file.
Z2m use this script but if all actions are done only with the manufacture name, why they have bind the other cluster, so I prefer test before.

Make a basic DDF just with the ZHASwitch and check logs to be sure.

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "Siglis",
    "modelid": "zigfred uno",
    "product": "xxxxxxxxx",
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_SWITCH",
            "restapi": "/sensors",
            "uuid": [ "$address.ext", "0x05", "0x0006"],
            "items": [
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/battery",
                    "awake": true
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/reachable"
                },
                {
                    "name": "state/lastupdated"
                },
                {
                    "name": "state/buttonevent"
                }
            ]
        }
    ]
}
```

(There is no bind yet in this json, probably need them)

WhistleMaster, 2022-08-18:
Thanks for the reply ! I've tried the DDF with the correct log options, unfortunately, there is nothing in the log about the events when the buttons are pressed, see attached file (I've filtered on the node id):

[_deconz_logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9367981/_deconz_logs.txt)

Smanar, 2022-08-18:
It's not deconz logs.

On deconz take a look in deconz / help / debug view, with enabling flags "info" and "info_l2", then press a button.

BTW the device is reconized in deconz with the DDF ? You have a new sensor visible (for exemple in phoscon / help/ api information / sensor (I haven't see it in your log, can just load the DDF and make a "hot reload" in the DDF editor menu)

Smanar, 2022-08-19:
As eval try

`"Item.val = ZclFrame.at(0) * 1000 + ZclFrame.at(1)"`

`payload: 03020000`

I take the first byte so "03" * 1000 + second byte so "02".
Its not perfect but you will have something using "deconz rules" X00Y

PS : You can remove manufacture code

Edit :
Not possible use the editor you miss the "cmd" field edit the file to have something like

        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0xFC42",
            "cmd": "0x02",
            "Item.val = ZclFrame.at(0) * 1000 + ZclFrame.at(1)"
          }
        },

WhistleMaster, 2022-08-21:
Thanks ! 

I've updated a little bit, it was missing the "eval" in front of the formula and also +1 to the first byte as the payloads are starting at 0. This gives:

```
    {
      "name": "state/buttonevent",
      "awake": true,
      "parse": {
        "cl": "0xFC42",
        "cmd": "0x02",
        "eval": "Item.val = (ZclFrame.at(0)+1) * 1000 + ZclFrame.at(1)"
      }
    },
```

I'm putting everything together to also have the other functions working with that and I will post the DDF.

WhistleMaster, 2022-08-21:
Done ! Everything is working with the following DDF. I also get the events in HA, perfect ! Thank you very much @Smanar for the help !

**Note**: in order to get the "double" press event, you have to enable it in the Zigfred app for each button. The app configures the device through Bluetooth.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Siglis",
  "modelid": "zigfred uno",
  "product": "zigfred uno",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/zigfred_uno.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x05",
        "0x0006"
      ],
      "items": [
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0xFC42",
            "cmd": "0x02",
            "eval": "Item.val = (ZclFrame.at(0)+1) * 1000 + ZclFrame.at(1)"
          }
        }
      ]
    },
    {
      "type": "$TYPE_COLOR_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x05"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "refresh.interval": 5
        },
        {
          "name": "state/colormode",
          "description": "The currently active color mode."
        },
        {
          "name": "state/effect",
          "description": "The currently active effect.",
          "default": "none"
        },
        {
          "name": "state/hue",
          "description": "The current color hue."
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "state/sat",
          "description": "The current color saturation.",
          "refresh.interval": 5
        },
        {
          "name": "state/x",
          "description": "The current color x coordinate.",
          "refresh.interval": 5
        },
        {
          "name": "state/y",
          "description": "The current color y coordinate."
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x06"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x07"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Smanar, 2022-08-21:
So the device is compatible in HA ?
Haven't checked the value returned, but they need to be X00Y, X is the button number and Y the event.

If all is fine I will make the PR.

WhistleMaster, 2022-08-21:
> So the device is compatible in HA ? Haven't checked the value returned, but they need to be X00Y, X is the button number and Y the event.

Yes, it is !

[image]

> If all is fine I will make the PR, or you want to make it ? (just remember to remove all "description" lines and the "path" from the DDF) 

I can make the PR, that would be my first one :)

> BTW you miss too principal items in the ZHAswitch like model id, manufacture name and ect ....

I can add it, what are the one required ?

Smanar, 2022-08-21:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Siglis",
  "modelid": "zigfred uno",
  "product": "zigfred uno",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x05",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0xFC42",
            "cmd": "0x02",
            "eval": "Item.val = (ZclFrame.at(0)+1) * 1000 + ZclFrame.at(1)"
          }
        }
      ]
    },
    {
      "type": "$TYPE_COLOR_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x05"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/bri",
          "refresh.interval": 5
        },
        {
          "name": "state/colormode"
        },
        {
          "name": "state/effect",
          "default": "none"
        },
        {
          "name": "state/hue",
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "state/sat",
          "refresh.interval": 5
        },
        {
          "name": "state/x",
          "refresh.interval": 5
        },
        {
          "name": "state/y",
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x06"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x07"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/bri",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Can compare with this one, and test it.

To make a PR it's easy, mostly for only one file.
- go https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices
- create a new folder+file (it's the same thing) like siglis/zigfred_uno_wired_switch.json
- copy paste the contain (you will be able to edit it later)
- submbit a PR

You will be helped by the site, as you don't have right on the officiel github, it will create a fork under you github account, make your change inside and prepare the PR.
There is no danger you can't edit the official github, all changes will be on your fork, and after the PR will be submitted they will be synchronised, so if you want to edit something in your fork, the change will be too in the PR.

The PR list is here https://github.com/dresden-elektronik/deconz-rest-plugin/pulls if you want to see example.

WhistleMaster, 2022-08-21:
Thanks ! Mine almost looks like your latest one, except with the following. Are those needed ?

- "name": "config/on"
- "name": "config/reachable"

This switch is "virtual" as there is no physical switch behind, it's only to handle the events.

Smanar, 2022-08-21:
I m seing somethinf strange the endpoint 0x05 is used for the ZHAswitch  and the TYPE_COLOR_DIMMABLE_LIGHT

WhistleMaster, 2022-08-21:
> I m seing something strange the endpoint 0x05 is used for the ZHAswitch and the TYPE_COLOR_DIMMABLE_LIGHT

Indeed. The events are defined in the endpoint 0x05, which is also the COLOR_DIMMABLE_LIGHT physically, and the cluster 0xFC42 handles the events. I've tried to add the buttoneven in the COLOR_DIMMABLE_LIGHT but it does not work.

> Yep, they are needed, all third app use it, config/on is used for rules, config/reacheable trigger an error if set to false.
Makes sense !

> I think you have too much dimmmable light.

I don't think so :) There is two dimmmable lights on the switch: one little led which is controllable as well on the switch (to make fancy lights effect on the switch) and the second on can be connected on the switch.

WhistleMaster, 2022-08-21:
> Edit: I think the useless one is this one
> 
> ```
>     {
>       "type": "$TYPE_COLOR_DIMMABLE_LIGHT",
>       "restapi": "/lights",
>       "uuid": [
>         "$address.ext",
>         "0x05"
>       ],
> ```

This is what I wrote [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6258#issuecomment-1216972114), it's a small led on the device itself which can be controllable as well. You can change the color and so on. It's also working in HA, you can control it there.

Smanar, 2022-08-21:
ok, so all is fine ^^

You will have the cluster in the uniqueid "0x0006" so will be not the same uniqueid. Can use the Cluster 0xFC42 is you want instead of 0x0006

WhistleMaster, 2022-08-21:
You mean here:

```
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x05",
        "0x0006"
```

would become:

```
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x05",
        "0xFC42"
```

Smanar, 2022-08-21:
Yep, but it's like you want, as the device will use only the cluster 0xFC42 and not the 0x0006, will be better for me, but it's just cosmetic, and as HA have already memorised the one with 0x0006, you will need to remove it.

WhistleMaster, 2022-08-21:
Sure, I'll do it now to test.

WhistleMaster, 2022-08-21:
Done #6280 ! Thank you very much for the help !

Smanar, 2022-08-21:
Perfect, thx for the new DDF ^^

BTW, if you can remove the "path" line, you can just edit it here https://github.com/WhistleMaster/deconz-rest-plugin/blob/patch-1/devices/siglis/zigfred_uno_wired_switch.json all will be synchronised.

`"path": "/devices/zigfred_uno.json",`

But in reality, I don't think deconz use this field.

WhistleMaster, 2022-08-21:
Done :)

Atokulus, 2022-08-22:
Dear all

I am dearly sorry, that I have not yet been able to respond to your integration request. It is quite nice, you took initiative in integrating the device 😄 Great job and kudos to @WhistleMaster and @Smanar 😁

I just wrote an extensive comment in your pull request. Looks quite promising!

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6280#issuecomment-1222110594

WhistleMaster, 2022-08-22:
Thank you very much @Atokulus, your help will be gladly appreciated on this matter ! So let's continue the discussion in the PR then.

closed 2022-09-20 by github-actions[bot]