dresden-elektronik/deconz-rest-plugin #6262: Ikea E1703 smart plugs keep disappearing and can't be re-paired reliably

Larx, 2022-08-13:
## Describe the bug
I don't know if this is a bug or a misconception of how to handle Ikea smart plugs. I have two IKEA E1703 smart plugs that have been working reliable for two years. I don't know how I paired them in that time, probably was pretty simply. Seemingly overnight both plugs have disappeared at the same time from my network a few weeks ago and I cannot re-pair them reliably. I can't say if there has been some software update in this time. The plugs are in different rooms.
I once got the plugs re-paired by touch-link (press the button for >5sec very near to the ConBee), but after a few days both plugs were lost again - again at the same time.
Sometimes I also suddenly get the plug to appear suddenly in the web ui during multiple of attempts to pair it, but that is not reliable, I got the feeling that sometimes I don't notice that the plug is available and start another pairing process, so that this entry is already stale.

Things are complicated because there don't seem to be up-to-date instructions how to really pair this device. The documents I find on the net mostly do not work, as if for some other device?!

- Touch-link seems to be long-press >5sec -> clicking -> light fading, then blinking.
- According to some sources the plug should then be re-plugged to immediately start pairing like after a factory reset. This does not work for me.
- Some sites say 4x press to start normal pairing process -> does nothing for me.
- But by chance I found some mode where I do long-press ca. 2 secs. followed by serveral shot-presses and then the light then starts pulsing. Doesn't pair, however.

The fact that I don't know how to start a pairing process does not really help here ;-).

## Steps to reproduce the behavior
Remove stale entries from the UI and/or the VNC display, if there are any.
Try to re-pair in several different ways, see above (add new light)
Nothing happens most of the time, see above.

## Expected behavior
Plug should be recognized es light.

## Screenshots
Nothing to see.

## Environment
Host system: Raspberry Pi 4
Running method: Marthoc Docker container
Firmware version: 26780700 (yes, that's a beta version, I flashed it in the hope it would work better than the last stable version)
deCONZ version: 2.18.00 / 7.8.2022
Device: ConBee II
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? Yes, USB 3.0 external SSD

## deCONZ Logs

## Additional context
Maybe related, somehow at the same time my battery-powered wall switch Busch-Jäger 1-gang has been appearing as a 2-gang switch in the ui, and I had to assign functions to the non-existant second rocker in order to make it work again.

Mimiix, 2022-08-13:
Can you share some logs while pairing? APS, APS_L2, error, ERROR_L2, info, INFO_L2.

Smanar, 2022-08-13:
Perhaps similar issue than https://forum.phoscon.de/t/how-to-include-ikea-tradfri-control-outlet-e1703/2320

Larx, 2022-08-13:
> Perhaps similar issue than https://forum.phoscon.de/t/how-to-include-ikea-tradfri-control-outlet-e1703/2320

I was able to connect the e1703 flawlessly when I bought them almost two years ago. So they should be supported. But I cannot re-add them nowadays.
Also the pairing procedure as also described in your link does not work at all for me.

I just was able to pair one again a few minutes ago by this procedure:
long press 5sec, plug clicks and led fades out and in, then press for about 2 sec, then about 4 to 5 fast clicks in a row. The led then pulses and Deconz found & connected the plug.
I don't know if this was a one-time success and I didn't save logs, unfortunately. And I don't know if this pairing will last...

Mimiix, 2022-08-13:
But they are paired now?

Larx, 2022-08-13:
I tried it out: For me pairing my E1703s works like this (I tried it two times with both plugs):

- Start pairing mode in Deconz
- Press button 5 secs, release -> light fades out, several clicks, light fades in again
- Press button 2 secs, then press button 4 times quickly in a row -> light begins to fade in and out
- Light will appear as "On/Off Plug Unit XX"

If I do only the first step, pairing works only if very near to the ConBee-Stick. If I do only the second step, nothing happens at all.

Now I will see if the units stay connected and will reopen this bug if the connection is lost again after a few days...

closed 2022-08-13 by Larx