dresden-elektronik/deconz-rest-plugin #6263: Ikea Fyrtur not showing in the Rest API for pairing but does show in VNC

JustAnotherPU, 2022-08-13:
**Describe the bug**

Ikea Fyrtur Blinds are not showing in the Rest API, therefore, I can't make the pairing process in the web interface but when I connect through VNC it does show them. 

**Steps to reproduce the behavior**

1 - Start searching for light in the web interface 
2 - Nothing happens 
3 - Connect via VNC to DeCONZ blinds are show and controllable from there 

**Expected behavior**

Be able to pair the device and use the REST API to control it from HA or Homebridge 

**Screenshots**
[image]

[image]

**Environment**

Host system: Qnap
Running method: DeConz and Homebridge running via Docker on a Qnap Nas
Firmware version: (26720700)
deCONZ version: (2.18.0)
Device: (ConBee II)
Do you use an USB extension cable: (yes, 16ft)
No other USB or Serial devices connected.

**deCONZ Logs**

N/A

**Additional context**

Succesfully connected both aqara sensor and an Ikea remote. issue is only with the blinds

Smanar, 2022-08-13:
It's you that have changed the node name ?
I don't remember is covering are visible in phoscon, you can't see them direclty in the API ? (Already have checked in HA or homebridge ?)

Thoses devices don't have DDF but can make a try with this one https://forum.phoscon.de/t/ikea-fyrtur-blinds/1858/6

JustAnotherPU, 2022-08-13:
@Smanar I did not change the node name. I did check in homebridge and nothing is shown unfortunetly

SwoopX, 2022-08-13:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6110

closed 2022-09-12 by github-actions[bot]