dresden-elektronik/deconz-rest-plugin #6265: Device name

jleeca, 2022-08-13:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2022-08-13:
Hello can you try this DDF just with adding your device to the list ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6257

jleeca, 2022-08-13:
I can confirm that the device works now when the DDF file you mentioned is used with the Manufacturer replaced to _TZ3000_7tbsruql.

closed 2022-08-13 by jleeca

Smanar, 2022-08-14:
Ok, I will ask to the guys that make the PR to add it too.

Middendorf, 2022-08-15:
Hallo 

Ich habe mit dem Hersteller geschrieben. Er hat mir diesen Link gegeben. Leider wird bei mir nur der Conbee2 Stick angezeigt. wie kann ich ein gerät in deconz app suchen? 

Über die PhosconApp blinkt mein Gerät so als ob er sich verbinden möchte, dann suche ich sensoren (andere Geräte) kurz darauf hört er auf zu blinken aber mein Türsensor wird nicht angezeigt. 

Ist der sensor defekt oder nur noch nicht in der APP (Software) hinterlegt?? 

Danke für eure Hilfe.

Mimiix, 2022-08-15:
@Middendorf Can you please post in english?

Middendorf, 2022-08-16:
Hi

Apologies for my bad English first.

I have written to the manufacturer. He gave me this link. Unfortunately, only the Conbee2 Stick is displayed for me. how can i search for a device in the deconz app?

My device flashes via the PhosconApp as if it wants to connect, then I search for sensors (other devices) shortly after it stops flashing but my door sensor is not displayed.

Is the sensor defective or just not stored in the APP (software)??

Thank you for your help.

Smanar, 2022-08-16:
If your device is the _TZ3000_7tbsruql , the PR is not yet merged, check https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6257

If you want to use it now, you need to create yourself the DDF on your OS.

Better to use phoscon than deconz for inclusion.

Middendorf, 2022-08-16:
Hi,

unfortunately I can't get it to run at all under Phoscon (displayed). If I go under sensors, set it to sync (flashes), then under look for new sensor, other devices. The search begins, the sensor flashes quickly, shortly afterwards it flashes briefly and then off. Unfortunately, no new sensor is displayed under Devices.
Is there another way to integrate the sensor?

Thanks Stefan

Mimiix, 2022-08-16:
@Middendorf  Is it shown in the RestAPI?

Phoscon needs to add support on their own.

Smanar, 2022-08-17:
@Middendorf you have added the DDF ?