dresden-elektronik/deconz-rest-plugin #6266: Water sensor _TZ3000_ww9i3e0y

jleeca, 2022-08-13:
## Device

- Product name: Water sensor (https://www.aliexpress.com/item/1005003446278673.html?spm=a2g0o.order_list.0.0.15101802pc6qWy)
- Manufacturer: _TZ3000_ww9i3e0y
- Model identifier: TS207
- Device type 
  - Sensor

## Screenshots

[image]
[image]

### Basic
[image]

### Identify
[image]

### Power Configuration
[image]

### IAS Zone
[image]

jleeca, 2022-08-14:
FYI, I just tried similar _TZ3000_d16y6col_water_sensor.json DDF file (after replacing the manufacturename to "_TZ3000_ww9i3e0y") but it's not working so it looks like there's more attributes which need to be changed?

Smanar, 2022-08-14:
There is perhaps a typo in the model id ?

`Model identifier: TS207`

You need to have the "Gold" Status in the DDF

`  "status": "Bronze",`

jleeca, 2022-08-15:
Changing the Status to "Gold" in the DDF fixed the issue. The model number TS0207 is correct from the original _TZ3000_d16y6col_water_sensor.json DDF file so not needed to modify it.

Smanar, 2022-08-15:
But you can make a pr for the DDf work for both devices

```
  "manufacturername": ["_TZ3000_d16y6col", "_TZ3000_ww9i3e0y"],
  "modelid": ["TS0207", "TS0207"],
```

I can make the change if you want ?

jleeca, 2022-08-15:
Yes, I still needed to modify manufacturernams for those two devices in DDF files. I only wanted to clarify that the model number TS0207 is correct. Anyway, please make a pr so others can use those two types of  devices.

Smanar, 2022-08-15:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

maykey22, 2022-10-27:
Hi, are there any plans to merge that PR soon?

Smanar, 2022-10-27:
>on 6 Aug ...

I will go to ask to manup

maykey22, 2022-12-04:
Any updates? I have this device and would like to use it 😀

Smanar, 2022-12-04:
No, I m  asking a "roadmap" ATM.
A massive PR merge was in project but there is some problem on recent deconz version, and I think the first objective is correct the blocking issues first.

maykey22, 2022-12-04:
I got it, IMO you should consider creating smaller PRs. Smaller pieces of code are easier to integrate.

Smanar, 2022-12-04:
Right, but the problem is lot of device can need the same file, or depend each others.

maykey22, 2022-12-04:
What is the strategy to deal with this? We try to discover all dependant devices? The consecutive commits to support new devices is the stabilization phase?

Smanar, 2022-12-05:
manup have asked same ^^. We will have conflict for sure.
For me better to merge small PR from "little" user, because they come rarely. And I can solve confflict on my PR after.