dresden-elektronik/deconz-rest-plugin #6267: MoesGo Zigbee 3-Gang Smart Switch No-Neutral

forast3r, 2022-08-14:
## Device

- Product name: MoesGo Zigbee 3-Gang Smart Switch No-Neutral
- Manufacturer: _TZ3000_qewo8dlz
- Model identifier: TS0013
- Device type : 
  - Light

purchased recently on "big A": https://www.a***.de/gp/product/B096X834WB/

Detected on deconz but not shown on phoscon

## Screenshots
Endopints and clusters of the node:
[image]

Node info panel
[image]

### Basic
[image]

### Identify
[image]

### Groups
(partial screenshot)
[image]

### Scenes
(partial screenshot)
[image]

### On/Off
(partial screenshot)
[image]

### Other clusters that are not mentioned above
All other clusters appear empty or without significant data.

Smanar, 2022-08-14:
Hello, to try can start with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_qewo8dlz",
  "modelid": "TS0013",
  "product": "Wired Switch 3 gangs",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x03"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0006"
    }
  ]
}
```

Can have issue with broadcast request, and this DDF don't use the scene cluster (IDK if used)

forast3r, 2022-08-14:
Hi, thanks for the quick reply.
I'm trying to figure how to create the file (I'm using HA deconz addon) because when ssh'ing in HA docker command does not exist.

forast3r, 2022-08-15:
Hi, managed to include the json in the deconz container and forcing the Hot Reload of the file.

As a result, the first two switches are working right, but the third appears greyed oud (unknown in HA):
[image]

In HA:
[image]

Locally toggling the state for the third switch does not refresh the state in phoscon.

What could I try?

Smanar, 2022-08-15:
Oups, my bad, I have used a battery switch json for the third one

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"_TZ3000_qewo8dlz",
   "modelid":"TS0013",
   "product":"Wired Switch 3 gangs",
   "sleeper":false,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_ON_OFF_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x01"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/on",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      },
      {
         "type":"$TYPE_ON_OFF_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x02"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/on",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      },
      {
         "type":"$TYPE_ON_OFF_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x03"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/on",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":5,
               "max":1800
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":2,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":5,
               "max":1800
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":3,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":5,
               "max":1800
            }
         ]
      }
   ]
}
```

No broadcast issue ? (others devices that work in same time than this one)

forast3r, 2022-08-15:
Tested the new one and the three lights are working as expected.

Will check if I detect some broadcast issue, but I don't plan switching lots of devices simultaneously.

If you need further tests let me know.

Regards

Smanar, 2022-08-16:
NP, have made a PR for it

forast3r, 2022-08-23:
Thanks for making the PR, I confirm that an identical 2nd unit is working fine (but at the same time the first has stopped working, I guess too many group changes in a row, will remove from the network and add again tomorrow)

Regarding the **broadcast issues,** I can confirm that at least in the new unit, sending a broadcast toggle command to a light group in Phoscon does not work, but sending the individual toggle commands does.

Smanar, 2022-08-23:
But this device is a "light' not a "switch", so it receive order from battery switch, I think in phoscon it will work if you add it in a switch (battery one) group.

This device don't have output cluster but if it have broadcast request (tuya are magic) they need to be blocked.

forast3r, 2022-08-23:
Thanks for the reply, I will experiment a little more with the groups and the battery switches (will reconfigure some battery switch to test this)

BTW this device is reported as a battery device (I guess it is because of the "no neutral" supply)

Can you explain this: 

> This device don't have output cluster but if it have broadcast request (tuya are magic) they need to be blocked.

And do you need me to check the logs ?

Smanar, 2022-08-24:
>BTW this device is reported as a battery device

In deconz ? Yes this device is an end device, so no router features, but some others "no neutral" are.

>Can you explain this:

From your capture, you have blue cluster (input) and grey one (output). The "command device" use his output cluster to command the "receptor device" using the input one.
The device look at 99% for a "receptor device" for me. But there is some unknow cluster and it s tuya so I can't be sure at 100%

forast3r, 2022-08-24:
Thanks for the clarification. Didn't know the color coding of input and output clusters.

Had time to remove and re-link the first device, now both devices are working as expected.

closed 2022-09-23 by github-actions[bot]

nebarh01, 2023-02-19:
Hello,
I have just puch3s this exact model, but the DDF doesn't appear to be correct... the device imports fine into deCONZ, Phoscon and HomeAssistant, buy when I turn on / off the virtual switch all the loads turn on / off together.
After editing the DDF the best I can figure out from the available documentation by changing the fn: to "tuya" rather than "zcl" for the "state/on" cluster, the virtual switches now turn on / off independently - so managed to make some progress there. However, the physical loads all still turn on / off together. It's like HA / Phoscon is sending the signal to turn on / off load 1 /2 /3, but tue device is receiving a command to turn all loads on regardless of switch virtual switch is pressed.
The physical switches work exactly as expected, and report in HA etc correctly too.

Happy to provide any information or detail to help,  obviously. Just looking for some help / pointers please as I think I have exhausted by ability unfortunately.

Smanar, 2023-02-19:
I think you have a more recent version, and your device need the tuya unlock sequence to work, else a request to 1 endpoint make all output react.
I m sorry but you can't do something with DDF, need to be done on c++ side.

nebarh01, 2023-02-19:
Ah, OK.
Is that something I can code to fix? Or is that going to be lower down in the code, maybe somewhere in the application that will need a recompile?

Smanar, 2023-02-20:
Have made a fast check, and yes z2m is using the tuya unlock sequence for TS0013

`await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);`

I can (try to) make the code, but I need someone with a machine with a Full OS on linux to compile the code and test it.

nebarh01, 2023-02-20:
@Smanar if you could, that would be fantastic! My Linux is not amazing but I have raspbian OS (debian?), deconz installed locally (not docker or anything) and HA in python venv. I'm more than happy to test anything and share results, info data etc, whatever you need.

I have been reading up to try and sort it myself, but I keep reading (all from you I think!) that this needs to be done in c++, but it seems lots of people need this and more and more devices are becoming unsupported :(. If you're able I'm more than happy to test and work with you!

Smanar, 2023-02-20:
Raspian is perfect ^^.

The compilation can take lot of ressources (during the compilation) but nothing dangerous for your data. I will make the code and explain you how to test it tommorow.
I think I will make a hack to test it fastly , and if it work a clean method to use it using DDF.

nebarh01, 2023-02-20:
Sounds like a plan @Smanar!!
I'm sure I can spare a few cycles ;)

Let me know what you need to do and I'll get it tested asap!

Smanar, 2023-02-21:
So I have some check and I have some bad and good news.

- It seem it's possible to do the magic using only DDF
- I have never succed with this kind of stuff, it's a little hacky (I will explain later)
- The DDF core is able to make a read request on more than 1 attribute in one request

So the new DDF to try

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_qewo8dlz",
  "modelid": "TS0013",
  "product": "Wired Switch 3 gangs",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
[... 204 lines cut ...]
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = Attr.val"
          },
          "default": 0
        }
```

I have added a special field.

- "config" field are only for "sensor" not "light", so we will need to hide it later, for the moment I think it will be not an issue, it's not the first device that do that.
- It's not possible to make a report so I hope it will trigger the poll with `"refresh.interval": 300,` so the request will be done every 5 mn
- I m using `"config/interfacemode"` so the result need to be a number, but the attribute 0xfffe is an enum8, can compare the result on the GUI.

So If I m right, the device need to work after 5mn, IDK if the magic tuya unlock sequence need to be done immediatly or can be done 5 mn after

nebarh01, 2023-02-21:
@Smanar, you are a legend!
Updated the DDF with the code you sent, works first time!

I have a 3 gang and 2 gang, updated both with the new DDF (removed ep:3 for the 2gang) and both worked perfectly!
Tested immediately after pairing and 5, 10, 15 minutes after - no change in behaviour, still working perfectly.
My 3 gang I deleted from deCONZ and then did a new pairing - worked perfectly.
My 2 gang I had already added, so just updated DDF and restarted deCONZ service and it started working straight away; no new pairing required.
I tested turning all the virtual switches on and off in HomeAssistant in different orders, and every time it turned the correct physical switch on / off.
I also tried turning the physical switches on / off and watched the virtual switches, and they all behaved perfectly too.

Looks like you cracked it with the first try! :D

You said that this was a bit of a hacky way to do it - do you still want to tidy the code? I can keep testing new versions if you want...

Smanar, 2023-02-22:
I am disgusted to see it was so easy, long months lot of devices are locked because they need this fuc**ing sequence.

I m searching a way to make it cleaner, but I don't have idea, perhaps a new field in DDF ?
I think this hack will be fine for the moment, just increase the poll to 24h, no need to make it every 5mn, I just hope for the first request don't happen after 24 h.
And can hide the field for compatiblity.

```
        {
          "name": "config/interfacemode",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": [
              "0x0004",
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = Attr.val"
          },
          "public": false,
          "default": 0
        }
```

nebarh01, 2023-02-22:
Hey,

So I tested your new code with the 24h instead of 5min - works fine.
Deleted and repaired device and it works perfectly still!

Sorry, I'm not sure where I'm supposed to put this bit? : const char *RConfigConfigured = "config/configured"; with "defaut":false
But, I would say that this is just more code, not tidying up maybe? If it works as it is and you said other devices are using this already, if it works and doesn't cause issues, maybe its fine and you don't need the extra code?

I *think* I took some logs (attached) that you wanted... the device ID in question is a4c13859509ab0b2. Im not sure what you class as spamming, but that ID appears many times.
I only ran the logs for 1min - I can run longer if you need, just let me know how long :)

[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/10807645/debug.txt)

BabaIsYou, 2023-02-22:
> I am disgusted to see it was so easy, long months lot of devices are locked because they need this fuc**ing sequence.

That's great. Perhaps could we put `"public": false` for avoiding access to this _useless_ attribute from REST-API.

Another candidate could be `config/controlsequence`  ?

Smanar, 2023-02-23:
Logs seem fine, the only spammed request is

>19:23:17:863 APS-DATA.request id: 207, addrmode: 0x03, addr: 0x00212effff07b613, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00

So not the same device and if I m right this request if for LQI

For the field to use, we will see with others devs. But I m think we can use for the moment

```
        {
          "name": "config/interfacemode",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": [
              "0x0004",
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = (Attr.val != 0)"
          },
          "public": false,
          "default": false
        }
```

The value will be false at start, and after the first request will be true (if the attribute 0xfffe containt all valus different than 0)

We can have a problem, some device don't support attribute read on the attribute 0xfffe, so this request will fail, hoping for deconz don't remake it in loop mode if it have error.

nebarh01, 2023-02-23:
That's excellent @Smanar.
I'll have to leave the coding to you!

Do you need / want me to test this new code, or is  it just a change in name for the variable?

I don't have any other different devices, so I can't test that bit unfortunately...

Smanar, 2023-02-23:
>Do you need / want me to test this new code, or is it just a change in name for the variable?

Yes, pls ^^, it's just a variable name, but as we will share this code, better to test it, can have an impact on result

>I don't have any other different devices, so I can't test that bit unfortunately...

No problem, thx for help BTW, there is so much device locked, will not be hard to find new one, @BabaIsYou have already asked on some old issues.

nebarh01, 2023-02-23:
@Smanar,

Please find attached logs after changing the code.
Switch works perfect straight after new pairing.
The ID should be the same - a4c13859509ab0b2
[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/10818846/debug.txt)

nebarh01, 2023-02-24:
Thank you @Smanar!!
Thank you from me and I suspect many others in the community for taking the time to figure this silliness out and find a solution!!

:D

nebarh01, 2023-03-11:
Hey @Smanar,
Just thought I would raise this here in case it's an issue...
I have noticed that these devices have randomly become unavailable - admittedly I don't know if its something in the ddf or an issue with my homeassistant (I am still working on testing this), but only these switches seem to have the issue; all my other devices seem fine. 

If I figure it out I'll post back.
P.s. hope you've been well!!

Smanar, 2023-03-11:
All the one that use the tuya unlock sequence ?

Can remove it to test, one time the device is connected, you don't need it, so you can remove it from the DDF, and restart deconz.

nebarh01, 2023-03-11:
Yes, it seems to be.
I'll try switching back to the built in ddf for the devices to test...

Smanar, 2023-03-11:
Can try to increase thoses fields

`"refresh.interval": 5`

Set the interval to 20

nebarh01, 2023-03-29:
@Smanar Sorry I haven't responded in a while.
Making the change you suggested didn't solve the issue and seemed to make really weird things happen as well!

I ended up rebuilding my Pi from scratch, and reinstalled / reconfigured deCONZ and HomeAssistant and a few other bits, and they seem more stable now thankfully. So I think it was either some corruption / misconfiguration on the Pi or maybe the location (I had moved it recently to redecorate, but oddly only saw issues with these devices). Either way, think we are good here! :D

Thanks so much for the help and suggestions!