dresden-elektronik/deconz-rest-plugin #6269: Beok TGR85B-ZB clone (_TZE200_2ekuz3dz) heating thermostate

mrblur, 2022-08-16:
## Device

- Product name: ZIGBEE THERMOSTAT (Beok TGR85B clone)
- Manufacturer: _TZE200_2ekuz3dz
- Model identifier: TS0601
- Device type :
  - Thermostat

This seems to be a clone of Beok TGR85B-ZB, it came with no branding and manual says only "Zigbee THERMOSTAT USER GUIDE".
It looks like this one: https://zigbee.blakadder.com/Beok_TGR85-ZB.html, seems to be added to z2m already at https://github.com/Koenkk/zigbee2mqtt/issues/12849 (and my device looks exactly like the one linked in here), should be supported by zigpy with https://github.com/zigpy/zha-device-handlers/pull/1507/commits.

So far my device was able to only join the network. It did flip the time by 5hr upon joining. And somehow ended up with time being set as 25:13 (:roll_eyes:). 

## Screenshots

[image]

### Basic
[image]
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

Smanar, 2022-08-17:
Long story but we need a PR a waiting list to include tuya TRV with DDF.

mrblur, 2022-08-24:
It seems that PR somehow got stuck.

Can we push it forward in any way? I'm open for debugging and testing. Have some experience with Qt and C++.

"Winter is coming" and so far my stupid device flips the time randomly during the day. I've seen hours of 27 and 35 already :roll_eyes: .

I've seen something about incorrect time synchronization on tuya flying around, is it possible to "patch" it over with a DDF?
I can live with it being unsupported, but this time sync issue breaks scheduling completely.

Smanar, 2022-08-25:
I have just updated the PR.

Sure you can compile the code on your side.
You have 2 possibilities, start from official code and mimic an exiting TRV that use the same dp than this one. From this link https://github.com/Koenkk/zigbee-herdsman-converters/pull/3840/files it seem this device use same dp than moes TRV (for exemple 40 for childlock and 16 for heatpoint)
Or start from the tuya TRV PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054 and just create a DDF for your device

On my side I prefer help you for the second one, because the first one will be never merged, and the second can improve the PR.

The time sysnchronisation use only the leagcy code, so if I m right there is no impact on the solution you wana try.

mrblur, 2022-09-16:
Keep this open, the issue with time sync is still there and irritating as never.

I also tried to compile the plugin myself, but it seems my Gentoo uses different libraries as the plugin fails to load. It looses all device names, there is no DDF editor available. Waiting for THE PR to get merged in hope this will fix all my issues :+1:

Smanar, 2022-09-16:
9 Days ago, there was  a PR probably usefull for you https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6305

mrblur, 2022-09-16:
Thanks @Smanar, already on latest release (unraid + docker), re-added device to the network, but it doesn't stop it from displaying "29:55" once in a while :laughing: 

Either I read the changelog wrong, or this is something different happening. Like the time sync somewhat is working, as the minutes and day of week is always good, only the hour display switches during the day. I can setup the hour manually and after some time it displays "25:14" again :shrug: 

I've seen something about tuya byte ordering on time sync while browsing old commits, could this be related?

Smanar, 2022-09-16:
So you are on version 2.18.1 ?
It's perhaps because your device is not supported, so it doesn't use the deconz code at all.

mrblur, 2022-09-16:
Yes, I am.

I see the app (at least for docker) is built on GCC 7.5, that is ancient on Gentoo (anything below 10 is..), I see there is a docker version available, would try again this weekend over docker, maybe this time the plugin would be compatible.

Smanar, 2022-09-16:
The problem is your device is not supported. So lof of thing are just ignored for your device.

jplahl, 2022-09-27:
@Smanar 
yesterday I posted the following thread in the phoscon forum before I found this issue.
[https://forum.phoscon.de/t/hama-tuya-thd-smart-radiator-trv-tze200-yw7cahqs-some-state-values-not-updated/2512](url)

I think my problems with the Hama smart radiatior have the same reason like all the tuya based thermostat devices.

So I compiled your git repository for testing your tuya changes. But for my hama thermostat nothing changes. Still no update for temperature or battery status. Maybe I missed something.

Smanar, 2022-09-27:
>So I compiled your git repository for testing your tuya changes

Hu, wich one ?

>Still no update for temperature or battery status

This is strange because it worked when you include the device or when you use the device manualy. And tuya device don't use bind and reporting. So if it work at least one time, no reason for it stop working ?

As you have the GUI, you can take a look in deconz log with "info" you can see thoses logs lines, for all incoming request from the device
```

        DBG_Printf(DBG_INFO, "Tuya debug 4 : Address 0x%016llX Payload %s\n", ind.srcAddress().ext(), qPrintable(zclFrame.payload().toHex()));
        DBG_Printf(DBG_INFO, "Tuya debug 5 : Status: %u Transid: %u Dp: %u (0x%02X,0x%02X) Fn: %u Data %d\n", status, transid, dp, dp_type, dp_identifier, fn, data);
```

Have you tried to include the device a second time ? (without deleting it)

jplahl, 2022-09-27:
> Hu, wich one ?
https://github.com/Smanar/deconz-rest-plugin/tree/tuya_trv_clean

> Have you tried to include the device a second time ? (without deleting it)
yes

>This is strange because it worked when you include the device or when you use the device manualy. And tuya device don't use bind and reporting. So if it work at least one time, no reason for it stop working ?

That's. If the mapping/code is wrong why are the values set the first time. I have no idea. It seems that only a change of the heatsetpoint value is transmitted.  If I manually lock or unlock the device the state is always locked. 

Here are some logs
change temperature at device:
`16:51:21:367 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000367020004000000d2
16:51:21:369 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 210
16:51:22:212 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000367020004000000dc
16:51:22:214 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 220
16:51:23:051 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000367020004000000d2
16:51:23:053 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 210
16:51:23:893 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000367020004000000be
16:51:23:895 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 190
16:51:25:309 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000367020004000000c8
16:51:25:311 Tuya debug 5 : Status: 0 Transid: 3 Dp: 615 (0x02,0x67) Fn: 0 Data 200`

rejoin of the device:
`17:00:22:940 send permit join, duration: 65
17:00:26:492 Device TTL 3418 s flags: 0x7
17:00:27:920 ZCL attribute report 0x540F57FFFE23E58A for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:00:28:825 Tuya : Schedule command
17:00:28:828 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000973000018000000000000000000000000000000001600000000000000
17:00:28:830 Tuya debug 5 : Status: 0 Transid: 9 Dp: 115 (0x00,0x73) Fn: 0 Data 0
17:00:28:831 Tuya : Unknow Schedule mode
17:00:28:913 Bind response success for 0x540f57fffe23e58a ep: 0x01 cluster: 0x0000
17:00:29:124 Tuya : Schedule command
17:00:29:126 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000d6e000018000000000000000000000000000000001400000000000000
17:00:29:128 Tuya debug 5 : Status: 0 Transid: 13 Dp: 110 (0x00,0x6E) Fn: 0 Data 0
17:00:29:130 Tuya : Unknow Schedule mode
17:00:29:307 Tuya : Schedule command
17:00:29:309 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 001177000018000000000000000000000000000000000000000000000000
17:00:29:311 Tuya debug 5 : Status: 0 Transid: 17 Dp: 119 (0x00,0x77) Fn: 0 Data 0
17:00:29:313 Tuya : Unknow Schedule mode
17:00:31:945 DB SELECT manufacturername FROM nodes WHERE mac LIKE '60:a4:23:ff:fe:53:73:4f%' COLLATE NOCASE: _TZ3000_kohbva1f
17:00:31:965 DB SELECT manufacturername FROM nodes WHERE mac LIKE '60:a4:23:ff:fe:53:6c:b8%' COLLATE NOCASE: _TZ3000_kohbva1f
17:00:31:982 DB SELECT manufacturername FROM nodes WHERE mac LIKE 'bc:33:ac:ff:fe:b1:3d:9f%' COLLATE NOCASE: _TZ3000_iivsrikg
17:00:31:998 DB SELECT manufacturername FROM nodes WHERE mac LIKE '60:a4:23:ff:fe:53:6b:41%' COLLATE NOCASE: _TZ3000_kohbva1f
17:00:32:939 Skip idle timer callback, too early: elapsed 927 msec
17:00:33:875 0x26BF seems to be a zombie recv errors 13
17:00:33:877 Node zombie state changed 0x84fd27fffeb8bb9a
17:00:34:355 0x2CAF seems to be a zombie recv errors 25
17:00:34:357 Node zombie state changed 0x680ae2fffe6e05aa
17:00:34:533 DeviceAnnce of SensorNode: 0x04CD15FFFE100F22 [1]
17:00:34:536 	nwk address changed 0x0000 -> 0xE5A9 [2]
17:00:34:538 device announce 0x04CD15FFFE100F22 (0xE5A9) mac capabilities 0x80
17:00:34:540 set fast probe address to 0x04CD15FFFE100F22 (0xE5A9)
17:00:34:542 FP indication 0x0000 / 0x0013 (0x04CD15FFFE100F22 / 0xE5A9)
17:00:34:543                       ...     (0x04CD15FFFE100F22 / 0xE5A9)
17:00:34:545 device announce 0x04CD15FFFE100F22 (0xE5A9) mac capabilities 0x80
17:00:34:546 DEV Tick.Join: event/device.anounce
17:00:34:548 DEV Tick: fast poll 0x04CD15FFFE100F22, mac capabilities: 0x80
17:00:35:439 [1] get node descriptor for 0x04cd15fffe100f22
17:00:35:441 ZDP get node descriptor for 0xE5A9
17:00:41:243 0xE5A9 nwk changed to 0x69F4
17:00:41:246 DeviceAnnce of SensorNode: 0x04CD15FFFE100F22 [1]
17:00:41:249 	nwk address changed 0xE5A9 -> 0x69F4 [2]
17:00:41:252 FP indication 0x0000 / 0x0013 (0x04CD15FFFE100F22 / 0x69F4)
17:00:41:254                       ...     (0x04CD15FFFE100F22 / 0xE5A9)
17:00:41:255 device announce 0x04CD15FFFE100F22 (0x69F4) mac capabilities 0x80
17:00:41:257 FP indication 0x0000 / 0x0013 (0x04CD15FFFE100F22 / 0x69F4)
17:00:41:259                       ...     (0x04CD15FFFE100F22 / 0xE5A9)
17:00:41:261 device announce 0x04CD15FFFE100F22 (0x69F4) mac capabilities 0x80
17:00:41:263 DEV Tick.Join: event/device.anounce
[... 99 lines cut ...]
17:00:55:849 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000c7600000c000000000000000000000000
17:00:55:851 Tuya debug 5 : Status: 0 Transid: 12 Dp: 118 (0x00,0x76) Fn: 0 Data 0
17:00:55:852 Tuya : Unknow Schedule mode
17:00:56:018 ZCL attribute report 0x84FD27FFFEABBE91 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:00:56:056 Tuya : Schedule command
17:00:56:058 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 00107200000c000000000000000000000000
17:00:56:060 Tuya debug 5 : Status: 0 Transid: 16 Dp: 114 (0x00,0x72) Fn: 0 Data 0
17:00:56:061 Tuya : Unknow Schedule mode
[... cut ...]
17:01:04:953 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x5C0272FFFE5DD8D4 (seems to be active)
17:01:05:483 Bind response success for 0x5c0272fffe5dd8d4 ep: 0x01 cluster: 0x0000
17:01:05:547 ZCL configure reporting rsp seq: 164 0x5C0272FFFE5DD8D4 for ep: 0x01 cluster: 0x0000 attr: 0x4000 status: 0x00
17:01:06:797 ZCL attribute report 0xCC86ECFFFE499B04 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:01:10:373 ZCL attribute report 0x60A423FFFE536B41 for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
`

Smanar, 2022-09-27:
Ha ok, this PR is for future, to add new Tuya TRV using the DDF editor, your device is already in the code, so it need to work.

The temperature is this one

17:00:50:793 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000266020004000000e6
17:00:50:795 Tuya debug 5 : Status: 0 Transid: 2 Dp: 614 (0x02,0x66) Fn: 0 Data 230

0x02 0x66 > 23 degre.
But it s during the inclusion, and it work for you at this moment.
After 5/10 mn do you still have return like thoses one ?

And if you try to set yourself the heatpoint (using API, it work manualy), it work or not ? If not, nothing visible in logs ?

jplahl, 2022-09-27:
it seems that status update are only send every 1 hours.

Setting heatsetpoint is working
`18:26:06:821 Send Tuya request 0x04CD15FFFE100F22 : Dp_type: 0x02, Dp_identifier 0x67, data: 000000be
18:26:11:688 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 005e67020004000000be
18:26:11:690 Tuya debug 5 : Status: 0 Transid: 94 Dp: 615 (0x02,0x67) Fn: 0 Data 190
18:26:11:806 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 005e67020004000000be
18:26:11:808 Tuya debug 5 : Status: 0 Transid: 94 Dp: 615 (0x02,0x67) Fn: 0 Data 190
18:26:11:930 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 005e67020004000000be
18:26:11:933 Tuya debug 5 : Status: 0 Transid: 94 Dp: 615 (0x02,0x67) Fn: 0 Data 190
18:26:12:034 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 005e67020004000000be
18:26:12:037 Tuya debug 5 : Status: 0 Transid: 94 Dp: 615 (0x02,0x67) Fn: 0 Data 190`

Setting locked is not working
`18:28:36:944 Send Tuya request 0x04CD15FFFE100F22 : Dp_type: 0x01, Dp_identifier 0x07, data: 01
no response`

Setting offset is working
`18:30:30:519 Send Tuya request 0x04CD15FFFE100F22 : Dp_type: 0x02, Dp_identifier 0x1B, data: ffffffff
18:30:34:719 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 00c71b020004ffffffff
18:30:34:721 Tuya debug 5 : Status: 0 Transid: 199 Dp: 539 (0x02,0x1B) Fn: 0 Data -1
18:30:34:836 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 00c71b020004ffffffff
18:30:34:839 Tuya debug 5 : Status: 0 Transid: 199 Dp: 539 (0x02,0x1B) Fn: 0 Data -1
18:30:34:970 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 00c71b020004ffffffff
18:30:34:975 Tuya debug 5 : Status: 0 Transid: 199 Dp: 539 (0x02,0x1B) Fn: 0 Data -1`

Also setting mode is working: heat -> auto
`18:33:51:447 Send Tuya request 0x04CD15FFFE100F22 : Dp_type: 0x01, Dp_identifier 0x6C, data: 01
18:33:51:450 Send Tuya request 0x04CD15FFFE100F22 : Dp_type: 0x01, Dp_identifier 0x65, data: 01
18:33:51:965 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 001b6c01000101
18:33:51:968 Tuya debug 5 : Status: 0 Transid: 27 Dp: 364 (0x01,0x6C) Fn: 0 Data 1
18:33:52:196 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 001c6501000101
18:33:52:199 Tuya debug 5 : Status: 0 Transid: 28 Dp: 357 (0x01,0x65) Fn: 0 Data 1
18:33:54:307 ZCL attribute report 0x60A423FFFE53734F for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
`

Smanar, 2022-09-27:
>it seems that status update are only send every 1 hours.

So the temperature is updated but only every  hour ? From my memory it's not something that can be changed ...

For the childlock can you change it manualy to compare the request 0x01 0x07 on your previous logs.

>8:26:06:821 Send Tuya request 0x04CD15FFFE100F22 : Dp_type: 0x02, Dp_identifier 0x67, data: 000000be 

In this exemple deconz is sending the tuya request 0x02 0x67

jplahl, 2022-09-27:
The device has two different "temperature" 

heatsetpoint -> desired temperature
This is working. heatsetpoint can be set with the REST API and a manually change is received. This value is in the config section of the device

temperature -> measured temperature
This is the actual measured temperature. This value is only set when adding the device and currently get never updated. This value is in the state section. 

When I manually lock oder unlock the device nothing is logged.

Smanar, 2022-09-28:
> This value is only set when adding the device and currently get never updated

Ok so this one is not updated every hour.

After 1 hour of working mode (after inclusion),  do you still have other report ? like

```
17:00:50:793 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000266020004000000e6
17:00:50:795 Tuya debug 5 : Status: 0 Transid: 2 Dp: 614 (0x02,0x66) Fn: 0 Data 230
```

If you don't make action on it, because it seem report are working if you change manualy the heatpoint.
And I don't see why a manual action on childlock don't make report if changing heatpoint is working, you are sure for battery ?

jplahl, 2022-09-28:
I think the device has a problem. I bought this device as a set with 2 thermostats and one gateway last week. Today I assigned the thermostat to the supplied gateway. After some minutes the gateway lost the connection to the device.

Batteries are fine, but I already replayed them to verify. Tomorrow I will try the second thermostat of the set.

jplahl, 2022-10-18:
Now it took me longer to test different things. The devices are working fine. The reason why the devices with the Tuya Gateway were offline after a short time were blocked ports. Both the gateway and the app on the mobile phone require port activations because the gateway and the app are sending data to the Tuya cloud. Of course I don't like that at all.

So I again paired my devices to my conbee network.

Every hour the devices are sending data. As I understand it, various values are not yet implemented here.
`
19:00:53:568 Tuya : Schedule command
19:00:53:568 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000973000018000000000000000000000000151515151515150000000000
19:00:53:568 Tuya debug 5 : Status: 0 Transid: 9 Dp: 115 (0x00,0x73) Fn: 0 Data 0
19:00:53:568 Tuya : Unknow Schedule mode
19:00:53:738 Tuya : Schedule command
19:00:53:738 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 000d6e000018000000000000000000000000151416161515120000000000
19:00:53:738 Tuya debug 5 : Status: 0 Transid: 13 Dp: 110 (0x00,0x6E) Fn: 0 Data 0
19:00:53:738 Tuya : Unknow Schedule mode
19:00:53:965 Tuya : Schedule command
19:00:53:965 Tuya debug 4 : Address 0x04CD15FFFE100F22 Payload 001177000018000000000000000000000000000000000000000000000000
19:00:53:965 Tuya debug 5 : Status: 0 Transid: 17 Dp: 119 (0x00,0x77) Fn: 0 Data 0
19:00:53:965 Tuya : Unknow Schedule mode
`

`
18:30:08:653 Tuya debug 4 : Address 0x540F57FFFEBC56DF Payload 000266020004000000e3
18:30:08:653 Tuya debug 5 : Status: 0 Transid: 2 Dp: 614 (0x02,0x66) Fn: 0 Data 227
`

PastCoder, 2022-10-20:
Hi,
great to see that work is progressing for this device. Could you already estimate how long it may take until it will be included in a (beta) release?

jplahl, 2022-10-23:
I bought a second Conbee to trace the traffic between the HAMA device and gateway with zshark. I was able to add the support for lime protection (Kalkschutz) and the away function.

`Enable limeprotection:  0x01 0x82  data::01:01
Disable lime protection: 0x01 0x82 data::01:00
`
`away on:    0x01 0x6a data::01:01
away off:   0x01 0x6a data:01:00
`
Now I want to add the schedule/program function:
This is an example of the command:
`00:47: 6d:00:00:12: 43:07:01:2c:00:be:02:1c:00:a0:04:38:00:c8:05:28:00:a0
`

I did several configurations and change values and was able to decode:
Monday: 0x01
Tuesday: 0x02
Wednesday: 0x04
Thursday: 0x08
Friday: 0x10
Saturday: 0x20
Sunday: 0x40

          command           day(s)     schedule 1     schedule 2        schedule 3        schedule 4
00:47: 6d:00:   00:12:   43:07:	  01:2c:00:be:	02:1c:00:a0:	04:38:00:c8:	05:28:00:a0

When I try to send a command of the already implemented code I can see entries in the debug log but nothing in the wireshark trace:
`17:48:09:633 Tuya data: 3430393839303430313230643031363038
17:48:09:634 Send Tuya request 0x04CD15FFFE100F22 : Dp_type: 0x00, Dp_identifier 0x70, data: 3430393839303430313230643031363038
17:48:09:690 0x04CD15FFFE100F22 error APSDE-DATA.confirm: 0xD0 on task`

Smanar, 2022-11-18:
Old topic ^^.

I have give up for the moment, waiting for the tuya support using DDF.

>@Smanar I've compared zigbee2mqtt time syncing code with deconz and noticed they use UTC time as relative number of seconds since 01-01-2000, and we use UNIX_EPOCH. There is J2000_EPOCH defined and seems to also be implemented, but you have changed it [here ](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/41745353eb8548079c02be417aa84e78d35dd36c#diff-e26cc016feb71ae607b1eecfd627df205b6faa96793f0c7cf676dd55f5ec4b51R930). Do you recall, why?

I can try to find the original issue, but it's from an user after snifff on the original gateway, but from my memory it's device dependent, it's for the that both are definied in the code UNIX_EPOCH and J2000_EPOCH.

>with UNIX_EPOCH only the hour is wrong

It's not because the winter/summer hour ? IDK if it exist on your country ?

>At time.cpp there is a link to tuya docs

In time.cpp ?

>PS. What is wrong with the PR for broader TRV support, it is in the queue for months?

Yeah :( Idk how many users are waiting for it ... The good news is the next step in deconz will be TRV, so I realy hope for this PR. The DDF core will be improved too, so perhaps we will be able to use it for them.

>Can you help me in in any way with this? :D

I m looking the device cluster list, are you sure this device use the tuya time sync ? He have too the time cluster.
Z2M is using the tuya synch and with J2000_EPOCH

mrblur, 2022-11-18:
> I can try to find the original issue, but it's from an user after snifff on the original gateway, but from my memory it's device dependent, it's for the that both are definied in the code UNIX_EPOCH and J2000_EPOCH.

Yeah, now I figured out how to build, update and force time sync, I played a bit yesterday.
With J2000_EPOCH also the day changes (rolls back by 1), with UNIX_EPOCH only the hour has 8 hours added.
> It's not because the winter/summer hour ? IDK if it exist on your country ?

It does, but I'm at UTC+1/2 (depending on dst) and the diff is 8 hours from local. Unsuprisingly, that is China's offset to UTC, I'm starting to believe this has firmware borked and hardcodes something :/

> 
> > At time.cpp there is a link to tuya docs
> 
> In time.cpp ? You mean tuya .cpp ?

Yeah, It was late yesterday :sweat_smile: 

> I m looking the device cluster list, are you sure this device use the tuya time sync ? He have too the time cluster. Z2M is using the tuya synch and effectively with J2000_EPOCH for this device

Well, I did change the code, resynced the time and got different results, so I'm saying: yeah, it uses tuya time sync :)
Could it be it gets confused on getting time synced two times? :thinking: 

What got me more confused is that switching to J2000 made matter worse on my device as it also skewed the day of week :rofl: 

I only have a single controller dongle, so it is not easy to turn off all the lights at home for playing with z2m again, I'll try to confirm it works/or not with z2m though :)

Meanwhile waiting for that DDF support :+1:

Smanar, 2022-11-18:
I have put a JS code you can try on JS script online.

Try to compare result with the one in deconz log ?

mrblur, 2022-11-18:
I need to run the code on the device itself, I don't trust it anymore ;)

The docs you provided say:

> Data | 8 | The time value is 8 bytes. It takes the format of 4-byte Unix timestamp + 4-byte local timestamp.
(...)
The Unix timestamp tracks time as a running count of seconds. The count begins at the Unix epoch on January 1st, 1970 at 00:00:00, so a Unix timestamp is the total seconds between any given date and the Unix epoch.
The local timestamp is calculated by adding the time zone offset and DST to the Unix timestamp.

That's exactly what getTime on tuya.cpp does.

Smanar, 2022-11-18:
Yep, it's on the first TRV we have added "MRT 200" or something like that, an user have sniffed packet and we have reproduced it.

But it seem it's device specific, and some devices can use different values, and probably this one ...

closed 2022-12-18 by github-actions[bot]

cagnulein, 2023-05-04:
any update on this device? I just bought it :(

Mimiix, 2023-05-04:
@Smanar ?

Smanar, 2023-05-04:
Can make a try with this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5844#issuecomment-1535190225

It seem both device use same DPID, except your device don't need the tuya unlock sequence.

cagnulein, 2023-05-04:
sure @Smanar , thanks! do you have a link of a guide that can i follow to use this DPID? sorry for the n00b question :(

cagnulein, 2023-05-05:
ok I found this https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
Let me try to add your file now

cagnulein, 2023-05-05:
@Smanar it worked! should I create a PR adding also this device?

Smanar, 2023-05-05:
Sure you can.
Just use a better title than "Create _TZE200_TS0601_thermostat.json" ^^

cagnulein, 2023-05-05:
ok i can do it the next week

sonny608, 2023-06-09:
Hi. How am I able to control the head using MQTT Explorer? I tried "iobroker/deconz/0/Sensors/11/heatsetpoint"
"iobroker/deconz/0/Sensors/11/heatsetpoint/set"
"iobroker/deconz/0/Sensors/11/heatsetpoint/cmnd"
"iobroker/deconz/0/Sensors/11/heatsetpoint/command" with no effect. Does anyone know the commands used by MQTT to change the temperature, turn on/off the device?

Smanar, 2023-06-09:
Deconz don't use MQTT, but a native REST protocol.

You need to send a body {"heatsetpoint":12} at the endpoint http://IP:PORT/api/API_KEY/sensors/ID/state

sonny608, 2023-06-09:
I followed your advice, but unfortunately it still didn't do anything
[image]
[image]
[image]

Smanar, 2023-06-10:
Ops, sorry my bad, it's not "state", but "config"

sonny608, 2023-06-10:
Changing from "state" to "config" helped with the on/off change. Changing the temperature via {"heatsetpoint":12} does not work yet.

Smanar, 2023-06-11:
You still have the error message "parameter not available" ?
Try with 1200 instead of 12

sonny608, 2023-06-11:
its working, thank you :) 👍