dresden-elektronik/deconz-rest-plugin #6270: Moes vibration sensor _TZ3000_bmfw9ykl

jleeca, 2022-08-17:
## Device

- Product name: Moes vibration sensor
- Manufacturer: _TZ3000_bmfw9ykl
- Model identifier: TS0210
- Device type : 
  - Sensor

## Screenshots

[image]
[image]

### Basic
[image]

### IAS Zone
[image]

### Power Configuration
[image]

Smanar, 2022-08-17:
Hello, can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_bmfw9ykl",
  "modelid": "TS0210",
  "vendor": "Tuya",
  "product": "Moes vibration sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHAVibration",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/tampered",
          "awake": true
        },
        {
          "name": "state/vibration",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

jleeca, 2022-08-17:
I just tried the DDF and found that the device is now correctly recognized and basically works fine.

However, I see once a vibration is detected it never get cleared. Is it possible to make the alarm auto clear after a timeout (e.g. after 10 seconds) or is this something that has to be handled by a application?

jleeca, 2022-08-17:
Upon a further research, it looks like DDF requires auto-reset but neither config/delay or config/duration worked when I tried as they only appears to be working for presence sensor but not for a vibration sensor.

Perhaps https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6165 is required to support auto-reset for this vibration sensor or it may necessary to change the device type to presence or occupancy sensor as a workaround although changing the sensor type is not ideal?

jleeca, 2022-08-18:
As I just wanted to make this work now than later, I'm using the following DDF which changes the type to ZHAPresence for making config/duration to work;
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_bmfw9ykl",
  "modelid": "TS0210",
  "vendor": "Tuya",
  "product": "Moes vibration sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "ZHAPresence",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/duration",
          "default": 1
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/presence",
          "awake": true,
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]

```
Although the device no longer appear as a vibration sensor but rather as a motion sensor, the state now gets reset automatically and the device is at least usable now.

Smanar, 2022-08-18:
Ha yessss, right.

In fact IAS sensor send a notification at every detection, it s not a on / off value by defaut.

dicer, 2023-02-15:
@jleeca Your DDF works fine for me as well. Thanks! Why was this closed though? Better to integrate it then not support the device at all, right?

jleeca, 2023-02-15:
Im not sure how the fix can be integrated or if it's already integrated but reopening for now as per the last comment from @dicer

Smanar, 2023-02-15:
Ha nope, the PR was not made.
Do you know how to make it ? I can make it if needed ?

But not sure what will be others devs reaction about the use of a presence sensor ^^

jleeca, 2023-02-15:
@Smanar, please go ahead to create a PR if you can. I have no idea how PR works.

Smanar, 2023-02-16:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6753

closed 2023-03-18 by github-actions[bot]