dresden-elektronik/deconz-rest-plugin #6271: Develco Products A/S	IOMZB-110

Patricen1, 2022-08-18:
## Device

- Product name: <html>
<body>

Develco Products IOMZB-110 IO Module
-- | --

</body>
</html>
- Manufacturer: Develco Products A/S
- Model identifier: IOZMB-110
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch
   - Other: 2 relay NO/NC outputs, 4 inputs with internal pull-up

## Screenshots

([url](url))
[image]
[image]

Mimiix, 2022-08-18:
Hi,

There's a lot of info missing and the template is not used correctly. Please create a new issue with all requested screenshots.

closing this one.

closed 2022-08-18 by Mimiix

zephyr2206, 2022-08-18:
Already requested here: [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5258](url)

Mimiix, 2022-08-18:
I'll re-open that one :)

zephyr2206, 2022-08-18:
Thanks :D