dresden-elektronik/deconz-rest-plugin #6272: Lonsonho Tuya Zigbee Smart US Plug 16A 20A Power Monitor _TZ3000_okaz9tjs

Drapiece, 2022-08-18:
## Device

- Product name: Lonsonho Tuya Zigbee Smart US Plug 16A 20A Power Monitor 
- Manufacturer: _TZ3000_okaz9tjs
- Model identifier: TS011F
- Device type :  
  - Switch

## Screenshots

[image]
[image]
[image]

[image]
[image]

This device has powermeter function in tuya, but
powermeter is not displayed.
It is simply being used as a switch.
Both 16A and 20A have the same configuration.

Hope to update.
Thank you.

Drapiece, 2022-08-18:
2f toiret washlet is testbed. :)

Drapiece, 2022-08-18:
> Hi,
> 
> CAn you please:
> 
> Create a request per device. Make sure to add all screenshot clusters.
> 
> Kind regards,

ok. I will upload all screenshot cluster tab

Drapiece, 2022-08-18:
> Hi,
> 
> CAn you please:
> 
> Create a request per device. Make sure to add all screenshot clusters.
> 
> Kind regards,

uploaded! please help me~

Mimiix, 2022-08-18:
Looks good!

@Smanar  Can you help with a DDF :D?

Drapiece, 2022-08-18:
> Looks good!
> 
> @Smanar Can you help with a DDF :D?

I'm so sorry. I'm beginner on github. sorry.
I did clean up context. 
thank you

Smanar, 2022-08-18:
Hello can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_okaz9tjs",
  "modelid": "TS011F",
  "product": "Smart US Plug 20A Power Monitor",
  "vendor": "Lonsonho",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {"at": "0x0000", "cl": "0x0702", "ep": 0, "fn": "zcl" },
          "parse": {"at": "0x0000", "cl": "0x0702", "ep": 0, "eval": "Item.val = Attr.val * 10"},
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Drapiece, 2022-08-18:
> Hello can try this DDF
>
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZ3000_okaz9tjs",
>   "modelid": "TS011F",
>   "product": "Smart US Plug 20A Power Monitor",
>   "vendor": [...]

Thank you! It works!
Consumption and now load power sensor activated!
[image]
Can you add amp sensor? 
thank you!

Smanar, 2022-08-19:
You mean the current ? It need to be, can you take a look direclty in the API ? using Phoscon / help / API Information / sensors.
You need to have it in the ZHAPower sensor.

Drapiece, 2022-08-19:
> 

oh!! thank you! I find it!
onemore thing. consumption is refreshed, but power is not refresh.
please edit ddf to refresh power.
It really helps. thank you!!

Smanar, 2022-08-20:
The ddf look fine for power too

```
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
```

Can you compare in deconz if the value is moving ? Cluster 0x0B04, attribute 0x050B (was 0 on your capture)
You can have convertion issue too, perhpaps need to *10 or /10.

Drapiece, 2022-08-20:
> 

[image]
[image]
power is not moving. consumption is moving.
what can I change ddf?
thank you

Smanar, 2022-08-20:
It's possible it s not working on the device, it stay at 0.

[image]

On this capture the value is 0 too, you are able to have another value ?

 Cluster 0x0B04 (ELECTRICAL MEASUREMENT) , attribute 0x050B

Drapiece, 2022-08-20:
> 

[image]
[image]
yea. it have another value :{
[image]
[image]
but, no change on HA

Smanar, 2022-08-20:
Ha yes right.
Can you take a look in deconz / debug view with enabling the flag "DDF" (not too much talkative) To see if you can find a a line about state/power ?

Drapiece, 2022-08-22:
thank you!
start move power!!!
It's almost at the end!!
I want setting 10 sec interval power, current, voltage, consumption.
but, I change interval and reboot, then 100 sec again.
is it impossible change to 10 sec?
really thank you!

Smanar, 2022-08-22:
You have them here

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
```

0x0505 is the voltage.
0x0508 is the current.
0x050B is the power.

min is the minium delay between 2 report.
max is the maximum delay.
change is the value change needed for a report.

Take care, decreasing too much delay is not good.

BTW what is the change on the DDF that make the power working ? (To make a PR wil the final DDF)

Drapiece, 2022-08-24:
Thank you for teaching!
really helpful!

add read and parse to power and current, then start moving.

This is final DDF from you

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_okaz9tjs",
  "modelid": "TS011F",
  "product": "Smart US Plug 20A Power Monitor",
  "vendor": "Lonsonho",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0B04"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "read": {"at": "0x0508", "cl": "0x0b04", "ep": 1, "fn": "zcl" },
          "parse": {"at": "0x0508", "cl": "0x0b04", "ep": 1, "eval": "Item.val = Attr.val"},
          "refresh.interval": 100
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {"at": "0x050b", "cl": "0x0b04", "ep": 1, "fn": "zcl" },
          "parse": {"at": "0x050b", "cl": "0x0b04", "ep": 1, "eval": "Item.val = Attr.val"},
          "refresh.interval": 100
        },
        {
          "name": "state/voltage",
          "read": {"at": "0x0505", "cl": "0x0b04", "ep": 1, "fn": "zcl" },
          "parse": {"at": "0x0505", "cl": "0x0b04", "ep": 1, "eval": "Item.val = Attr.val"},
          "refresh.interval": 100
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0051",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {"at": "0x0000", "cl": "0x0702", "ep": 0, "fn": "zcl" },
          "parse": {"at": "0x0000", "cl": "0x0702", "ep": 0, "eval": "Item.val = Attr.val * 10"},
          "refresh.interval": 10
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
`
Thank you so very much!

would you make me same brand 16A powermeter socket?
I modified and applied 20A.
It works in HA, but The ON/OFF display of the power switch is not normal.
(If turn it off, it will come back to on after a while)
DDF will be attached to the 20A request.

thank you!

Smanar, 2022-08-24:
So this DDF is finished ?

Wana make a PR with it or do you want I make it ?

For the issue on your second device, I realy have no clue, it s same device, with same DDF, I m seing on Z2M they are using the tuya unlock sequence but it s for enable missing field, nothing about this issue ...

Drapiece, 2022-08-25:
> So this DDF is finished ?
> 
> Wana make a PR with it or do you want I make it ?
> 
> For the issue on your second device, I realy have no clue, it s same device, with same DDF, I m seing on Z2M they are using the tuya unlock sequence but it s for enable missing field, nothing about this issue ... You had same issue before the DDF (when you have only on/off and no measurement) ?

thank you for your support!

Yes, this DDF is finisied!

I don't know mean PR.
I want this add device in release stable. (if can)

No measurement original before insert DDF.
normal switch, normal status.
after insert DDF, switch status is not normal.

Smanar, 2022-08-25:
PR are here https://github.com/dresden-elektronik/deconz-rest-plugin/pulls
It will be under the creator name (the owner).

As it just adding 1 file it not so hard. Just go in the good folder https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices then select "add file" you can use the same method to create a subfolder. As you don't have right on the official github, it will automaticallt create a fork under your name, create the file and prepare a PR

Drapiece, 2022-08-26:
Thank you for advice.

But, this is made from you.

I didn’t(couldn’t) made this.

Please register this DDF.

 

If you have time, Please check 16A DDF.

Thank you.

Smanar, 2022-08-26:
Have added the DDf on a previous PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

For the 16 A I have answered here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6272#issuecomment-1225936542

Can try the tuya unlock sequence, you need to read thoses attribute in the same order, on Cluster 0x0000

```
    Attribute: manufacturerName
    Attribute: zclVersion
    Attribute: appVersion
    Attribute: modelId
    Attribute: powerSource
    Attribute: 0xfffe
```

proggprod, 2022-09-05:
Hi.
Got this working today, but I can't seem to adjust the reporting  interval. 
What should I be adjusting? I thought it was the Max interval in the Bindings tab of the DDF editor -
[image]

I'm interested in the Watts in particular, but the update seems really slow. Maybe once a minute up to once every 4-5 mins.
I have a different plug (_TZ3000_cphmq0q7) that updates its Watts reading constantly, maybe every second or two.

If I click the 'read' button in the deCONZ UI it updates the value instantly.
[image]

Smanar, 2022-09-05:
It's possible the device (with the actual fimware) don't support reporting (It's something new on tuya plug), or the device have itself a minimum.

For the report :
- Min is the maximum time without report
- Max is the maximum
- Change is the difference value that trigger a report

You can check in logs if it s a report or a poll

closed 2022-10-05 by github-actions[bot]