dresden-elektronik/deconz-rest-plugin #6276: Aqara multi sensor not refreshing values regularly anymore after update to deconz 2.17.01

centrofly, 2022-08-18:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: (26720700)
 - deCONZ version: (2.17.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

SwoopX, 2022-08-18:
This is not a bug, just to be clear on that right away 🙂 

I assume you upgraded from a version prior to 2.15.0? This PR is intended to bring back what you interpret as bug https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6268

centrofly, 2022-08-18:
Thanks for the info, but I must admit I do not fully understand what you described in the PR.

Why would you not classify the behavior I found as a bug, if the quality of data received is not as good (up to date) as before?

SwoopX, 2022-08-18:
Well, what the device does is sending the reports for the 3 values in a non-configurable way, more or less zigbee compliant, in more randomly chosen intervals. Deconz receives and processes them, so everything works as expected.

Now Xiaomi has some, not exactly zigbee compliant "special" reports, which are sent in a roughly 50 mins interval, content-wise different for every device. Those are what you perceive as reduced readings.

With the new code base, it had to be chosen what will be taken to give you updated values, which was the regular reports, as they occur more often (and you cannot have both). The PR is intended to bring back leveraging the special report in addition.

LeoeLeoeL, 2022-08-19:
I can say, since Aqara TVOC DDF went to GOLD (the latest beta) air quality entity updates often while humidity not.
In previous release it was the opposite.

tamtam1111, 2022-08-22:
> This is not a bug, just to be clear on that right away 🙂
> 
> I assume you upgraded from a version prior to 2.15.0? This PR is intended to bring back what you interpret as bug #6268

The "fix“ would be very welcome

MaxCere, 2022-08-29:
Any workaround or ETA on when we will see the changes here? https://github.com/deconz-community/deconz-docker

MaxCere, 2022-08-30:
> Any workaround

Rolled back to 2.14.01 as a temporary workaround works, waiting for the improvement for a stable solution.

Hakko31, 2022-08-30:
I have the same error pattern. The sensors no longer update for hours.

MaxCere, 2022-09-16:
I upgraded to 2.18.1 but still, sensors are refreshing less frequently.

Smanar, 2022-09-16:
```
Expected behavior

Values should at least update once every hour, or when temperature changes 0.5 degree Celsius
```

what is your reporting interval ?

MaxCere, 2022-09-16:
It's 0.5 degrees like you said but on 2.14.01 and using other Zigbee dongle updates are much more frequent....how can you tell that this is expected behavior? Take a look at the below graph, flat line is with 2.18.01 and the more frequent is with 2.14.01

User experience it's a lot better with more updates, I will migrate to another dongle if it remains the expected behavior.

[image]

Smanar, 2022-09-16:
Your device is the "lumi.weather" ?
Strange I haven't the same result on my side, I have 0.1 °C variation.

MaxCere, 2022-09-16:
Device is "lumi.sensor_ht", didn't touch any DDF. 

I will try again run 2.18.01 to avoid my errors

Smanar, 2022-09-17:
Ok So I don't known at all this device but from the Z2M project it seem this device use the both cluster for temperature report, the temperature one and the Xiaomi one.

The second one is broken or WIP https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6268

Can you try like in this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6309

With replacing 

```
        {
          "name": "state/temperature",
          "public": false
        },
```

by

```
        {
          "name": "state/temperature",
          "awake": true,
          "parse": {
            "at": "0xff01",
            "ep": 1,
            "eval": "Item.val = Attr.val * 100",
            "fn": "xiaomi:special",
            "idx": "0x03"
          }
        },
```

grandyann, 2022-09-27:
I've got the exact same behavior here with a lumi.sensor_ht:
- it was sending frequent updates (more or less every hour or 0.5° change) before I updated to 2.17.01
- Not better once updated to 2.18.01
- I tried the change suggested above ( https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6276#issuecomment-1250062618 ) but it didn't improve anything

Smanar, 2022-09-27:
If I m right the "idx" to try for light level is 0x0b not 0x03 like on the sample.

grandyann, 2022-09-28:
Note that my device is a xiaomi temperature and humidity sensor (not a light).

So I tried with 0x03, and also 0x0b, but nothing did really work.
Note that I also tried with 0x02 because I found some info here: https://zigate.fr/xiaomi-capteur-temperature-humidite-clusters/ (sorry, in french), but I'm really not sure which value I should take.
Is there any place with the datasheet for each device ?

xavax59, 2022-09-29:
Hello, i've the same problem. To solve the problem, I change the status DDF xiaomi_wsdcgq11lm_temp_hum_press_sensor.json  (here /usr/share/deCONZ/devices/xiaomi/). I change Gold to Bronze

stigvi, 2022-09-29:
> Hello, i've the same problem. To solve the problem, I change the status DDF xiaomi_wsdcgq11lm_temp_hum_press_sensor.json (here /usr/share/deCONZ/devices/xiaomi/). I change Gold to Bronze

Yes, this is what I did, too. IMO, these sensors should still be in a state other than gold "officially"

Smanar, 2022-09-29:
Ha right, you can disable the "Gold" status.
Better to correct the report, but yes you are right, better to set the status "Bronze" than "Gold" officialy.

glm-ypinczon, 2022-09-30:
Thx a lot for the tip @stigvi , it made a huge difference, and now everything seems to work the way it used to before updating 👍

grandyann, 2022-09-30:
Thx a lot for the tip @xavax59 and @stigvi , it made a huge difference, and now everything seems to work the way it used to before updating 👍

BabaIsYou, 2022-10-04:
> IMO, these sensors should still be in a state other than gold "officially"

Agree, more and more often the DFF seems published in "Gold" without having been completely validated, for convenience. I think they should be released in "Silver" when first included in a release, then set to "Gold" on the next release if no issue has been reported.

Smanar, 2022-10-04:
Right, but I think you are the only one that doing that.
Even me I put always "Gold" out of laziness. But always for new devices, because the DDF can't broke an old one.

BabaIsYou, 2022-10-04:
> Right, but I think you are the only one that doing that. Even me I put always "Gold" out of laziness. 

lol ! I'm just following this documentation : [DDF levels](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-change-DDF-modes-and-enable-or-disable-DDF-levels)

> But always for new devices, because the DDF can't broke an old one.
Not quite sure about that ... The new DDF will superseed the old one or legacy code auto generated one.

Smanar, 2022-10-04:
>lol ! I'm just following this documentation

And I m sure no one can say you are wrong, but set the "Glod" state is easier and faster, the"least effort syndrome"

>Not quite sure about that ... The new DDF will superseed the old one or legacy code auto generated one.

It's new device, so there is no legacy code, no previous DDF. It can be a broken DDF or nothing ^^.
I have only /2 devices that are not in "Gold", but return are too much random for the moment ....

niawag, 2022-10-20:
Hi all, I'm having connection problems with a xiaomi sensor (Xiaomi Mi Smart Temperature and Humidity Sensor - v 20160516 in phoscon), I'm not sure my problem are related to this issue but it may be. The sensor is really close to the conbee2 (less than 2m with a soft wall between eachother).
How can I troubleshoot it ?
Should I try to change the DDF inside the docker container? Or should I get back to a prior deconz version, if so, which one?

centrofly, 2022-11-17:
> Thx a lot for the tip @xavax59 and @stigvi , it made a huge difference, and now everything seems to work the way it used to before updating 👍

@grandyann: What exactly did you change? I don not understand how/where to change DDF status from gold to bronze.

Smanar, 2022-11-17:
Hello, use a text editor to edit the file with this path. You will see in the file "Status" in the first 10 line.

rasssta, 2022-12-12:
Is there any work going on for this bug? Because we are many that consider it to be a bug... I thought some of my sensors were broken recently because of this issue. Applying the workaround did magic for me. 

[image]

Take a guess when I applied the workaround...

stigvi, 2022-12-12:
> Is there any work going on for this bug? Because we are many that consider it to be a bug... I thought some of my sensors were broken recently because of this issue. Applying the workaround did magic for me.
> 
> [image]
> 
> Take a guess when I applied the workaround...

I got tired of this and the Philips Hue Dimmer bug not fixed for months. So I decided to switch over to zigbee2mqtt with a Sonoff coordinator and that was a positive experience.

rasssta, 2022-12-12:
Yeah, I'm leaning towards doing the same (again, sigh). I moved away from Homey as I got tired of the crappy update interval/0.5c increments for my temp sensors, so if this is going to stay as a "feature" I'm off to the next one.

sceiler, 2022-12-27:
@rasssta can you please point me to a step-by-step guide or tell me yourself how to apply this workaround? I have got 2 different Aqara sensors with the same issue

rasssta, 2022-12-27:
> @rasssta can you please point me to a step-by-step guide or tell me yourself how to apply this workaround? I have got 2 different Aqara sensors with the same issue

You have one above: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6276#issuecomment-1261927464

sceiler, 2022-12-27:
Yeah I saw this but where do I find these JSON files? How do I apply it to my Conbee II/deConz?

stigvi, 2022-12-28:
> 

I have switched over to zigbee2mqtt and dont have deconz anymore, so I have to write this out of memory. But if you open or go into the user interface for deconz, you will find severeal small strips at the top or bottom of the sceen. That is more sections or screens at the VNC session. One of those small strips is the DDF editor.

So you start with clicking on the node in the network map so that the node is the active one. Then select something like "edit DDF" in the menu. Then you open the DDF editor with clicking on the smal strip at the top or bottom. Select silver for that DDF and save and finally hot reload.

If you have this as a docker in Home Assistant, you have to do this at every restart as the changes you do are reset when the docker is restarted.

sceiler, 2022-12-29:
> > 
> 
> I have switched over to zigbee2mqtt and dont have deconz anymore, so I have to write this out of memory. But if you open or go into the user interface for deconz, you will find several small strips at the top or bottom of the screen. That is more sections or screens on the VNC session. One of those small strips is the DDF editor.
> 
> So you start with clicking on the node in the network map so that the node is the active one. Then select something like "edit DDF" in the menu. Then you open the DDF editor with clicking on the smal strip at the top or bottom. Select silver for that DDF and save and finally hot reload.
> 
> If you have this as an addon in Home Assistant, you have to do this at every restart as the changes you do are reset when the docker is restarted.

thanks a lot. I got it working with your instructions. unfortunately it doesn't seem to help in my case. might be because it is a different sensor than the above mentioned or something else. I ordered a new Sonoff gateway and will replace my conbee ii with it. hopefully that will help long-term

Mimiix, 2022-12-29:
> thanks a lot. I got it working with your instructions. unfortunately it doesn't seem to help in my case. might be because it is a different sensor than the above mentioned or something else. I [...]

You can use the conbee with Z2m :)

sceiler, 2022-12-29:
tbh I was considering a lot if I should change at all with all the migration work required but after researching around the internet it seems the conbee ii is a bit outdated from a hardware perspective. even though it should not matter at all with my rather basic setup I need to / want to redo most of my zigbee stuff anyway so I might as well switch to the latest zigbee hardware available.

Smanar, 2022-12-29:
>it seems the conbee ii is a bit outdated from a hardware perspective

From memory ? The microcontroller speed ?

I still have a conbee 1

cputoaster, 2023-01-10:
Just a me-too comment: I cannot pair the lumi.weather sensors anymore with latest stable docker. And I don't know how to change it easily (just copying the json to the docker mounted devices folder and changing to bronze did not work, its probably still taking the one from the image). Maybe reverting it to Bronze for the next release would be a good intermediary fix?

Smanar, 2023-01-10:
There is soon a new deconz version, without all bugs on 2.19.X branch
To disable a DDF you can set in to "Bronze" yes, but you can too delete it.

But no, for me the solution is correcting the DDF.

closed 2023-02-09 by github-actions[bot]

deluxestyle, 2023-02-14:
Can this be opened again? I still face this issue

deluxestyle, 2023-02-15:
> Have you tried to disable the DDF ?

Means to set the DDF to Bronze?
Yes, but everytime a new version comes I have to do it again

Smanar, 2023-02-15:
>Yes, but everytime a new version comes I have to do it again

Ha ok, what is your OS ? you can put it in an "devices" user folder, but if the user folder is prioritary on the "deconz" folder I m not sure if it still same if the deconz one have an highter status.

pergolafabio, 2023-02-15:
> Perhaps https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues
> 
> DDF don't help for connection issue, only recognition. Can be the router if the sensor use one.

Hi, your reply is interesting... I have 3 multi sensors..

1 connected to conbee, the 2 others connected to Ikea plugs... When I do a cold boot, all sensors are connected.. hit after a while/hours, the 2 connected to Ikea are gone, not connecting anymore.... Untill I do a cold boot again...

What can I do to troubleshoot this? Running Deconz as addon in HA

Smanar, 2023-02-15:
Xiaomi / Aqara have this issue with some routers https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

But was fine with ikea ... A cold boot is a device reset for you ? or you reboot the host ?

You have take a look on the network picture on the GUI ?

pergolafabio, 2023-02-15:
Ok I have a look later, yes when I do a complete host reboot...

pergolafabio, 2023-02-16:
Had a look indeed, ths is how it shows now
accordint to this link, stuff should be missing, but in my case its all there...
the 2 multi sensors are just gone from network

[image]

Smanar, 2023-02-16:
Just for test can you try to put the sensor closer an ikea switch ?

pergolafabio, 2023-02-16:
ok, will do that...
I did a coldboot , and they are reporting back as you can see below
But still i dont see any lines of those 2 that are giving issues? What could explain that?

[image]
[image]

pergolafabio, 2023-02-16:
i also now noticed, the ones thats connected "multi_onder" seems it now also took an ikea as a repeater, while it was first direct connected to the conbee stick
And even the "multi_onder" that always worked , the one with the line , is the one with the biggest distance, even another floot
So i dont think distance is an issue
multi_boven, and multi_zolder are quite close to ikea repeaters

Smanar, 2023-02-16:
But what are you calling a "coldboot", a deconz restart ?
In phoscon you can select the device and look for "last seen" if it s > 1h, lot of chances for the device have DC (even they are not already marked as unrechable)

What is the model id of ikea switch ?

pergolafabio, 2023-02-16:
Ok, will have a look later...

I mean a cold boot, shutdown the host, no power...
After the coldboot, they are reporting back for a few hours...

Restarting the addon/deconz, doesn't 

Not sure if there is a difference in just restarting or power down/coldboot

Smanar, 2023-02-16:
Ok so realy something strange, I don't see how a cold boot can have a different reaction than a deconz restart, and on a sleepig device ...

The gateway is still powered if you just restart the Host, so to restart it you need to cut the power (or unplug/replug it) and some machine still power the USB port even on power off mode.

A xiaomi/aqara that have left the network never come back.

pergolafabio, 2023-02-16:
Ok will have a look later...
Yeah I'm aware of Xiaomi, once paired , they don't change route indeed...
It's indeed strange that only after cold boot they come back, I check again tomorrow and will provide the firmware info later

pergolafabio, 2023-02-16:
deconz: 20.20.01
gw: 26390500
conbee first model
below is the ikea plug info, allthough dont think its related anymore, since the working sensos is also connected to it... i tought at first it was connected to conbee stick itself

[image]

pergolafabio, 2023-02-16:
gonna repair them again, maybe it will solve the issue anway

Smanar, 2023-02-17:
IDK, I have take a look fastly on google, haven't see special issue about the ikea device ....

pergolafabio, 2023-02-17:
Yeah indeed , gonna buy some new batts and repair them again, thx for all info

[4 more replies]