dresden-elektronik/deconz-rest-plugin #6283: Device name

patvdleer, 2022-08-23:
## Device

- Product name: `ROB_200-029-0`
- Manufacturer: `ROBB smarrt`
- Model identifier: `ROB_200-029-0`
- Device type:
  - Other: Window Covering Controller (shutter)

Product link:
  - NL: https://www.robbshop.nl/robb-smarrt-zigbee-rolluikschakelaar
  - EN: https://www.robbshop.nl/en/robb-smarrt-zigbee-roller-shutter-switch 

Currently shows up as a dimmable light.

## Screenshots

[image]

### Basic

[image]

### Identify

[image]

### Alarms

N.a.

### Device Temperature

N.a.

### Groups

[image]

[image]

### Scenes

[image]

[image]

[image]

[image]

### On/Off

[image]

[image]

### Level Control

[image]

[image]

### Color Control

N.a.

### Simple Metering

N.a.

### Diagnostics

[image]

### Other clusters that are not mentioned above

### 0102 Window Covering

[image]
[image]
[image]

### 1000 ZLL commisioning

[image]

### F2 - 0021 Green power

[image]

Smanar, 2022-08-24:
Ha yes it s because your device have light cluster and covering cluster.
Easy to correct using DDF, 2 solutions, can make change yourself or share your actual DDF for I edit it.

In deconz, right clic, edit DDF, save the file and share it.

If you want to make the change yourself (I don't know your DDF so I can be wrong)

Remove state/bri, state/on
Chnage the type from "dimmable light" to `"type": "$TYPE_WINDOW_COVERING_DEVICE",`
Make hot reload in the menu

patvdleer, 2022-08-25:
[image]

I did get to remove the state/bri and state/on by I dont seem to be able to change the state.

Smanar, 2022-08-25:
Can you try with replacing your DDf by this one

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ROBB smarrt",
  "modelid": "ROB_200-029-0",
  "product": "ROB_200-029-0",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift"
        },
        {
          "name": "state/open"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Take care at path choosed by deconz, or you put it direclty in the DDF folder (in "devices)
If they are in the good folder, deconz select them automatically, else you probably need to load them using the editor.

patvdleer, 2022-08-25:
I placed it in the folder, `/usr/share/hassio/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices`,  next to the file created by deCONZ originally. I loaded it and the path is confirmed in the DDF preview. Hot reloaded, but there doesn't seem to change anything. Phoscon and HASS still see it as a dimmable light.

Smanar, 2022-08-25:
It's not my DDF, I don't use the "path" filed.

patvdleer, 2022-08-25:
I didn't remove or clean it, -> `Dimmable light 20`

---

I unpaired it again, fixed the files 
- `rob_200-029-0.json` (your version)
- `rob_200-029-0.orig.json` (original)

Added again -> `Dimmable light 21`

---

Deleted `rob_200-029-0.orig.json`, unpaired, -> `Dimmable light 22`

Smanar, 2022-08-26:
Ok.
What is your deconz version, on lastest version the legacy code is slowed down compared to DDF code.

Can you try to enable deconz log in deconz / help / debug view just with the flag "DDF" and make a hot reload ?

patvdleer, 2022-08-26:
Hass: 2022.8.6 
deconz: 2.17.1

[image]

Smanar, 2022-08-27:
And from logs we can't see the used type but the code create only lift and open, wo seem not so bad.
Can you check direclty in the API using phoscon / help / API Information / light / ID

I think the issue is because the legacy code that create a device json before the DDF core.

patvdleer, 2022-08-27:
I now can control it via Hass!

```json
{
    "etag": "c697919f10fe0b860ddf53ec1d7ac3c3",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-08-27T10:13Z",
    "manufacturername": "ROBB smarrt",
    "modelid": "ROB_200-029-0",
    "name": "Shutters 1",
    "state": {
        "lift": 100,
        "open": false,
        "reachable": true
    },
    "swversion": "2.5.3_r47",
    "type": "Window covering device",
    "uniqueid": "94:de:b8:ff:fe:8a:21:0d-01"
}
```

Phoscon settings view:
[image]

Phoscon regular view: (unable to control it via Phoscon)
[image]

Hass:
[image]

Smanar, 2022-08-27:
So it work now ?
What was the trick ? use "hot reload" ?

patvdleer, 2022-08-27:
I have no clue... I restarted home assistant, reloaded the addon not the new setting did how up.

Smanar, 2022-08-28:
Lol, ok, I will submit the PR as without it the device can never be reconised as covering.

closed 2022-08-28 by manup

patvdleer, 2022-08-29:
There seems to be one remaining issue, Home assistant disables the button to go down if they're fully up and vice versa. I can control them just fine via deconz vnc. If I let them go up or down just a bid and stop them I then can control them via HA.
Since I have no clue where this issue might originate I'm posting this issue here, please let me know if I should address this with the HA devs.

Shutter 1 and 2 are closed, shutter 3 is open.  

[image]

patvdleer, 2022-08-29:
@Smanar 

[image]

patvdleer, 2022-08-31:
Is this what you're looking for? 

The shutters are now fully open.

[image]

[image]

[image]

[image]

[image]

Smanar, 2022-08-31:
Nope the device json, the value returned by the api for it (and used by HA)
You can see them in phoscon / help / API Information / light / "device ID"

somethinh like that
```

{
    "manufacturername": "_TZE200_zah67ekd",
    "modelid": "TS0601",
    "state": {
        "lift": 0,
        "open": true
    },
    "swversion": null,
    "type": "Window covering device"
}
```

And I think thoses value are reversed for you.

patvdleer, 2022-08-31:
these values dont change, even after reloading, on and open stay on false.

# rest api

```json
                    {
    "etag": "e64177b6deddedc6ed820081b14ffca0",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-08-31T15:57Z",
    "manufacturername": "ROBB smarrt",
    "modelid": "ROB_200-029-0",
    "name": "Shutters 2",
    "state": {
        "bri": 127,
        "lift": 100,
        "on": false,
        "open": false,
        "reachable": true
    },
    "swversion": "2.5.3_r47",
    "type": "Window covering device",
    "uniqueid": "94:de:b8:ff:fe:8a:21:32-01"
}
```

# web socket 
```json
{
    "18:03:07:117": {
        "e": "changed",
        "id": "19",
        "r": "lights",
        "state": {
            "bri": 127,
            "lift": 0,
            "on": false,
            "open": null,
            "reachable": true
        },
        "t": "event",
        "uniqueid": "94:de:b8:ff:fe:8a:21:32-01"
    }
}
```

Smanar, 2022-08-31:
But there is something changed ? You don't use the DDF ? On your previous post you have that as json https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6283#issuecomment-1229165639

No state/bri and state/on (as they was removed on the DDF)

For a closed shutter you need to have
```
open = false
lift = 100%
```

and yes on your capture only lift is moving.

patvdleer, 2022-08-31:
ow my bad, I overlooked that one, here's a fresh websocket log from open to close. 

```json
{
    "21:14:37:8": {
        "e": "changed",
        "id": "19",
        "r": "lights",
        "state": {
            "bri": 127,
            "lift": 100,
            "on": false,
            "open": null,
            "reachable": true
        },
        "t": "event",
        "uniqueid": "94:de:b8:ff:fe:8a:21:32-01"
    },
    "21:14:37:83": {
        "e": "changed",
        "id": "19",
        "r": "lights",
        "state": {
            "bri": 127,
            "lift": 100,
            "on": false,
            "open": null,
            "reachable": true
        },
        "t": "event",
        "uniqueid": "94:de:b8:ff:fe:8a:21:32-01"
    },
    "21:15:19:512": {
        "attr": {
            "id": "19",
            "lastannounced": null,
            "lastseen": "2022-08-31T19:15Z",
            "manufacturername": "ROBB smarrt",
            "modelid": "ROB_200-029-0",
            "name": "Shutters 2",
            "swversion": "2.5.3_r47",
            "type": "Window covering device",
            "uniqueid": "94:de:b8:ff:fe:8a:21:32-01"
        },
        "e": "changed",
        "id": "19",
        "r": "lights",
        "t": "event",
        "uniqueid": "94:de:b8:ff:fe:8a:21:32-01"
    },
    "21:15:19:517": {
        "e": "changed",
        "id": "19",
        "r": "lights",
        "state": {
            "bri": 127,
            "lift": 0,
            "on": false,
            "open": null,
            "reachable": true
        },
        "t": "event",
        "uniqueid": "94:de:b8:ff:fe:8a:21:32-01"
    }
}
```

Smanar, 2022-09-01:
Ok so lift value seem good, you miss the state/open, I will add it on  the DDF
But I don't understand why you have bri and on ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ROBB smarrt",
  "modelid": "ROB_200-029-0",
  "product": "ROB_200-029-0",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = Attr.val < 100",
            "fn": "zcl"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

patvdleer, 2022-09-01:
> But I don't understand why you have bri and on ? 
Bri and on?

Sadly the new DDF doesn't fix it...

patvdleer, 2022-09-01:
I've updated the DDF to 

```json
{
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = Attr.val == 100",
            "fn": "zcl"
          }
```

Check the icons:

The woonkamer (livingroom) is half closed, the other 2 are open.

[image]

Smanar, 2022-09-02:
>Bri and on?

Yeah on a previous DDF you have 

```
        "state": {
            "bri": 127,
            "lift": 0,
            "on": false,
            "open": null,
            "reachable": true
        },
```

And now you have again just

```
        "state": {
            "lift": 0,
            "open": null,
            "reachable": true
        },
```

Ok so your device is reversed

>From fully open to fully closed
```
            "lift": 0,
            "open": true,
```

And you need to reverse both
So in the DDF you can use
```

        {
          "name": "state/lift",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = 100 - Attr.val;",
            "fn": "zcl"
          }
        },
        {
          "name": "state/open",
          "parse": {
            "at": "0x0008",
            "cl": "0x0102",
            "ep": 1,
            "eval": "Item.val = Attr.val > 0",
            "fn": "zcl"
          }
        },
```

patvdleer, 2022-09-02:
Awesome this fixes it!

Smanar, 2022-09-02:
So I update the previous PR with the new changes ?

patvdleer, 2022-09-02:
yes please

patvdleer, 2022-09-02:
Crap..... I found a bug, I can't set the position to 0 anymore, it sets it back to 100. I can still close it via the buttons but not the slider

Smanar, 2022-09-02:
So it work with value from 100 to 1, but not the 0 values ?
It was working before the last DDF modification ? Can you try "patching" only state/lift or state/open to check the guilty ?

patvdleer, 2022-09-19:
my apologies for the delay, have been swamped with work, will try to debug asap