dresden-elektronik/deconz-rest-plugin #6292: Door/Window Sensor ZD08Device name

matze20211, 2022-08-26:
## Device

- Product name: Door Window Sensor Model ZD08 (with anti-tamper button)
- Manufacturer: _TZ3000_7d8yme6f
- Model identifier: TS0203
- Device type : 
  - Sensor
  
## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2022-09-06:
Hello can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203"],
  "vendor": "Tuya",
  "product": "Smart Zigbee Door Window Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

matze20211, 2022-09-06:
Hi, thank you, it is mainly working, but the tamper button isn't recognized. and is shown with motion detector symbol in phoscon:
[image]

Smanar, 2022-09-07:
Oups I forget the tamper.
Can you add in the DDF

```
            {
               "name":"state/tampered"
            }
```

The second problem can come from phoscon, you can see the device json in phoscon / help / API information / sensors
Or in third app.

matze20211, 2022-09-08:
Super, now all is working fine.

Smanar, 2022-09-08:
Nice, thx, PR submitted https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6307

closed 2022-09-09 by manup