dresden-elektronik/deconz-rest-plugin #6299: Aqara Motion Sensors - P1(MS-S02) and RTCGQ11LM Lightlevel not updating

oneseventhree, 2022-09-05:
## Describe the bug

The Lightlevel only updates when motion is "detected"

## Steps to reproduce the behavior

## Expected behavior

Lightlevel used to update constantly even when motion was not detected

## Screenshots

## Environment

 - Host system: PC
 - Running method: Windows
 - Firmware version: 26660700
 - deCONZ version: 2.18.00 / 07/08/2022
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2022-09-05:
Thanks for the report.  I'll ask Manup to check! :)

SwoopX, 2022-09-05:
The P1 never worked differently and that's also described on their product page.

For the first gen, I cannot recall, but mine is working fine.

manup, 2022-09-05:
For the RTCGQ11LM this is how the device works, only updates during motion. Plus the light level is also reported once per hour in the special report command.

manup, 2022-09-05:
Ha, just had a look on the RTCGQ11LM DDF, I think we no longer catch the hourly light level report.
@SwoopX perhaps we should just attach special report handling to the `state/lightlevel` item and keep `state/lux` doing the Illuminance cluster handling?

oneseventhree, 2022-09-05:
> For the RTCGQ11LM this is how the device works, only updates during motion. Plus the light level is also reported once per hour in the special report command.

How about for the P1? :)

SwoopX, 2022-09-07:
> Ha, just had a look on the RTCGQ11LM DDF, I think we no longer catch the hourly light level report. @SwoopX perhaps we should just attach special report handling to the `state/lightlevel` item and keep `state/lux` doing the Illuminance cluster handling?

While this would work for the RTCGQ11LM, I must admit I'm not too much of a fan to that. Other device have more than one useful item from the special reporting. Also, there might be one significant caveat, which is triggering of rules. Any item getting its value via `R.item('XYZ').val` doesn't emit the required event and therefore, also cannot be used to complete state changes.

manup, 2022-09-07:
Indeed I think doing a `R.item('XYZ').val = 555` should also emit proper events :thinking: not sure yet how to do this but should be useful in general. "event all the things" :)

SwoopX, 2022-09-08:
> Indeed I think doing a `R.item('XYZ').val = 555` should also emit proper events 🤔 not sure yet how to do this but should be useful in general. "event all the things" :)

Hehe, exactly. If there's no event, it never happened 😂

Squashyware, 2022-10-05:
I am also not getting any updates to lux levels from the RTCGQ11LM sensor even with motion.
The value displayed on the sensor page remains static when motion is triggered.
If I can provide any info to help resolve this bug please let me know what you need

Nutti85, 2022-10-21:
I also have this problem, seemingly at random. Rebooting or re-adding the device fixes the issue for a few hours or days and it drops out again.

oneseventhree, 2022-10-21:
You should all update your firmware to 26780700. I had issues and this fixed it:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6436#issuecomment-1286823848

djsomi, 2022-11-06:
Hi Guys,

Same issue here, latest firmware on Deconz 26780700, latest firmware on Aqara Motion Sensor P1. 
Lux data received every 10 minutes, but it is incorrect. If I trigger a motion, correct lux data arrives immediately.
Same (or worse) issue is present sensors paired with Aqara hub, lux reading is waaay less often and again incorrect value till motion is triggered.

Hope this is some FW issue what Aqara will/can fix, lux reading is useless if this sensor operating this way.

BR
Zoltan

[image]
[image]

closed 2022-12-06 by github-actions[bot]