dresden-elektronik/deconz-rest-plugin #6315: Syntax error in DDF for HUE lightsensor R.item('state/lux').val after 2.16.1 (also present on 2.18.1)

olemr, 2022-09-10:
## Describe the bug
All versions after 2.16.1 (including yesterdays 2.18.1) have the Philips HUE Motion-sensor light-sensor Item broken.
None of dark/daylight/lux is longer updated.
Enabling the DDF log and hitting cluster Read produces this:

[image]

Seems like a syntax error:
```
Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000))
, err: SyntaxError: Syntax error
``` 
## Steps to reproduce the behavior
This error has been persistent for me since 2.16.1

## Expected behavior
No syntax error in code and the sensor to work.

## Screenshots
See description above

## Environment
[image]

- Hostsystem:
omr@shs2: uname -a
Linux shs2 4.15.0-66-generic 75-Ubuntu SMP Tue Oct 1 05:24:09 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
omr@shs2: cat /proc/cpuinfo
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 69
model name      : Intel(R) Celeron(R) 2957U @ 1.40GHz
stepping        : 1
microcode       : 0x26
cpu MHz         : 1396.831
cache size      : 2048 KB
physical id     : 0
siblings        : 2
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg cx16 xtpr pdcm pcid sse4_1 sse4_2 movbe popcnt tsc_deadline_timer xsave rdrand lahf_lm abm cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust erms invpcid xsaveopt dtherm arat pln pts md_clear flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs
bogomips        : 2793.53
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
Happens both under systemctl and GUI

 - Firmware version: (26xxyy00)
 26400500
 - deCONZ version: (2.xx.yy)
 2.18.1 and all version after 2.16.1
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 ConBee I
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 Yes, a powered USB HUB
 - Is there any other USB or serial devices connected to the host system? If so: Which?
 Yes, the HUB and a Z-Wave stick

## deCONZ Logs
```
Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000))
, err: SyntaxError: Syntax error
``` 

-->

## Additional context

Network size is ~113 devices, most powered routers
Mix of HUE, Ikea, Aqara, Lumi and Namron (Sunricher)
Also mentioned here: 
https://forum.phoscon.de/t/philips-hue-outdoor-light-sensor-broken-since-2-17-0-still-a-problem-on-2-18-1-ddf-syntax-error/2429?u=omr

ebaauw, 2022-09-10:
Is this the old indoor sensor, SML001, or the old outdoor sensor, SML002?

Are you sure you're using the DDF as distributed by deCONZ, and not some old local copy?

It works for me, just double-checked v2.18.1-beta with a new outdoor motion sensor (SML004):
```
13:47:54:200 00:17:88:01:0b:d1:08:d7-02-0400/state/lightlevel expression: /* global Attr, Item, R */

const v = Attr.val
const tholddark = R.item('config/tholddark').val
const tholdoffset = R.item('config/tholdoffset').val

Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000))
 --> 282
```

olemr, 2022-09-10:
_Is this the old indoor sensor, SML001, or the old outdoor sensor, SML002?_
Yes, have one of each, and see the same issue on both. Doesn't have any newer for reference.
[image]
[image]

_Are you sure you're using the DDF as distributed by deCONZ, and not some old local copy?_
Yes. The only DDF I have touched, was a Namron K8 a while back.

Here is the DDF setting:
[image]

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_PHILIPS",
  "modelid": "SML002",
  "vendor": "Philips",
  "product": "Hue outdoor motion sensor",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/philips/sml002_motion_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0x0406"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/alert",
          "default": "none"
        },
        {
          "name": "config/battery",
          "refresh.interval": 7300,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 2
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 2,
            "eval": "Item.val = Attr.val / 2"
          },
[... 485 lines cut ...]

```
16:10:06:492 failed to evaluate expression for 00:17:88:01:06:44:01:13-02-0400/state/lightlevel: /* global Attr, Item, R */

const v = Attr.val
const tholddark = R.item('config/tholddark').val
const tholdoffset = R.item('config/tholdoffset').val

Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000))
, err: SyntaxError: Syntax error
16:10:06:540 failed to evaluate expression for 00:17:88:01:06:44:01:13-02-0400/state/lightlevel: /* global Attr, Item, R */

const v = Attr.val
const tholddark = R.item('config/tholddark').val
const tholdoffset = R.item('config/tholdoffset').val

Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000))
, err: SyntaxError: Syntax error
```

ebaauw, 2022-09-10:
And what's the content of `/usr/share/deCONZ/devices/philips/0400_lightlevel.js`?  Does it end with a comma (`,`), or is that just the error message?

Otherwise I'm suspecting some issue with the Javascript environment on your system, like the linked issue.  However, I don't think a missing function would throw a SyntaxError

Smanar, 2022-09-10:
And if you put a ";" at the end of all lines ?

olemr, 2022-09-10:
";" did not help ...
```
17:07:48:327 failed to evaluate expression for 00:17:88:01:06:44:01:13-02-0400/state/lightlevel: /* global Attr, Item, R */

const v = Attr.val;
const tholddark = R.item('config/tholddark').val;
const tholdoffset = R.item('config/tholdoffset').val;

Item.val = v;
R.item('state/dark').val = v <= tholddark;
R.item('state/daylight').val = v > tholddark + tholdoffset;
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000));
, err: SyntaxError: Syntax error
```

olemr, 2022-09-10:
_Otherwise I'm suspecting some issue with the Javascript environment on your system, like the linked issue. However, I don't think a missing function would throw a SyntaxError_
But why does a downgrade to 2.16.1 makes it work again?
I have done a system apy-get update/upgrade today. Shouldn't that update the Javascript environment as well?

olemr, 2022-09-10:
I'm at a loss ...
Have tried putting the whole script inside { }, install nodejs on the host, adding ";" to each line ..
added a few consol.log() statements at the end and the syntax error always come after the last statement.
Even tried this from stack overflow, but I'm not sure where the console output goes ...:

```
18:32:29:066 	asdu: 185701000000219b6901000021010002000021feff
18:32:29:067 ZCL got data for node=0xE80F, cl=0x0400, at=0x0000, status=0x00, type=0x21
18:32:29:068 ZCL got data for node=0xE80F, cl=0x0400, at=0x0001, status=0x00, type=0x21
18:32:29:069 ZCL got data for node=0xE80F, cl=0x0400, at=0x0002, status=0x00, type=0x21
18:32:29:072 failed to evaluate expression for 00:17:88:01:06:44:01:13-02-0400/state/lightlevel: /* global Attr, Item, R */

<script>
  window.onerror = function (e) {
    console.log('Error: ', e);
  };
</script>

<script>
const v = Attr.val
const tholddark = R.item('config/tholddark').val
const tholdoffset = R.item('config/tholdoffset').val

Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000))
</script>
, err: SyntaxError: Syntax error
18:32:29:394 DEV add binding cluster: 0xFFFD,  0x000B57FFFE3148F1
18:32:29:395 timeout for response to Mgmt_Bind_req id 129 to 0xD0CF5EFFFE6F7048
18:32:29:396 APS-DATA.request id: 202, addrmode: 0x03, addr: 0x0017880108394845, profile: 0x0000, cluster: 0x0033, ep: 0x00 -> 0x00 queue: 13 len: 2 tx.options 0x00
18:32:29:397 	asdu (leng
```

Smanar, 2022-09-10:
>But why does a downgrade to 2.16.1 make it work again?

Good question, apart the math.log10, I don't see something news on this part ...

ebaauw, 2022-09-10:
> But why does a downgrade to 2.16.1 make it work again?

Because that didn't yet use a DDF for these sensors (or the DDF status was Draft).

ebaauw, 2022-09-10:
> But the only line using "," is

That is assuming a `,` in the Javascript is causing the SyntaxError, but it seems to be part of the error message instead.

> I have done a system apt-get update/upgrade today. Shouldn't that update the Javascript environment as well?

The Javascript engine is included in Qt.  Nothing to do with Java.

I don't think this issue is related to the issue with the Xiaomi sensor at all.  The Xiaomi reports `lux` and `lightlevel` needs to be computed from it, using `Math.log10()`.  This was missing in some Qt Javascript engines.  Note that the missing function resulted in a TypeError, not a SyntaxError:
```
Item.val = lux;, err: TypeError: Property 'log10' of object [object Math] is not a function
```
Also note that the `,` is part of the error message, not of the Javascript.

The Hue motion sensor reports `lightlevel`, and `lux` needs to be computed.  Note that the Javascript from the previous version also was reported not working, but we don't have an error log for that.
```
, err: SyntaxError: Syntax error
```
This seems to suggest that Javascript is choking on some non-printable stuff, or maybe it's choking an a premature end of file or end of input.  CR/NL differences?  A buffer too short, or not terminated correctly?

Note that the Javascript was changed again in v2.18.0, and v2.17.0 was already reported not working.  Unfortunately, we don't have the error log for that version of the Javascript, but I seriously doubt either Javascript file is at fault.  Again, the v2.18 file works for me in v2.18.1 for the SML004, and it did work when I tested the change for the SML001 and SML003.  I don't have the old outdoor sensor, but the DDF for the four different versions of the device are almost identical.

@olemr do you have any other devices which use a Javascript file from their DDF?  And do they work correctly.

olemr, 2022-09-10:
@Smanar tried with just 1 line of code:
```
20:42:12:389 failed to evaluate expression for 00:17:88:01:06:44:01:13-02-0400/state/lightlevel: const v = Attr.val;
, err: SyntaxError: Syntax error
```
Deleted everything in the file and wrote that one liner anew by hand.
Dumped the file:
```
omr@shs2:~$ hexdump -v  -e '/1  "%_ad#    "' -e '/1    "%02X hex"' -e '/1 " = %03i dec"' -e '/1 " = %03o oct"' -e '/1 " = _%c\_\n"'  /usr/share/deCONZ/devices/philips/0400_lightlevel.js
0#    63 hex = 099 dec = 143 oct = _c_
1#    6F hex = 111 dec = 157 oct = _o_
2#    6E hex = 110 dec = 156 oct = _n_
3#    73 hex = 115 dec = 163 oct = _s_
4#    74 hex = 116 dec = 164 oct = _t_
5#    20 hex = 032 dec = 040 oct = _ _
6#    76 hex = 118 dec = 166 oct = _v_
7#    20 hex = 032 dec = 040 oct = _ _
8#    3D hex = 061 dec = 075 oct = _=_
9#    20 hex = 032 dec = 040 oct = _ _
10#    41 hex = 065 dec = 101 oct = _A_
11#    74 hex = 116 dec = 164 oct = _t_
12#    74 hex = 116 dec = 164 oct = _t_
13#    72 hex = 114 dec = 162 oct = _r_
14#    2E hex = 046 dec = 056 oct = _._
15#    76 hex = 118 dec = 166 oct = _v_
16#    61 hex = 097 dec = 141 oct = _a_
17#    6C hex = 108 dec = 154 oct = _l_
18#    3B hex = 059 dec = 073 oct = _;_
19#    0A hex = 010 dec = 012 oct = _
_

```

@ebaauw will start hunting for other .js in my DDFs

olemr, 2022-09-10:
@ebaauw yes, seems like the Philips HUE dimmer switch calls the fc00_buttonevent.js script:
```
/* global Item, R, ZclFrame */

if (ZclFrame.at(2) === 0x00) {
  Item.val = ZclFrame.at(0) * 1000 + ZclFrame.at(4)
  R.item('state/eventduration').val = ZclFrame.at(7) << 8 | ZclFrame.at(6)
}
```
Log when pressing a button (down+release)
```
21:00:36:037 expression: /* global Item, R, ZclFrame */

if (ZclFrame.at(2) === 0x00) {
  Item.val = ZclFrame.at(0) * 1000 + ZclFrame.at(4)
  R.item('state/eventduration').val = ZclFrame.at(7) << 8 | ZclFrame.at(6)
}
 --> 0
21:00:36:407 expression: /* global Item, R, ZclFrame */

if (ZclFrame.at(2) === 0x00) {
  Item.val = ZclFrame.at(0) * 1000 + ZclFrame.at(4)
  R.item('state/eventduration').val = ZclFrame.at(7) << 8 | ZclFrame.at(6)
}
 --> 1
```

No errors there ...

ebaauw, 2022-09-10:
What if you don't use `const` for the light level script?  Just `Item.val = Attr.val;`?

olemr, 2022-09-10:
That did the trick!
```
21:08:04:707 00:17:88:01:06:44:01:13-02-0400/state/lightlevel expression: /* global Attr, Item, R */

v = Attr.val
tholddark = R.item('config/tholddark').val
tholdoffset = R.item('config/tholdoffset').val

Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000)) 
 --> 2
```

Wow, thank you! :-)

ebaauw, 2022-09-10:
So, a Javascript engine that throws a SyntaxError on `const`?!

Does it also throw the error the very first time after deCONZ is restarted?  Maybe subsequent script invocations re-use the same instance of the Javascript engine and the syntax error is thrown trying to re-define a constant?

olemr, 2022-09-10:
Went out and shone a flashlight at the sensor:
[image]

Working again :-)

olemr, 2022-09-10:
_Does it also throw the error the very first time after deCONZ is restarted? Maybe subsequent script invocations re-use the same instance of the Javascript engine and the syntax error is thrown trying to re-define a constant?_

I think the very first attempt is failing, because after each .js editing I have to restart deconz, and as soon as the GUI starts, I go right to the cluster, enable the DDF logging and hit Read. Really doubt any data from the sensor arrives before I do that.

olemr, 2022-09-10:
I'm a bit novice concerning javascripts, were does the engine reside? 
In the deconz package or as part of my Ubuntu system?

ebaauw, 2022-09-10:
As I said above, I think it's part of the Qt runtime.  I think this is a separate package, listed as a dependency of the deCONZ package, and installed automatically.  I don't know the details, and, frankly, I don't think I want to ;-).  We had some platform-specific bugs before, like, if I remember correctly `char` being unsigned on ARM but signed on Intel.  And we've seen a neutered `Math` implementation in Javascript (missing the `log10()`).

SwoopX, 2022-09-10:
Very remarkable and great that you were able to nail it down to `const`. I recall there's a different issue out here where the same error was observed.

Would be interesting to get as close as possbile to the Qt version(s) being troublesome, if that's possible at all. Btw, deconz uses Qjsengine in the background for all JS related stuff as far as I can tell, so this is Qt internal.

@olemr Could you share your Qt version? You can either use `apt show libqt5core5a` or `qmake --version`. My dev runs on x64, Ubuntu 20.04, Qt version 5.12.8, however, my prod is on Raspbian Buster, Qt version 5.11.3, and both run fine.

olemr, 2022-09-10:
@SwoopX 
```
Here you go ..
omr@shs2:~$ apt show -a libqt5core5a
Package: libqt5core5a
Version: 5.9.5+dfsg-0ubuntu2.6
Priority: optional
Section: libs
Source: qtbase-opensource-src
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 5998 kB
Provides: qtbase-abi-5-9-5
Depends: libc6 (>= 2.17), libdouble-conversion1 (>= 2.0.0), libgcc1 (>= 1:3.4), libglib2.0-0 (>= 2.22.0), libicu60 (>= 60.1-1~), libstdc++6 (>= 5), zlib1g (>= 1:1.1.4)
Recommends: qttranslations5-l10n
Suggests: libthai0
Breaks: libqt5scintilla2-12v5 (<< 2.9.2+dfsg-2~)
Homepage: http://qt-project.org/
Task: ubuntu-live, kubuntu-desktop, lubuntu-qt-core, lubuntu-qt-desktop, ubuntustudio-video, ubuntustudio-publishing, ubuntustudio-photography, ubuntustudio-graphics, ubuntustudio-audio, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop
Supported: 5y
Download-Size: 2035 kB
APT-Manual-Installed: no
APT-Sources: http://no.archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages
Description: Qt 5 core module
 Qt is a cross-platform C++ application framework. Qt's primary feature
 is its rich set of widgets that provide standard GUI functionality.
 .
 The QtCore module contains core non-GUI functionality.

Package: libqt5core5a
Version: 5.9.5+dfsg-0ubuntu1
Priority: optional
Section: libs
Source: qtbase-opensource-src
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Qt/KDE Maintainers <debian-qt-kde@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 5997 kB
Provides: qtbase-abi-5-9-5
Depends: libc6 (>= 2.17), libdouble-conversion1 (>= 2.0.0), libgcc1 (>= 1:3.4), libglib2.0-0 (>= 2.22.0), libicu60 (>= 60.1-1~), libstdc++6 (>= 5), zlib1g (>= 1:1.1.4)
Recommends: qttranslations5-l10n
Suggests: libthai0
Breaks: libqt5scintilla2-12v5 (<< 2.9.2+dfsg-2~)
Homepage: http://qt-project.org/
Task: ubuntu-live, kubuntu-desktop, lubuntu-qt-core, lubuntu-qt-desktop, ubuntustudio-video, ubuntustudio-publishing, ubuntustudio-photography, ubuntustudio-graphics, ubuntustudio-audio, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop
Supported: 5y
Download-Size: 2035 kB
APT-Sources: http://no.archive.ubuntu.com/ubuntu bionic/main amd64 Packages
Description: Qt 5 core module
 Qt is a cross-platform C++ application framework. Qt's primary feature
 is its rich set of widgets that provide standard GUI functionality.
 .
 The QtCore module contains core non-GUI functionality.

omr@shs2:~$ qmake --version
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu
omr@shs2:~$
```

ebaauw, 2022-09-11:
That looks like two installations, `5.9.5+dfsg-0ubuntu2.6` and `5.9.5+dfsg-0ubuntu1`.  That version is ancient; current version is 6; I think only 5.15.2 is in LTS.  That's what runs on my Raspberry Pi-s (both 32-bit and 64-bit).  I think the Javascript engine is in the `libqt5qml5` package.

Your `qmake` installation uses an even older version, 4.8.7.  Not sure that matters unless you compile the REST API plugin yourself.

Your Ubuntu version seems pretty old as well, current is 22.04 LTS.  You might need to upgrade that to get a newer version of Qt.

olemr, 2022-09-11:
_You might need to upgrade that to get a newer version of Qt._
Yes, it's time.. :-)
Have to do 20.04.0 LTS first, then 22.04.0 LTS ...

olemr, 2022-09-11:
On Ubuntu 22.04.1 jammy everything works fine even with const (as expected):
```
omr@shs2:~$ apt show -a libqt5core5a
Package: libqt5core5a
Version: 5.15.3+dfsg-2ubuntu0.1
```

```
18:53:54:251 00:17:88:01:06:44:01:13-02-0400/state/lightlevel expression: /* global Attr, Item, R */

const v = Attr.val
const tholddark = R.item('config/tholddark').val
const tholdoffset = R.item('config/tholdoffset').val

Item.val = v
R.item('state/dark').val = v <= tholddark
R.item('state/daylight').val = v > tholddark + tholdoffset
R.item('state/lux').val = Math.round(Math.pow(10, (Math.max(0, Math.min(v, 60001)) - 1) / 10000))
 --> 889

```

ebaauw, 2022-09-11:
Cool, I think this issue can be closed then.

@manup, @Mimiix the install guides for ConBee/ConBee II on phoscon.de might need to be updated.  Now they mention:
> Supported Ubuntu versions
> 
> ~~Ubuntu 16.04 LTS 64-Bit PC (AMD64)~~
> Ubuntu 18.04 LTS 64-Bit PC (AMD64)
> Ubuntu 20.04 LTS 64-Bit PC (AMD64)

Mimiix, 2022-09-12:
@manup or @Gautama please update those :)!

closed 2022-09-12 by Mimiix

olemr, 2022-09-13:
You might want to have a look at https://forum.phoscon.de/t/cannot-start-deconz-2-18-1-without-gui-on-ubuntu-22-04-0/2435/2?u=omr
before adding 22.04.0 to the list.