dresden-elektronik/deconz-rest-plugin #6318: MYUET ME167 - Tuya Thermostat - Clone Derivat

Miraculix666, 2022-09-10:
## Device

- Product name: MYUET ME167
- Manufacturer: _TZE200_bvu2wnxz
- Model identifier: TS0601

One of many places to order:
https://de.aliexpress.com/item/1005003952150617.html?spm=a2g0o.productlist.0.0.6036199bHV4yKu&algo_pvid=8f5681bd-f635-4e7a-9704-7c2681383b62&algo_exp_id=8f5681bd-f635-4e7a-9704-7c2681383b62-8&pdp_ext_f=%7B%22sku_id%22%3A%2212000027550116432%22%7D&pdp_npi=2%40dis%21EUR%2147.94%2128.29%21%21%21%21%21%402100bb4a16628399681366003e742e%2112000027550116432%21sea&curPageLogUid=76pPMqvayuTP

- Device type : 
  - Sensor
  - Thermostat

## photos will be added

##Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Debug log is following next days, there should be clones, but I don't know how to create a propper ddf.

THX for help

Smanar, 2022-09-12:
Hello, sorry, but for the moment it's not possible to add Tuya TRV using DDF.

Miraculix666, 2022-09-12:
> Hello, sorry, but for the moment it's not possible to add Tuya TRV using DDF.

:-(

Okay, what is the best way to go further?
It seems to be very similar to this one.

https://zigbee.blakadder.com/Hama_00176592.html
https://www.zigbee2mqtt.io/devices/SEA801-Zigbee_SEA802-Zigbee.html

Smanar, 2022-09-12:
There is a PR tu support them, but still in waiting list.

Miraculix666, 2022-09-13:
After this video, I am sure that it is a SEA802

https://youtu.be/0mE70FTBQG0

Since i only have one raspbee, I have to change my setup for the debugging,
My idea is to good up on straight deconz image, ein directly change my network channel, in that way I should only have that one device in my network that we investigate, is that correct?

What tests should I do to provide helpful output?

Is the above mentioned Thermostat supported by now?

If it is, it should be possible to add line to the device list and compile, correct? (I read something like this in another thread...)

I would try this.

Testing in this context should be possible with Rest client, I think.

Smanar, 2022-09-13:
The model mean nothing, there is new tuya clone every day.
This model is not supported by the legacy code, and not possible to add it using DDF yet.

Miraculix666, 2022-09-14:
:-(

Seemd to me from the other threads that basic functions are available.

Bad buy...

Is there anything that I can provide to get it up and running?

Smanar, 2022-09-14:
For the moment, need to wait for the core be in place, to add new devices.

Miraculix666, 2022-09-15:
> For the moment, need to wait for the core be in place, to add new devices.

Is there a est. timeline for this? If not, I need to buy also a tuya base (is there one device preferred?).

Maybe I could help then, similar to this procedure:

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_5-create-a-cloud-project

Smanar, 2022-09-15:
Lol, I m seing you are realy motivated, but the problem is this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054
On my side I m waiting for this PR is valided before adding more tuya TRV, because if something is not accepted need to remake all for all devices added.

Miraculix666, 2022-09-17:
> Lol, I m seing you are realy motivated, but the problem is this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6054
> On my side I m waiting for this PR is valided before adding more tuya TRV, because if something is not accepted need to remake all for all devices added.

Winter is coming,... 
And it's maybe getting (more) nasty in Europe this year.

I had to dump my old fs20 thermostat system. 

Ok,

I'm trying to getting an overview of this vast and crumbled theme. 
I also have another tuya thermostat that I could test (working fine till now, but I'm only using the basic functions. It is this one, labeled my Moe's https://zigbee.blakadder.com/Hysen_HY368.html)

It seems that with a little luck it could be done in the near future.
🤞

Would debug be helpfull and a Szenario as i described be good for that?

Miraculix666, 2022-10-05:
I couldn't wait any longer, got now a cheap tuya ZigBee hub for the thermostats. I also connected with my to tuya dev account. Just let me know if I can help out with some logs or something else...

GaWr26, 2022-10-14:
I also just got two of these so would be great if I could get them to work in Deconz

DoraDeu, 2022-12-03:
Could this topic be re-opened? This model has not included into the converter yet and there is still just an external converter as a workaround.

Stooovie, 2022-12-21:
Still nothing? The external converter doesn't work for me either

Smanar, 2022-12-21:
Nope sorry, still waiting for support for tuya TRV using DDF.

Miraculix666, 2023-01-12:
since there's no action I'll probably change my ZigBee system to ZHA or
z2m. Skyconnect is already delivered...

2023, 03:00:

closed 2023-02-10 by github-actions[bot]

Miraculix666, 2023-02-10:
Still pending....