dresden-elektronik/deconz-rest-plugin #6324: Tesla Smart Thermostatic Valve TV100 (_TZE200_husqqvux)

Asuanin, 2022-09-12:
## Device

- Product name: Tesla Smart Thermostatic Valve TV100
- Manufacturer: _TZE200_husqqvux
- Model identifier: TS0601
- Device type : Thermostat
- BlakAdder link: https://zigbee.blakadder.com/Tesla_Smart_TSL-TRV-TV01ZG.html

[image]

## Screenshots
[image]

[image]

[image]

Mimiix, 2022-09-12:
Can you please add all screenshots?

Asuanin, 2022-09-12:
I added screenshots for Groups, Scenes and OTAU

BabaIsYou, 2023-05-25:
Can try this DDF ?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_sur6q7ko", "_TZE200_hue3yfsn", "_TZE200_e9ba97vf","_TZE200_husqqvux"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Tuya TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 35, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 3]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 3 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

closed 2023-06-23 by github-actions[bot]