dresden-elektronik/deconz-rest-plugin #6325: Immax NEO Smart kontroler (L) V4 (_TZ3000_llfaquvp)

Asuanin, 2022-09-12:
## Device

- Product name: Immax NEO Smart kontroler (L) V4 2-Button
- Manufacturer: _TZ3000_llfaquvp
- Model identifier: TS0012
- Device type : Light (Dimmer)

[image]

## Screenshots

### Basic
[image]

[image]

[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]

### On/Off
[image]
[image]

Smanar, 2022-09-12:
Hello, can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_llfaquvp",
  "modelid": "TS0012",
  "vendor": "NEO,
  "product": "Immax NEO Smart kontroler",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Asuanin, 2022-09-12:
@Smanar 
Thanks, this works fine, i'll try it with a dimmable bulb tomorrow.

Asuanin, 2022-09-13:
Information about dimmable feature is from manufacturer (Immax)

Asuanin, 2022-09-13:
Ok, the manufacturer is lying, even the official app over Lidl gateway does not support dimming

Test OK, Smanar's DDF works

It can be closed

Smanar, 2022-09-13:
Ok , have added the device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

Asuanin, 2022-09-13:
Thank you and have a nice day

closed 2022-09-13 by Asuanin