dresden-elektronik/deconz-rest-plugin #6328: Support STEP_HUE command in COLOR_CONTROL cluster

dagroe, 2022-09-13:
Is it possible to add the `STEP_HUE = 0x02` command in the `COLOR_CONTROL` cluster? I have a device that sends these commands but since they are not listed in the `button_maps.schema.json`, I guess I cannot add them in my `button_map`, or is there a workaround?

Thanks!

## Device

custom device that sends `STEP_HUE` commands.

## Screenshots
[image]

### Basic

[image]

SwoopX, 2022-09-13:
I presume you're asking in a quite humorisic way to add support for your device (including the respective button map)? 🙂 Because you're writing that you want to have something included in the zigbee standard itself, which is already available and defined.

If so, please provide the device details as requested by the template and the commands emitted on the applicable button presses. Thanks!

dagroe, 2022-09-14:
Sorry, I think I didn't explain my problem well enough. I am trying to add support for my own device. I created a `button_maps` entry like this:

```
 "dgElectronicsDimmer": {
            "vendor": "DG Electronics",
            "doc": "DG Electronics Dimmer",
            "modelids": ["RT01"],
            "buttons": [
                {"S_BUTTON_1": "On/Off"},
                {"S_BUTTON_2": "Dim Up"},
                {"S_BUTTON_3": "Dim Down"},
                {"S_BUTTON_4": "Color Up"},
                {"S_BUTTON_5": "Color Down"}
            ],
            "map": [
                [1, "0x08", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],
                [1, "0x08", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim Up"],
                [1, "0x08", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim Down"],
                [1, "0x08", "COLOR_CONTROL", "STEP_HUE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Hue Up"],
                [1, "0x08", "COLOR_CONTROL", "STEP_HUE", "3", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Hue Down"]
            ]
        }
```

I was thinking that whatever I put in my `button_maps` entry would have to pass through schema validation via `button_maps.schema.json`. In `button_maps.schema.json` the command `STEP_HUE` is not defined as a `COLOR_CONTROL` command (even though it is part of the zigbee standard). The only commands defined are `[ "MOVE_HUE", "MOVE_TO_SATURATION", "MOVE_TO_COLOR", "MOVE_TO_COLOR_TEMPERATURE", "STOP_MOVE_STEP", "MOVE_COLOR_TEMPERATURE", "STEP_COLOR_TEMPERATURE" ]`. So I am asking if you can add the `STEP_HUE` command to the schema or can I just use the value `0x02` instead of `STEP_HUE` in my map?

Thanks!

SwoopX, 2022-09-14:
That makes more sense now, thanks. I can update the schema later on.

SwoopX, 2022-09-14:
I've created #6332.

Iirc, the schema is only used to evaluate PRs adding or changing button maps. When you want to do something locally, you need to add the command for the respective cluster at the very beginning of the button map file, allowing to use the command at will in the maps.

dagroe, 2022-09-21:
Thanks a lot!

closed 2022-09-21 by dagroe