dresden-elektronik/deconz-rest-plugin #6329: Edit Node Name not working

Ltek, 2022-09-13:
## Describe the bug
Cannot Edit Node name
Set node name, as shown in the user manual, does not work. 

## Steps to reproduce the behavior
See attached screenshot from user manual

## Expected behavior
can change the node name text 

## Screenshots
attached

## Environment
2.18.1

## deCONZ Logs

## Additional context

Mimiix, 2022-09-13:
Can you please fill in the template as intended?

Ltek, 2022-09-13:
Sorry, thought it was pretty straightforward. The docs show we can, and we cant. I dont think its dependent on OS or such. I'll update the first post

BabaIsYou, 2022-10-03:
> I dont think its dependent on OS or such

I can confirm that it doesn't work that way too on RPi with v2.15.2-beta

SwoopX, 2022-10-03:
Node names cannot be edited via deconz GUI, it's recommended to do it via REST API.

closed 2022-10-03 by SwoopX

Ltek, 2022-10-04:
@SwoopX  You have not provided a justification for not allowing Node Names to be edited when it can do EVERYTHING ELSE to a node. Its nonsensical to disallow a name change. 
According to @Smanar ...
"Deconz manage the zigbee network
The API plugin is just a part of deconz
You can have a device in deconz but not in the API"

.. thus deConz and the API are separate entities. Thus, deConz would be using the REST API to make the Node Name change, yes?

your answer was basically saying something like, "see this fantastic file system management app we have! You can copy files, add new ones, delete them, move them around, view and edit the contents... but you cannot change a file name. You'll need to use a different app for that. We dont want you to do 'everything' in one place... that would be too convenient."

Smanar, 2022-10-04:
Lol, it s more a "working issue".
In deconz you have only 1 node displayed, but in the API you can have 5/6 entries for the same device.
So there is problem wich one deconz need to use as name for the node name is deconz (and can make a new debate on this problem too)
And worse, If I change the node name is deconz ? Wich one entry in the API I need to rename too ?

Previously this feature was to change a setting in the device, but not supported by all devices, so was deprecated.

Ltek, 2022-10-04:
The ask is to edit "Name" (screenshot attached). Seems like about the most simple ask possible?
This field is used to name the device in Deconz's UI and JowiHue/Homseer - and I'd assume ALL other apps accessing the REST API.

[image]

Mimiix, 2022-10-04:
As the issue has been closed by a dev and a answer is given, proceeding the discussion doesn't have any use at this moment.