dresden-elektronik/deconz-rest-plugin #6330: Functional devices get hex names in deconz GUI

SwoopX, 2022-09-13:
## Describe the bug

Under not yet clear conditions, nodes do not get properly processed during the start of deconz and some might show up with a hex name. Indications where things go wrong can be seen in the debug output of function `handleDDFInitRequest()` as below (2 devices affected at the same time in my production):

```
20:59:11:265 DEV found DDF for 0x0000000000000000, path: /home/deconz/.local/share/dresden-elektronik/deCONZ/devices/sp_120.json
[...]
20:59:12:891 DEV found DDF for 0x0000002B00000002, path: /home/deconz/.local/share/dresden-elektronik/deCONZ/devices/170-33505_170-33605_smart_socket.json
```

Some analysis revealed that `event.deviceKey()` apparently gets, good question, corrupted? Note the change in the debug log below and also the change of the event ID.

Other potentially disruptive side effects are not clear yet.

## Steps to reproduce the behavior

This is not consistently reproducible. I had to restart my dev environment like 10 times for it to occur. Regardless, that condition manifests if one or more nodes do show a hex name out of a sudden after restart.

## Expected behavior

Supported devices being covered by DDF should get completely loaded and data processed via the intended code.

## Screenshots

2nd screenshot depicts how it is supposed to be

[image]

[image]

## Environment

 - Host system: irrelevant
 - Running method: irrelevant
 - Firmware version: irrelevant
 - deCONZ version: 2.18.01 and presumably many versions below
 - Device: irrelevant
 - Do you use an USB extension cable: irrelevant
 - Is there any other USB or serial devices connected to the host system? irrelevant

## deCONZ Logs

This is a more extensive debug log of mine. The interesting part is `22:28:35:827 EventEmitter - Event ID: 68, Node ext: 0x0015BC0036000C0D` in the middle of it (note the event ID gets updated).

```
22:28:35:817 handleDDFInitRequest - Event ID: , Event devicekey: 0x0015BC0036000C0D
22:28:35:817 DEV path: /usr/share/deCONZ/devices/frient/aqszb-110_voc_sensor.json, result value: -1
22:28:35:817 DEV_InitCompatNodeFromDescription
22:28:35:818 Item count: 12
22:28:35:818 Available items: attr/name
22:28:35:818 Available items: attr/manufacturername
22:28:35:818 Available items: attr/modelid
22:28:35:818 Available items: attr/type
22:28:35:818 Available items: attr/swversion
22:28:35:818 Available items: attr/id
22:28:35:818 Available items: attr/uniqueid
22:28:35:818 Available items: attr/lastannounced
22:28:35:818 Available items: attr/lastseen
22:28:35:818 Available items: config/on
22:28:35:818 Available items: config/reachable
22:28:35:819 Available items: state/lastupdated
22:28:35:819 attr/id
22:28:35:819 sub-device: 00:15:bc:00:36:00:0c:0d-26-fc03, has item: attr/id
22:28:35:819 attr/lastannounced
22:28:35:819 sub-device: 00:15:bc:00:36:00:0c:0d-26-fc03, has item: attr/lastannounced
22:28:35:819 SQL exec failed: SELECT attr FROM sensors WHERE uniqueid = '00:15:bc:00:36:00:0c:0d-26-fc03' AND deletedState = 'normal', error: no such column: attr (1)
22:28:35:819 attr/lastseen
22:28:35:819 sub-device: 00:15:bc:00:36:00:0c:0d-26-fc03, has item: attr/lastseen
22:28:35:819 attr/manufacturername
22:28:35:819 sub-device: 00:15:bc:00:36:00:0c:0d-26-fc03, has item: attr/manufacturername
[... 99 lines cut ...]
22:28:35:829 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, has item: attr/type
22:28:35:829 attr/uniqueid
22:28:35:829 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, has item: attr/uniqueid
22:28:35:829 config/battery
22:28:35:830 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, create item: config/battery
22:28:35:830 config/offset
22:28:35:830 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, create item: config/offset
22:28:35:830 config/on
22:28:35:830 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, has item: config/on
22:28:35:830 config/reachable
22:28:35:830 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, has item: config/reachable
22:28:35:830 state/humidity
22:28:35:830 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, create item: state/humidity
22:28:35:830 state/lastupdated
22:28:35:830 sub-device: 00:15:bc:00:36:00:0c:0d-26-0405, has item: state/lastupdated
22:28:35:830 DEV add binding cluster: 0x0001,  0x0015BC0036000C0D
22:28:35:830 DEV add binding cluster: 0x0402,  0x0015BC0036000C0D
22:28:35:831 DEV add binding cluster: 0x0405,  0x0015BC0036000C0D
22:28:35:831 DEV add binding cluster: 0xFC03,  0x0015BC0036000C0D
22:28:35:831 DEV found DDF for 0x0000000000000262, path: /usr/share/deCONZ/devices/frient/aqszb-110_voc_sensor.json
```

## Additional context

N/A

bee-con, 2022-09-14:
I had that with one Philips Hue motion sensor after upgrading to v2.18.1-beta.
In the end, I deleted the device and paired it again.

LeoeLeoeL, 2022-09-18:
I saw this behaviour only after a F5.
To see again the correct nema I have to restar deconz

closed 2022-09-19 by manup