dresden-elektronik/deconz-rest-plugin #6333: Device name

Idaho947, 2022-09-15:
- Product name: Zigbee tuya door sensor
- Manufacturer: _TZ3000_rgchmad8
- Model identifier: TS0203
- Device type : Sensor

## Screenshots

[image]

Basic
[image]
[image]

Power
[image]

Node info panel
[image]

Smanar, 2022-09-15:
Hello, can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_rgchmad8"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203", "TS0203"],
  "vendor": "Tuya",
  "product": "Smart Zigbee Door Window Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
           "name":"state/tampered"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Idaho947, 2022-09-16:
I try it and nothing change. I try a hot reload in gui but it is always invisible in phoscon and the api

Smanar, 2022-09-16:
> I try a hot reload in gui

If you have the GUI, when you have done the right clic and "edit DDF", it s the good file that have been choose ? You can see the path and the file in the windows title bar.

If you enable log with flag "info" when making "hot reload" you need to see something like
```
12:35:47:792 update ddf TS0601 index 1
12:35:47:794 Hot reload device: 84:fd:27:ff:fe:89:ae:21
12:35:47:902 DEV found DDF for 0x84FD27FFFE892A21, path: /home/pi/.local/share/dresden-elektronik/deCONZ/devices/dimmer_module.json

```

Idaho947, 2022-09-16:
yes when I select edit ddf the file is here :
[image]

but I can see anything in the log when I click on hot reload

Smanar, 2022-09-16:
On your capture, on the window editor title we can see "untitled file" mean the file selection haven't worked.

But if it work now ..... (bad planet alignement)

for the battery, on your capture, for the "power configuraton" values are not get, all values are 0, So I have choose the classic attribute, the 0x0021.
It s the same DDF used for other model 

>"manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_rgchmad8"],

So there is no reasons for your devices use different one, perhaps need some time to get the value.

Idaho947, 2022-09-16:
Yes now, after the reboot I've got the good name on the windows

Smanar, 2022-09-16:
Tell me if you still haven't the battery after some hours.

Idaho947, 2022-09-16:
Fully charged and still 0 since 4 hours

Smanar, 2022-09-16:
Ok so not normal.

Something to test, read the attribute value in deconz, It's the attribute 0x0021 in the cluster 0x0001.
If the value is updated when you read the value (and the value is > 0) it s a binding issue, else there is something wrong in the DDF
Remember you need to wake up the device in same time you read the attribute, with short press on the reset button for exemple.

As the binding don't seem so bad in the DDF, and if the issue is the binding, just re-include the device without deleting it, deconz remake all setting during the inclusion (as the binding)

Idaho947, 2022-09-17:
%Ok this morning everithing is fine, battery 98%.

Smanar, 2022-09-17:
Without change on the DDF ? So I can use this one for the PR ?

Idaho947, 2022-09-17:
Yes you can use this one.

Smanar, 2022-09-17:
Have edded the device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

closed 2022-09-20 by Idaho947