dresden-elektronik/deconz-rest-plugin #6336: IKEA TRADFRI motion E1745 sensor duration

dracula385, 2022-09-16:
## Describe the bug

When using the Ikea E1745 Motion sensor, and motion is detected after 60 sec sensor goes to OFF state until 90 sec cooldown period. That causes the sensor reported to HA goes constantly ON-OFFON-OFF when there is a constant motion because of extra 30 sec delay

## Expected behavior

If there is a constant motion there sensor should stay in ON state like in screenshot below.
Motion Sensor 1, Motion Sensor 2, Drax Room Motion Sensor are connected via Deconz.
Motion Sensor Test occupancy is connected via Zigbe2mqtt where there is a 90 sec duration set for the sensor to match the delay

## Screenshots

 [image]

## Environment

 - Host system: PC
 - IKEA E1745 Motion Sensor
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: (26720700)
 - deCONZ version: (2.17.07)
 - Device:ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Sonoff Zigbee 3.0 USB dongle

## deCONZ Logs

## Additional context

ebaauw, 2022-09-16:
Set `config.duration` to 0 to prevent deCONZ from clearing `state.presence` based on that setting, using `config.delay` as reported by the sensor instead.

Mimiix, 2022-09-17:
Hi,

As this is not a bug report as described in #5113 I am closing this issue. Feel free to open an topic on the forum.

Kind regards,
Dennis

closed 2022-09-17 by Mimiix

dracula385, 2022-09-17:
Can I just mention here you cannot set the duration to 0. I will open a topic of forum. Thanks