dresden-elektronik/deconz-rest-plugin #6340: Xiaomi Smart Plug EU loses power metering data every couple of minutes

exi, 2022-09-21:
## Describe the bug
I have setup my Xiaomi Smart Plug (EU) (lumi.plug.maeu01) but the power metering attributes
randomly appear and disappear from deconz.
I configured it in HA and the data looks like this:
[Screenshot 2022-09-21 at 16-15-06 History – Home Assistant](https://user-images.githubusercontent.com/449463/191528612-3eb311e3-ddb1-432b-9273-9507581df3d6.png)

In the deconz Interface, the power metering cluster also appears/disappears randomly.
## Steps to reproduce the behavior
1. Configure the Xiaomi Smart Plug
2. Wait

## Expected behavior
The power metering clusters should be available permanently.

## Screenshots
[Screenshot 2022-09-21 at 16-15-06 History – Home Assistant](https://user-images.githubusercontent.com/449463/191528612-3eb311e3-ddb1-432b-9273-9507581df3d6.png)

## Environment
 - Host system: PC
 - Running method: Arch with HA docker and deconz-community docker
 - Firmware version: 7.6.202226720700
 - deCONZ version: 2.17.00
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

exi, 2022-09-21:
To add to this, the other sensor readings are not looking much better:
[image]

exi, 2022-09-21:
Excerpt from the deconz logs:
```
20:31:47:076 GW firmware version is up to date: 0x26720700
20:31:52:712 0x54EF4410004EC253 extract Xiaomi special attribute 0x00F7
20:31:52:713 	64 on/off 1
20:31:52:713 	03 Device temperature 30 °C
20:31:52:713 	98 power 57.334000 (57)
20:31:52:713 	95 consumption 0.292420 (292)
20:31:52:713 	96 voltage 2280.000000 (228)
20:31:52:714 	97 current 251.464905 (251)
20:31:52:714 	05 RSSI dB (?) 2 (0x0002)
20:31:52:714 	9a unknown 0 (0x00)
20:31:52:714 	08 unknown 297 (0x0129)
20:31:52:714 	09 unknown 1024 (0x0400)
20:31:52:715 	0b unknown 0 (0x00)
20:31:52:715 	9b Consumer connected (yes/no) 1
20:31:52:715 	0a Parent NWK 0 (0x0000)
20:31:52:715 	0c unknown 1 (0x01)
20:31:52:716 ZCL attribute report 0x54EF4410004EC253 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F
20:31:53:576 SensorNode id: 22 (Consumption 22) no longer available
20:31:53:576 SensorNode id: 23 (Power 23) no longer available
20:31:54:577 SensorNode id: 18 (Consumption 18) no longer available
```

The "no longer available" message comes up every couple of seconds.

exi, 2022-09-22:
As a workaround, in the deconz interface, the plug exposes a writable attribute called `Reporting Interval`.
The default value is 300, but setting it to "10" seems to at least prevent deconz from "forgetting" that it existed.

[image]

SwoopX, 2022-09-23:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-smart-plugs-not-working-correctly and update your deconz version to the latest

exi, 2022-09-23:
Thank you. I upgraded deconz to 2.18.2 and the issue went away!
Did not have to firmware upgrade the device.

With the latest deconz version, I can set the reporting interval back to 300 and the measurements don't randomly go away.

closed 2022-09-23 by exi