dresden-elektronik/deconz-rest-plugin #6342: no more power consumption

sansonifabio, 2022-09-22:
## Describe the bug
Good morning Since I updated to version 2.18.02 / September 19, 2022, no more power consumption is transmitted from my sockets. There were no problems with version v2.17.1. Can someone check this please.

## Steps to reproduce the behavior
updated to version 2.18.02

## Expected behavior

## Screenshots

## Environment

 

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: 26780700
 - deCONZ version: 2.18.02 
 - Device:  RaspBee II
 

## deCONZ Logs

## Additional context

Mimiix, 2022-09-22:
Hi,

It's impossible for us to understand what sockets you mean. Either way, i see something with "lumi", assuming this mightbe related: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6340

sansonifabio, 2022-09-22:
Hi;
It's about these values ​​that are no longer updated, it seems as if these data points are virtually non-existent.
I use ioBroker with the Deconz adapter to read the data, which has always worked so far.
`[Plug](https://www.amazon.de/Xiaomi-Ferngesteuerte-elektronschen-Sprachassistent-Stromverbrauchsmessung/dp/B083F91ZX3)`
[image]

Mimiix, 2022-09-22:
Hi, 

CAn you check the basic cluster in deconz? What model ID / Manufacturere name are these? 
 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

sansonifabio, 2022-09-22:
Model Identifier lumi.plug.mmeu01

sansonifabio, 2022-09-22:
so ich habe noch einen Plug mit dem selben Model Identifier lumi.plug.mmeu01 da kommen Daten immer an. Habe jetzt einmal den Plug gelöscht und neu angelernt jetzt werden jede Menge Datenpunkte angelegt aber es dauert bis sich sich ändern. oder aktualisieren.

[image]

Mimiix, 2022-09-22:
Lets wait for a dev to check.

Can you please keep it to english only :)?

raoulx24, 2022-09-22:
Hi there,

I have the same problem, I did some investigations, I wanted to post also, but I found this thread, so I will provide here some info.

I have 5 devices, of type lumi.plug.mneu01. Out of 5, one is still providing info (sensors exist in API), the rest no.

While looking in the docker logs, I found the following relevant lines:
`02:40:27:728 DB skip loading sensor Power 24 24, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:728 DB skip loading sensor Consumption 25 25, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:729 DB skip loading sensor Power 2 2, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:729 DB skip loading sensor Consumption 3 3, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:729 DB skip loading sensor Power 34 34, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:729 DB skip loading sensor Consumption 35 35, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:729 DB skip loading sensor Power 22 22, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:730 DB skip loading sensor Consumption 23 23, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:730 DB skip loading sensor Power 31 31, handled by DDF Mijia smart plug ZNCZ04LM`
`02:40:27:730 DB skip loading sensor Consumption 32 32, handled by DDF Mijia smart plug ZNCZ04LM`

A bit down, we have the following lines
`02:40:27:904 skip legacy loading 04:cf:8c:df:3c:7e:3f:20-01 / lumi.plug.mmeu01`
`02:40:27:905 skip legacy loading 04:cf:8c:df:3c:7e:3f:20-01 / lumi.plug.mmeu01`
`02:40:27:905 skip legacy loading 04:cf:8c:df:3c:7e:3f:20-01 / lumi.plug.mmeu01`
`02:40:27:906 skip legacy loading 04:cf:8c:df:3c:7e:3f:20-01 / lumi.plug.mmeu01`
`02:40:27:906 skip legacy loading 04:cf:8c:df:3c:7e:3f:20-01 / lumi.plug.mmeu01`

It is the id (mac?) of the only device that is still providing info **and** it is 5 times. The rest of ids do not appear in this format.

Then, this one, an id for a non reporting device:
`02:40:28:090 DEV no DDF for 0x04CF8CDF3C7E52B3, modelId: lumi.plug.mmeu01`
`02:40:28:090 DEV create on-the-fly DDF for 0x04CF8CDF3C7E52B3`
or another one, an id for a non reporting device:
`02:40:28:122 DEV found DDF for 0x04CF8CDF3C7E41E6, path:`

And even down, the device is 0x04CF8CDF3C7E52B3 pooled:
`02:40:50:267 0x04CF8CDF3C7E52B3 extract Xiaomi special attribute 0x00F7`
`02:40:50:267 	64 on/off 1`
`02:40:50:268 	03 Device temperature 31 °C`
`02:40:50:268 	98 power 1.663000 (2)`
`02:40:50:268 	95 consumption 124.085373 (124085)`
`02:40:50:269 	96 voltage 2300.000000 (230)`
`02:40:50:269 	97 current 7.230434 (7)`
`02:40:50:269 	05 RSSI dB (?) 8 (0x0008)`
`02:40:50:270 	9a unknown 0 (0x00)`
`02:40:50:270 	08 unknown 278 (0x0116)`
`02:40:50:270 	07 unknown 0 (0x0000000000000000)`
`02:40:50:270 	09 unknown 3073 (0x0C01)`
`02:40:50:271 	0b unknown 0 (0x00)`
`02:40:50:271 	9b Consumer connected (yes/no) 1`
`02:40:50:271 ZCL attribute report 0x04CF8CDF3C7E52B3 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F`
`02:40:52:975 0x04CF8CDF3C7E5423 extract Xiaomi special attribute 0x00F7`
`02:40:52:975 	64 on/off 1`
`02:40:52:976 	03 Device temperature 30 °C`
`02:40:52:976 	98 power 1.716000 (2)`
`02:40:52:976 	95 consumption 230.796967 (230797)`
`02:40:52:977 	96 voltage 2230.000000 (223)`
`02:40:52:977 	97 current 7.695067 (8)`
`02:40:52:977 	05 RSSI dB (?) 8 (0x0008)`
`02:40:52:978 	9a unknown 0 (0x00)`
`02:40:52:978 	08 unknown 278 (0x0116)`
`02:40:52:978 	07 unknown 0 (0x0000000000000000)`
`02:40:52:979 	09 unknown 4358 (0x1106)`
`02:40:52:979 	0b unknown 0 (0x00)`
`02:40:52:979 	9b Consumer connected (yes/no) 1`
and
`03:10:32:296 0x04cf8cdf3c7e52b3 found group 0xFFF0`
and
`03:12:21:488 ZCL attribute report 0x04CF8CDF3C7E52B3 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F`
`03:12:25:506 binding for cluster 0x0006 of 0x04CF8CDF3C7E52B3 exists (verified by reporting)`
`03:12:25:508 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x04CF8CDF3C7E52B3 (seems to be active)`

For the other one, 
`02:41:48:080 0x04CF8CDF3C7E41E6 extract Xiaomi special attribute 0x00F7`
`02:41:48:081 	64 on/off 1`
`02:41:48:081 	03 Device temperature 28 °C`
`02:41:48:081 	98 power 15.906000 (16)`
`02:41:48:081 	95 consumption 811.153931 (811154)`
`02:41:48:081 	96 voltage 2250.000000 (225)`
`02:41:48:082 	97 current 70.693336 (71)`
`02:41:48:082 	05 RSSI dB (?) 8 (0x0008)`
`02:41:48:082 	9a unknown 0 (0x00)`
`02:41:48:082 	08 unknown 278 (0x0116)`
`02:41:48:082 	07 unknown 0 (0x0000000000000000)`
`02:41:48:082 	09 unknown 2816 (0x0B00)`
`02:41:48:082 	0b unknown 0 (0x00)`
`02:41:48:083 	9b Consumer connected (yes/no) 1`
`02:41:48:083 ZCL attribute report 0x04CF8CDF3C7E41E6 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F`
`02:41:49:507 binding for cluster 0x0006 of 0x04CF8CDF3C7E41E6 exists (verified by reporting)`
`02:41:49:509 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x04CF8CDF3C7E41E6 (seems to be active)`

If you need full logs, or any other info, I am ready to provide them.

Thank you for your time.

raoulx24, 2022-09-22:
Maybe it helps (although I think it is because of ddf), in VNC they look different:
[image]

sansonifabio, 2022-09-23:
Hello, I can confirm that's the case with me too, with the exception of one, the rest of the sockets no longer send anything via deconz.

SwoopX, 2022-09-23:
Please refer to https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-smart-plugs-not-working-correctly

Mask3007, 2022-09-24:
Well after updating I had some decimal issues in the home assistant addon. Other reported it here 

https://github.com/home-assistant/addons/issues/2679

I updated the firmware according to the link. It seems to fix the decimal issue form the other topic but added another. It seems that the power state is not temp. saved:
[image]
Our tumble dryer uses approx 600w fully laoded so the "peaks" seams to be plausible. But then immeadiately drops to 0w.

[image]

sansonifabio, 2022-09-24:
Hello, I'm getting too old for this. 
Dealing with it at 60 is complicated. No firmware is displayed for me, no matter which device I take. Resetting did not bring any real success. It takes forever until something changes, for example with the current consumption. Another problem is that new data points are now being created that do not have the original name, see the picture. So if I take the name of the washing machine, sensor values ​​are shown to me with other names. Now how am I supposed to find out what's what?

Grüße Fabio danke für die Hilfe

[image]

Mimiix, 2022-09-24:
@SwoopX is this a deconz implementation issue or a device bug / user issue.

erik-afry, 2022-09-24:
Well it happened in the new deCONZ version, exactly after update, and is fixed upon rollback of deCONZ, so it is a deCONZ implementation issue.

erik3rik, 2022-09-24:
Well it was introduced when updating deCONZ, and is fixed by rolling deCONZ back again. So it is a deCONZ implementation issue.

Mask3007, 2022-09-24:
Since my Problem only maybe relates to the same topic here and I am not sure if the Problem is from "deCONZ", "deCONZ-rest-plugin" or the "home assistant Integration" I created a forum post: https://forum.phoscon.de/t/aqara-powersocket-consumption-issues/2494

SwoopX, 2022-09-24:
> @SwoopX is this a deconz implementation issue or a device bug / user issue. If it's the latter, I'd rather close this and forward users to the forums.

Well, as it can be read in the original issue, the device seemed to be jumpy right from the beginning on what it does expose. No fixed pattern to reproduce that and quite some attempts to press it into one scheme to have it somehow work. I always suspected the firmware to have issues, though I never mentioned in the issue.

With the new device firmware available since a few months, all that jumping around has stopped as far as I can tell based on my tests and it is finally exposed in one fixed state.

Any plug now making trouble was never meant to be allowed to play with deconz in the first place and apparently the measures to block that were not as successful as they should have been.

marc965, 2022-09-25:
Hi all!

I can confirm that ALL Lumi Plugs (lumi.plug.mmeu01) dont provide Power and Consumption any more.
The ONLY thing that changes is updating Homeassistant addon deconz to version 6.16.0 (updating vom 6.15.0).
From this time this addon was updated i only get the power state visible in homeassistant (on(off).

Product: Conbee II
Version: 2.18.02 / 9/19/2022
deconz Firmware not changed with this update: version 26720700

Running on:
Home Assistant 2022.9.6
Supervisor 2022.09.1
Operating System 9.0
Frontend 20220907.2 - latest

Buat de deconz log is providing this:
11:46:37:682 0x04CF8CDF3C8A37D6 extract Xiaomi special attribute 0x00F7
11:46:37:682 	64 on/off 1
11:46:37:682 	03 Device temperature 27 °C
11:46:37:683 	98 power 8.819000 (9)
11:46:37:683 	95 consumption 31.422968 (31423)
11:46:37:683 	96 voltage 2310.000000 (231)
11:46:37:683 	97 current 38.177490 (38)
11:46:37:684 	05 RSSI dB (?) 2 (0x0002)
11:46:37:684 	9a unknown 0 (0x00)
11:46:37:684 	08 unknown 278 (0x0116)
11:46:37:684 	07 unknown 0 (0x0000000000000000)
11:46:37:685 	09 unknown 2816 (0x0B00)
11:46:37:685 	0b unknown 0 (0x00)
11:46:37:685 	9b Consumer connected (yes/no) 1

I dont understand why this values are not shown in homeassistant:
[image]

I think its perhaps a change of source code in the addon thats not being considered for the integration.

[config_entry-deconz-a198b989cee311b87b9589379e8b448b.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9640462/config_entry-deconz-a198b989cee311b87b9589379e8b448b.json.txt)

[config_entry-deconz-a19448b.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/9640463/config_entry-deconz-a19448b.json.txt)

Koal4, 2022-09-25:
Also have the same problem.
Updated deconz yesterday through Home Assistant and since then I don't get any meter sensors anymore. Only on/off works so definitely a bug that is caused by the deconz update.

How can I go back to the previous version btw?

marc965, 2022-09-25:
> Also have the same problem. Updated deconz yesterday through Home Assistant and since then I don't get any meter sensors anymore. Only on/off works so definitely a bug that is caused by the deconz update.
> 
> How can I go back to the previous version btw?

If you made a backup previously as recommended by the update process (defult activated) go to /settings/system/backup
[image]

Go to restore and then restart Homeassistant.
Thats all

Koal4, 2022-09-25:
> If you made a backup previously as recommended by the update process (defult activated) go to /settings/system/backup [image]
>
> Go to restore and then restart Homeassistant. Thats all

Thanks, I was looking at Deconz website but should have looked here lol.
Anyway I'm back on 2.17.01 / 7/1/2022 in Home Assistant and the meters are working again. Thanks!!

erik3rik, 2022-09-25:
Same here, rolling back solves the issue. I guess we'll have to wait for a new update before doing anything else.

raoulx24, 2022-09-25:
So, I did the FW update on only one device (just to test) and weird things started to happen. So, from my memory, it was something like:
- after the FW update, in VNC the device looked ok, but no sensors (re)appeared (consumption and power)
- after a deconz restart, the device disappeared from VNC, but one like this appeared (it is not the one, just example) - note the missing second dot:
[image]
- ok, I said, let's try another inclusion, so, add new lights, reset the device, it reappeared correctly (includin the name, it changed back to 'Bedroom plug etc', and only one of the sensors (I think the total consumption was missing)
- another removal, readd, deconz restart, now it lloks ok in VNC, but I do not have any sensors

Buuut... Somewhere on the way, another device started to report the Consumption, Power (sensors), although I did not upgrade the FW on it.

So, I give up and I ordered a CC2652P and I will switch to ZHA in HA (at least for the plugs in the beginning).

I saved the deconz docker logs; if they might help you, just let me know.

Cheers

ondrackajan, 2022-09-27:
Same issue with Xiaomi smart plugs. I have rolled back to 6.15.0 and the problem disappeared.

Mask3007, 2022-10-02:
@raoulx24 can you please confirm after you added the plug with the new FW works in ZFA? I am considering same thing...

Asche1985, 2022-10-02:
I have the same issue. Updated to 2.18.02 and my power meter shows 0 watt. :-(

manup, 2022-10-03:
Copy pasta from related: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6372

The wiki has a few remarks for the device in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-smart-plugs-not-working-correctly from what I understand the plug should work with the DDF when it's updated to the recent OTA firmware. The older Xiaomi firmware is a bit messy.

Side note: The following PR adds the "known issues" section with a link to this wiki in the DDF in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6386  to show up in http://127.0.0.1:8000/devices/xiaomi/xiaomi_zncz04lm_smart_plug_v24/

Mask3007, 2022-10-03:
Yeah as mentioned I did this FW Update and totally regret it. Since then I have only more issues with my aqara smart plug. Luckily It's Not only me, so hoping for a fix. 
Parallel ordered a sonoff zigbee stick from aliexpress either (soft) switching to ZHA or downgrading smart plug firmware. Before Firmware Update I only had 10x readings which just could be adjusted manually. 
https://forum.phoscon.de/t/aqara-powersocket-consumption-issues/2494/5

raoulx24, 2022-10-03:
@Mask3007, after a couple of hours in zigbee2mqtt (I decided to go the hardcore way), two plugs, one older FW and one latest look ok, so no need to 'downgrade' the FW is needed (and, btw, I do not think it is possible).

I will not start here to make a comparison of deconz and zigbee2mqtt, as I consider that both products are solid and I respect a lot the work done in deconz/phoscon/conbee (I am a devops manager in a cloud company, so I consider that I can have a pertinent opinion)... But, in this specific case, I think it is not handled properly here.

I also had some Aqara Power Plugs and I remember that they started to behave strangely in Consumption reporting, so I changed them with other ones.

As I general and final comment, I can understand the problems in zigbee and especially in deconz. The approval process seems to be way too permissive, there are so many manufacturers which implement the protocol more or less to the letter (some of them seems like they think that they invented the wheel or implement some strang things to be sure that you will use only their ecosystem - here I am looking to Tuya, mainly... At least compared with zwave (which I also use in my house), where all this problems really do not exist. Zigebee vs ZWave is like comparing Android world to Apple (and I use a Google Pixel device, so some might say that I am somewhere in the middle)

Cheers and good luck to all of you

nunoary, 2022-10-12:
I also upgraded my plugs and I also regretted it at first, but I managed to figure out the method to reset and re-pair the device and I'm happy with the upgrade. For me the problem is not with the deConz but with the plugs that require some patience to re-pair. I used the method for about 15 plugs and everything is working perfectly with DDF version 24 for these plugs. If anyone is still interested I can leave here the steps I did, it might help someone.

Mask3007, 2022-10-13:
@nunoary please tell us about it.

nunoary, 2022-10-13:
@Mask3007 Ofcourse.

After updating the sockets to version 24, these are the steps I took:

1st Temporarily disable third-party applications that may be used (HA, JowiHue, etc). I'm not sure if this step is really necessary, but that's how I did it. In my case I disabled the Homeseer JowiHue plugin.

2nd Unplug the plug from power source

3rd Delete the node in deCONZ GUI

4th Reconnect the plug to the power source but in a place that is close to the Zigbee Gateway (Conbee, Conbee II, Rasp...). In my case I used an extension to be able to put the plug right next to the gateway for when I start pairing

5º With the plug plugged in, open the "Control" panel in the deCONZ GUI and select the tab (I don't remember the name now but it's the second tab) and click on "Permit Join". Immediately start the plug reset (press the ON/OFF button until the blue light starts to flash). When the blue light flashes, release the button and press it from second to second (approximately) until pairing is complete (a good indication is when the "Permit Join" timing is over)

6º Check if the node in deCONZ has the correct name (something like "Smart Plug 45"). In my case the first pairing attempt never works until the second attempt.

7th In case everything doesn't go well the first time (name, endpoints etc...) repeat the steps from the 3rd.

8th If everything goes well, on the second try it will work. To confirm this is what I did: confirm the name, read node descriptor, read simple descriptor and confirmed that the plug was using version 24 DDF.

9th In case you don't get it in the first, second, third...attempts, don't despair. I found that if we unplug the power sockets and let it go for a few hours (I expected about six hours) without power and then I tried again and that was it.

10th In my case, as I use Homeseer, I had to delete the device in Homeseer for it to create a new one with the updated features (ON/OFF; Whatts; KW; V; A).

I apologize for my english, it's not my language but if you have any questions just let me know.

waywit, 2022-10-13:
Hi there, I have had also issue before(22) and after upgrade (24) of my plugs.
- Some of them showed up with firmware 0.0.0_0024 in Phoscon, some with 01-23-2022
- I figured out that all "0.0.0_0024"running correctly and the rest need to be repaired
- I figured it our with "delete" all unwanted, restarted deconz, and started to repair them
- Now all my plugs showing as "0.0.0_0024" Firmware in phoscon GUI :-)

But there is still a difference in deCONZ GUI between some plugs:

Some of them look exactly like the ones on the linked wiki page:
[image]

But some of them have additional cluster attributes listed (???):
[image]

Does anybody know if this might be okay or do they need to be paired again?

nunoary, 2022-10-13:
@waywit 

What is the version of Xiaomi 1 Plug when you read the "Basic" cluster. If it says version 24 you need to delete the node and re-pair again. If it says 22 then you need to upgrade the plug.

Mask3007, 2022-10-13:
I have the quite similar aqara version of the plug SP-EUC01 (lumi.plug.maeu01) with the newest from deCONZ recommended Version. I can see that the Version 41 (date 12-06-2021) is recognized. Now after a two week disconnect of that plug I got some of my issues solved but still having issues. Mainly related to this topic. The consumption sensor is still missing, at least it is in deconz, but in the rest api its only showing the switch and ONE Sensor. 
[image]

(I have currently only one ""modelid": "lumi.plug.maeu01" paired therefore searching for it was easy.

Result of rest API (power):
{
            "config": {
                "on": true,
                "reachable": true,
                "temperature": 2800
            },
            "ep": 21,
            "etag": ".........",
            "lastannounced": null,
            "lastseen": "2022-10-13T16:18Z",
            "manufacturername": "LUMI",
            "modelid": "lumi.plug.maeu01",
            "name": "Power 48",
            "state": {
                "lastupdated": "2022-10-13T16:18:51.684",
                "power": 0
            },
            "swversion": "12-06-2021",
            "type": "ZHAPower",
            "uniqueid": ".........."
        }

Result of rest API (switch):
{
            "etag": "...............",
            "hascolor": false,
            "lastannounced": null,
            "lastseen": "2022-10-13T16:18Z",
            "manufacturername": "LUMI",
            "modelid": "lumi.plug.maeu01",
            "name": "Aqara Trockner plug",
            "state": {
                "alert": "none",
                "on": true,
                "reachable": true
            },
            "swversion": "12-06-2021",
            "type": "Smart plug",
            "uniqueid": "...................."
        }

Also I think the update interval of the rest api seems to be a little bit delayed or updates are skipped (Maybe because I tried it only with a low powered phone charger). At this I am not sure.

I hate it that I have to repeatly repair devices. Last days I had issues with my aqara motion sensor not working. I had to repair it. It came out of nowhere. I have really no idea what is happening in deconz or the zigbee network...

nunoary, 2022-10-13:
@Mask3007 Believe me I know what you're saying. I've been through this myself and I understand your frustration, I've only recently thought seriously about switching to another system. But I decided to make one more attempt and finally managed to get my Xiaomi/Aqara devices working as expected. Most of my equipment is from this brand, I even have Chinese ("lumi.plug") plugs and these integration with deCONZ is even more difficult than the European ones. 

I also have some "lumi.plug.maeu01" with version 41 and they are working perfectly, KWh included.

I'm sure your problem was with the pairing. Delete the node and re-pair again until you are sure it is ok. As I said before, if you can't make the correct pairing on the third attempt, it's best to unplug the plug from power socket, wait a few hours and try again. With me on the second try it's fine.

Mask3007, 2022-10-13:
@nunoary I paired my plug now at least 15 times. One time I had the consumption sensor, but then other issues (Power dropping immediately to 0 again). And since I had so many errors I hat the plug laying around here for at least two weeks. I think that are more hours than necessary.
Well there are many users complaining having issues, you included as you mentioned you had some to repair multiple times... There in my opinion this is definately a bug and not a feature. I think the plug is not using the correct ddf but not sure why and also I don't know how to change it manually.

nunoary, 2022-10-13:
Two weeks without power is more than enough :)

I have 33 Xiaomi plugs (15 "lumi.plug" + 15 "lumi.plug.mmeu01" + 3 "lumi.plug.mmau01"). The first one must be five or six years old... they never worked well and last weekend when i discovered the method i deleted all (nodes in deCONZ GUI) one by one and re-included one by one, it took about 2 days and now i am very satisfied.

Try the following:

1- Deletes the node in deCONZ GUI;

2- Power on (with a socket or extendable socket) the plug and bring it close to the gateway (5 to 10 cm);

3- Wait a few seconds to confirm that another node was not automatically created in the deCONZ GUI when you plugged it in. If created it, delete it again;

4- In deCONZ GUI click on "Panels" and select "Control";

5- Click on the "Control" tab and then on the "Pairing" tab;

6- Click on "Enable permit join"; Note: I've had more success using deCONZ GUI than Phoscon

7- Starts the reset of the socket, that is, keep pressing the ON/OFF button until the blue light starts to flash;

8- Immediately start clicking the ON/OFF button with a spacing of 1 to 2 seconds and notice what happens in the deCONZ GUI, mainly in the node name and in the permit join timer. This process it takes a few seconds but you should only stop pressing the ON/OFF button a few seconds after the permit join timer expires. The timer stops when the process is finished;

9- Keep pressing the ON/OFF button and notice what happens to the socket light. On the first attempt, one or more of the following 3 situations will happen: the light flashes blue, the light flashes red or the name in deCONZ GUI it is of type 0x000...no problem, go to the next step;

10- Delete the new node in the deCONZ GUI and redo steps 6, 7 and 8;

11- When the timer ends and you confirm that the node has the correct name, like "Smart Plug 12" you continue to press the button every two seconds for about another 10 seconds. Notice that the light now turns blue when ON and turns off when OFF (normal operation). Leave the socket on and turn on a device to consume energy, in my case it was a mini dehumidifier that consumes 20W;

12- Immediately go to the deCONZ GUI and confirm the endpoints. The three that must appear are 0x01, 0x15 and 0x1F. If you only have 0x01 and 0x15 there is no problem step 14 will take care of that;

13- Right click on the node and click on "Edit DDF". At the top of the DDF page is the name of the file used. For "lumi.plug.mmau01" the name should be "xiaomi/xiaomi_sp-euc01_smart_plug.json". If the name is not showing and it is for example "Untitled file", no problem the next steps will solve that.

14- Right click on the node again and select "Read node descriptor" and then "Read simple descriptors". Then select the Basic cluster and click Read, do the same for the 000C clusters to read the W and KW values. Note: if you double-click on the 0x0055 Present Value line on the Cluster Info tab, you will be able to read the current value and configure the reporting times, but be warned that in the case of "lumi.plug.mmau01" it is not possible to configure it.

Mask3007, 2022-10-14:
@nunoary 

First, wow and thanks for the detailed instructions. Thanks for trying your best but...

I tried it now 5 times... First time it was unplugged since yesterday. I never got the new DDF to get used. Still having "Untitled File".  

immediately after my sonoff stick arrives I will try ZHA to eleminate a hardware issue...

waywit, 2022-10-14:
Thank you all for the support... I have re-paired and re-paired and re-paired all my non-standard plugs and now they have all Firmware "24" and look all the same in deCONZ GUI, like described in the wiki. The connections are all stable now and work very fine with home bridge now. The only thing I have mentioned ist an issue with OpenHAB... only "power" and "consumption" values are shown there, I actually miss "current" and "voltage", which are listed in the API, but this I will try to get solved in the openhab addons:

```
  "31": {
    "etag": "9e4beeb8148d544374c0685368b43689",
    "hascolor": false,
    "lastannounced": "2022-10-13T21:41:38Z",
    "lastseen": "2022-10-14T12:22Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.plug.mmeu01",
    "name": "Drucker",
    "state": {
      "alert": "none",
      "on": true,
      "reachable": true
    },
    "swversion": "0.0.0_0024",
    "type": "Smart plug",
    "uniqueid": "04:cf:8c:df:3c:7b:1e:a4-01"
  }

  "207": {
    "config": {
      "on": true,
      "reachable": true,
      "temperature": null
    },
    "ep": 21,
    "etag": "e63e68ae8b78ced631fab35a0879c481",
    "lastannounced": "2022-10-13T21:41:38Z",
    "lastseen": "2022-10-14T12:22Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.plug.mmeu01",
    "name": "Power 207",
    "state": {
      "current": 8,
      "lastupdated": "2022-10-14T12:22:27.723",
      "power": 2,
      "voltage": 220
    },
    "swversion": "0.0.0_0024",
    "type": "ZHAPower",
    "uniqueid": "04:cf:8c:df:3c:7b:1e:a4-15-000c"
  }

  "208": {
    "config": {
      "on": true,
      "reachable": true,
      "temperature": null
    },

    "ep": 31,
    "etag": "0357b2e46737a9c35fd1221842d46d91",
    "lastannounced": "2022-10-13T21:41:38Z",
    "lastseen": "2022-10-14T12:22Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.plug.mmeu01",
    "name": "Consumption 208",
    "state": {
      "consumption": 43,
      "lastupdated": "2022-10-14T12:18:23.219"
    },
    "swversion": "0.0.0_0024",
    "type": "ZHAConsumption",
    "uniqueid": "04:cf:8c:df:3c:7b:1e:a4-1f-000c"
  }

```
Every plug has exactly 2 additional sensors one power and one consumption sensors. Before the update some had up to six sensors and some of them did not get updates...

nunoary, 2022-10-14:
@Mask3007 

Are you sure you have the file in the ".../devices/xiaomi/" directory? If you have, try to select the DDF file manually.

On DDF editor:
- File 
- Open File 
- go to you directory ".../deCONZ/devices/xiaomi" and select the file "xiaomi_sp-euc01_smart_plug.json" for the "lumi.plug.maeu01" plug
- Finally click in "Hot Reload".

sieren, 2022-10-24:
Chiming in here that the process is indeed super painful right now.
Got it to work in the end, but don't ask me how. It took like 10 repairs until it finally went up the way it should.

sansonifabio, 2022-10-25:
Hello,

So I have now successfully updated the firmware on three Lumi. Then the bindings/plug deleted and then learned new ones. The data points are all there but not updating. No watt specification or temperature as well as consumption. I then tried various things that were discussed here but also no changes. Since it was running with the v2.17.1 and not now with the nezeren versions, it can only be due to deconz or, I'm actually a simple user who doesn't know about it? For example, if I restart the deconz adapter in ioBroker, the current data is fetched, but that's about it. No data arrives during operation, this may also have something to do with the API?
Kind regards
Fabio

nunoary, 2022-10-25:
@sansonifabio 
What do you mean "bindings/plug deleted????

After update the firmware of the plug's you need to delete the device on deCONZ GUI and re-pair again.

sansonifabio, 2022-10-25:
"**you need to delete the device on deCONZ GUI and re-pair again.**"

That's what I meant, I did that too, but it didn't work.

sansonifabio, 2022-10-26:
In principle I see it like @erik3rikI' ve tried and done and done for three days now and apart from the fact that I messed up my Zigbee network because the routers have been reset, nothing has changed. I sincerely hope that the developers will find the error and fix it, it worked before. Many Thanks.

Koal4, 2022-10-27:
I also tried the suggested tips and can't get it to work properly at all. I have all the options that kbase descibed which you should see. When pairing it in Deconz I can read out everything but when I reboot my Home assistant device I loose some. 
And when it works in HA i have big gaps with no data. As if it is not receiving while the meter is a few cm away from the HA device with the Zigbee controller. Before this deconz update and other firmware this smart plug was doing his job just fine so don't understand this fix or update and then saying it was not working how it should. Seems to me that more ppl have issues then before.
I recently purchased another one and back then I noticed it was on a newer firmware (12-06-21). Same as this this version but this one doesn't have any issues at all in my setup.

Mask3007, 2022-10-29:
I can confirm, that it really might be an issue with deconz. As I mentioned before I also ordered a sonoff stick to test it parallel with other zigbee software and it arrived. Since after about >20 re-pairs I last got the socket working with 2x Power 2x temp and 1x Cons. So before the test with other integrations I jsut gave it another try but before I updated deCONZ to the latest in HA available version (6.17.0 with deCONZ 2.19.1). 
After this re-pair (with every step in between) I readded it to Home Assistant aaaaand ... 1x Power Sensor 1x temp ... NO cons. sensor.
So I installed zigbee2mqtt to test if the device itself is the problem. I set it up, paired it in zigbee2mqtt and I can confirm it is not the device itself. Now I also got back my voltage and current sensor so at total (Cons. , current, temp, link quality, power, update and voltage). I also can set wether the device is on or off by default after a power outage. Everything works so fine even the update interval ist much more often. 
So for me I came to the conclusion that there must be a bug in deCONZ. I can not say wether it is just dirty data (not everything removed after deleting the node) or it is the ddf (which in my case I never got to be displayed as "xiaomi..." when I edit ddf) or with the binding itself. 

Offtopic:
But the main thing that really bothers me is the minimal support. Even that there are many people complaining this issue.
Well I don't say deCONZ is a bad product, not even at all and hey it's free so who would I be to claim that something needs to be repaired. It works quite well and was, except when there was an update, quite stable und "just works". But every deCONZ Update was thrilling and in my case not so stable at all. In my eyes it is not such a good move to make breaking changes in existing behaviours in the software just beause 

> "Any plug now making trouble was never meant to be allowed to play with deconz in the first place and apparently the measures to block that were not as successful as they should have been."

I will try out deCONZ and other zigbee home assistant integrations in parallel and will see which is working for me best.

marc965, 2022-10-30:
Hi!
Just updatet to deCONZ Current version: 6.17.0 (Phoscon App 2.19.01 / 19.9.2022) Firmware: 26720700

All my Lumi.plug.mmeu01 are now showing power, consumption and current again.
All my Lumi.plug.mmeu01 are not updaded, last working version was 6.15.0. Its working again as it was.

```
15:11:20:165 ZCL attribute report 0x04CF8CDF3C8A37D6 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
15:11:21:601 0x04CF8CDF3C8A340E extract Xiaomi special attribute 0x00F7
15:11:21:601 	64 on/off 1
15:11:21:601 	03 Device temperature 25 °C
15:11:21:602 	98 power 125.519997 (126)
15:11:21:602 	95 consumption 97.154686 (97155)
15:11:21:602 	96 voltage 2330.000000 (233)
15:11:21:602 	97 current 538.712463 (539)
15:11:21:603 	05 RSSI dB (?) 4 (0x0004)
15:11:21:603 	9a unknown 0 (0x00)
15:11:21:603 	08 unknown 278 (0x0116)
15:11:21:603 	07 unknown 0 (0x0000000000000000)
15:11:21:604 	09 unknown 3329 (0x0D01)
15:11:21:604 	0b unknown 0 (0x00)
15:11:21:604 	9b Consumer connected (yes/no) 1
```

thanks for fixing this.
Any explantion for us what have been changed?

raoulx24, 2022-10-30:
> Hi! Just updatet to deCONZ Current version: 6.17.0 (Phoscon App 2.19.01 / 19.9.2022) Firmware: 26720700
>
> All my Lumi.plug.mmeu01 are now showing power, consumption and current again. All my [...]

The polling is done, as I also showed in a previous reply, but this does not mean that they're exposed in api. You should check in deconz api if they (the sensors) are exposed. It is a bit complicated, as you should search for the zigbee id in the json output of sensors; I think you can find it somewhere in help section of phoscongw, api something (sorry for not being more specific, but I switched from deconz and I cannot take a look).

I think your devices are not properly handled after pairing; based on previous replies, I doubt that you will be able to use them in the newer versions of deconz without fw update. Be warned that some people complained that after the fw update, they started to have other problems (some missing reportings); so do it (if so) fully aware.

Cheers

Viktor45, 2022-11-04:
> You should check in deconz api if they (the sensors) are exposed

got issue with this, have two Lumi.plug.mmeu01 plugs, but on every reboot, got only single(!) sensors ZHAPower/ZHAConsumption from one plug or another plug randomly exposed in deconz api - sensors 

6.17.0 (Phoscon App 2.19.01 / 19.9.2022) Firmware: 26720700

closed 2022-11-08 by manup