dresden-elektronik/deconz-rest-plugin #6343: EVA Han Powermeter

sevrinrobstad, 2022-09-22:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: EVA
- Model identifier: Meter REader
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

Another user posted a request for this device here
https://forum.phoscon.de/t/eva-metering-han-powermeeter/1941

Link to device:
https://shop.evasmart.no/produkt/smarthus/maleravleser

sevrinrobstad, 2022-10-14:
I hope someone can add this at some point..

Smanar, 2022-11-07:
Hello can you share the cluster list pls ? to see wich one cluster are used by the device ?

sevrinrobstad, 2022-11-08:
Hi, and thank you. Yes - this seems to works. I had to change the "power" sensor from attribute 0x0508 to 0x0304, then it turned up with usage in WATT as of right now.

Smanar, 2022-11-08:
>I had to change the "power" sensor from attribute 0x0508 to 0x0304, then it turned up with usage in WATT as of right now.

Hu ?
You have changed it just in the bind part ? and 0x0508 is the current, I have used 0x050B for the power. But It can't work as it, as I have set all the DDF by defaut, and legacy code don't use 0x0304.

To have it working need to correct it in the bind part and make

```
        {
          "name": "state/power",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0304",
            "cl": "0x0b04",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "read": {
            "at": "0x0304",
            "cl": "0x0b04",
            "ep": 2,
            "fn": "zcl"
          }
        }
```

sevrinrobstad, 2022-11-08:
Yea, I did the change in the code as you describe over.
The Phoscon/help/API Info/sensors trick was good , 
It looks like more info is coming through coming through to phoscon, so that make's it HA's fault.
But the most important one was power, so I'm quite happy :)

```
  {
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "xxxxxx",
    "lastannounced": null,
    "lastseen": "2022-11-08T19:20Z",
    "manufacturername": "Eva",
    "modelid": "Meter Reader",
    "name": "Power 58",
    "state": {
        "current": 4700,
        "lastupdated": "2022-11-08T19:20:54.598",
        "power": 1613,
        "voltage": 2370
    },
    "swversion": "2.0",
    "type": "ZHAPower",
    "uniqueid": "xxxxx"
}
```

Smanar, 2022-11-08:
>so that make's it HA's fault.

Can be just a synchronisation issue.
It's for that I m asking if the sensor is not working (real issue) or not present, probably a synchronisation problem (can happen if HA have included the device before the DDF was finished, so the device not completely in the API, and HA scan entries only 1 time)

Need some convertion too, like need voltage / 10, you know how to do that using DDF ? `Item.val = Attr.val / 10`

PR #7034 (2023-07-20): DDF for Eva Powermeter
Fix #6343 .
closed 2022-12-08 by github-actions[bot]