dresden-elektronik/deconz-rest-plugin #6344: Tuya Remote 4 switch

Remc000tje, 2022-09-22:
## Device

- Product name: Remote 4 Keys 12 Scenes.
- Manufacturer: _TZ3000_dziaict4
- Model identifier: TS0044
- Device type : 
  - Remote
  - Switch (4x)

optical it is exactly the same as: Remote 4 Keys 12 Scenes Zigbee ID: TS0044 | _TZ3000_ygvf9xzp

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2022-09-27:
Hello, can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_ygvf9xzp", "_TZ3000_ee8nrt2l", "_TZ3000_dziaict4"],
  "modelid": ["TS0044", "TS0044", "TS0044"],
  "product": "Tuya remote 4 gangs",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

Remc000tje, 2022-10-04:
It will proabably work, but I have no option to put enter the map where the DDF file is located or where one can edit. 

I am working on a synology VM , with HA Home Assistant 2022.9.4 , Supervisor 2022.09.1 , Operating System 8.2 , Frontend-versie: 20220907.2 - latest.

I tried to SSH into the NAS, but there I cannot find the Deconz location..

gr. Remco

Remc000tje, 2022-10-11:
it will work, saw it working on other HA. I am not able to edit in my VM, searched for help on many fora.
I can edit the DDF in Deconz itself, but cannot edit "Bindings".

gr. Remco

Smanar, 2022-10-12:
>it will work, saw it working on other HA

If you can confirm it's working I can make a PR.

>but cannot edit "Bindings".

You can do it with drag and drop https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#bindings-panel (there is a video too)

Remc000tje, 2022-10-15:
Confirmed working status! please make a PR.

[image]
[image]

Smanar, 2022-10-15:
Added here  https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6244

Remc000tje, 2022-11-06:
Thanks Smanar, so in the next update it will work fine.

kind regads, Remco

closed 2022-12-07 by github-actions[bot]