dresden-elektronik/deconz-rest-plugin #6349: Tropernic Intelligent Lock TY0A01

proggprod, 2022-09-24:
## Device

- Product name: Tropernic Intelligent Lock, zigbee version
- Manufacturer: _TYST12_bfejtdig
- Model identifier: TY0A01 
- Device type : Please remove all unrelated device types. 
  - Lock
  
- Product link with pictures https://www.aliexpress.com/item/1005004243881094.html

## Screenshots

[image]

[image]
### Basic
[image]

### Time
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### Tuya specific
[image]

### OTAU
[image]

### Time
[image]

proggprod, 2022-10-24:
Not solved

closed 2022-11-24 by github-actions[bot]

Mimiix, 2023-02-22:
@Smanar Can this be added?

Smanar, 2023-02-22:
I don't found on google information about DP used by this device "_TYST12_bfejtdig".
And it use the tuya cluster.

Tomasz45, 2023-10-07:
Ktoś rozwiązał problem ?

Smanar, 2023-10-07:
I haven't found someone that have find the dpid used by this device
I have found this information https://developer.tuya.com/en/docs/iot/zigbee-doorlock-dp?id=K9fembhbeab0p
But no warranty this devices use the same one, Tuya shit on rules/standard even when they are from them.