dresden-elektronik/deconz-rest-plugin #6352: BW-SHP15 _TZ3000_mraovvmm suddenly stopped measuring

SebLz, 2022-09-24:
## Describe the bug
After about 6 monthes of good services, one of my BW-SHP15 plug stopped measuring power and consumption. The consumption counter is not increasing anymore, power and voltage are always 0 even though the plug itself correctly delivers power to the load (I've been using it to charge an hybrid car (2000+ W under load)). I tried to unpair, then pair again with no effect. Could this be a hardware issue ?

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

[image]
[image]
[image]

## Environment

 - Host system: NUC
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: (26720700)
 - deCONZ version: (2.18.02)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? Yes bluetooth gongle

## deCONZ Logs

## Additional context

Mimiix, 2022-09-24:
Hi,

Do you have more of these which also have the same manufacturer name?

SebLz, 2022-09-24:
> 

Hello, yes I have 3 others that still work fine

Mimiix, 2022-09-25:
Hi,

In that case I am closing this issue as this is not a bug report as described in #5113. Feel free to open an topic on the forum.

Kind regards,
Dennis

closed 2022-09-25 by Mimiix