dresden-elektronik/deconz-rest-plugin #6360: Frient/Develco ZHEMI101 energy meter counting in MWh instead of kWh

engerberg, 2022-09-27:
## Describe the bug
When upgrading to Home Assistant addon deCONZ 6.16, the energy meter Develco/Frient ZHEMI101 started counting total consumption in MWh instead of kWh. 

## Steps to reproduce the behavior
- UsingFrient/Develco ZHEMI101 energy meter
- Running deCONZ 2.16 through Home Assistant OS
- Total consumption for energy meter is calculated in MWh

## Expected behavior
- Total consumption calculated in kWh

## Screenshots
Switched from 7640 kWh to 7.64 MWh

[image]

## Environment
 - Host system: PC 
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26580700)
 - deCONZ version: (2.18.2)
 - Device: ConBee II

## deCONZ Logs

## Additional context
Downgrading to HA addon 6.15 (deCONZ 2.17.1) got it back to normal again.

geltzer, 2022-09-27:
Same issue here

M203, 2022-09-27:
Me as well. Does it work to downgrade deconz? Will it mess up historic data?

engerberg, 2022-09-27:
> Me as well. Does it work to downgrade deconz? Will it mess up historic data?

Works to downgrade, but I now have a jump between ~7640 kWh to 7.64 and back to 7640 again over a day or two, which makes it complicated to plot the historic data

SwoopX, 2022-09-27:
For a quick solution, you can either edit the DDF (and remove the division by 1000) or delete the DDF. Both requires a restart.

kusk, 2022-09-28:
Seeing the same issue.

pfefferle, 2022-09-28:
same for me

Mimiix, 2022-09-28:
@SwoopX you adjusted the ddf in the codebase?

pfefferle, 2022-09-30:
I tried to change the `pulseconfiguration` using the API, but it seems to reset short afterwards.

kusk, 2022-09-30:
Will this be fixed in an upcoming version?

Smanar, 2022-10-01:
If someone can try the correction and confirm it's working, I will make the PR

jenseo, 2022-10-01:
@Smanar I can confirm it works after changing the value to `Attr.val` without the division in the deconz zigbee vnc application, not sure if this was the correct way to do it...?

jenseo, 2022-10-01:
Hmm, might have been to quick to draw that conclusion. Now, after the energy dashboard in HA has started updating the data, it says current energy usage is 11,777.457. So that seems way to high unfortunately..

Smanar, 2022-10-02:
So it haven't worked, or the convertion is not enought ?

jenseo, 2022-10-02:
This is how it looks when I look at the yearly data, so it seems the `/ 1000` might have been correct, but it didn't seem to report any data (or very low volumes). 

[image]

Now when I have reverted back to adding `/ 1000` again, it seems to be back to underdeliver the amount of kwh. So the question is if something has changed and that we need to do like `/ 100` or something?

jenseo, 2022-10-02:
@Smanar ☝️

Smanar, 2022-10-02:
Ha yes, If I remember HA make a calculation. The value returned is a counter, and HA display value by day, so it mean the code remember all values (at least the one at 0h or 24h)  and make calculation.
So to be sure you need to wait at least 1 complete day

jenseo, 2022-10-04:
@Smanar Makes sense! I have removed the `/ 1000` once again and will wait for it to update properly in the energy dashboard to see how it goes :)

Here's what it's reporting at the current moment anyways:

[image]

jenseo, 2022-10-04:
It looks off, but maybe that's due to the calculation you talked about, since it says almost 12,000 kWh. Otherwise I guess it's an issue with the `.` (dot) that's interpreted as a comma or something..?

pfefferle, 2022-10-04:
Either you have to not remove the `/ 1000` but lower it to `/ 10` or you can remove it, but lower the `pulseconfiguration`.

jenseo, 2022-10-04:
@pfefferle I'm out on deep water here with changing stuff in the VNC deconz app, but when I changed it there, it reset like the next hour or so. I see that there's also a pulseconfiguration as you say. I think I will await a fix for this that's hopefully released soon, since I'm not sure exactly what I'm doing here 😄

popy2k14, 2022-10-04:
Exact same issue here after upgrading to Home Assistant addon deCONZ 6.16 (= deconz 2.18.2).
Is there an working workaround?

@jenseo does removing the division and wait a whole day work?

popy2k14, 2022-10-04:
In my case, removing the division by 1000 and hot reload the DDF worked!
Before it showed 0,434 kWh and now 434,71 kWh counting up after the commas.

See here:
[image]

I will let it run for an day and see if the readings are plausible.

jenseo, 2022-10-04:
@popy2k14 for me the removal of division reset itself after an hour or so, so haven't been able to verify it unfortunately. Hope this can get fixed soon, the data is all messed up now, I guess it needs to be reset once it starts working again..

kusk, 2022-10-04:
Removing / 1000 work fine for me.

jenseo, 2022-10-04:
> Removing / 1000 work fine for me.

@kusk Where do you remove it without it resetting again after some time?

popy2k14, 2022-10-04:
@jenseo connect with VNC to deconz, select the device, select Edit -> Edit DDF in the menu.
Under Items -> consumption -> Expression -> remove the " / 1000".
Select File -> Save & File -> Hot Reload.

i have now correct my values in the recorder database with sqlite web addon and multiple statements.

Hope it sticks.

popy2k14, 2022-10-04:
@SwoopX please fix with upcomming release. As confirmed now multiple times, removing `/ 1000` works.

BacktrackBtk, 2022-10-29:
Been following this thread a bit since last update messed up the values.
6.16 update gives me off the chart values. Removing  / 1000 is not the solution for everyone.
[image]

popy2k14, 2022-10-29:
In 2.19.1 the /1000 is fixed.
Since HA was recording also during the switch to /1000 and with now removed /1000 it's messed up.
You have to delete your HA recorder database, like here described: https://community.home-assistant.io/t/how-do-i-delete-the-database-on-home-assistant/222962/3?u=popy

But be aware, every recorded backlog is deleted!!!!!!!!!!!!
In my case it's not a problem, but i dont know your setup.

Configuration is not affected because it's all stored in yaml files.

BacktrackBtk, 2022-10-29:
Thanks for the answer!
I meant 2.19.1. I looked at the wrong version.
I will try what you suggest

popy2k14, 2022-10-29:
You're welcome.

closed 2022-11-28 by github-actions[bot]

jenseo, 2022-11-28:
Hello, I had this issue and hoped that the upgrade + deleting the database would solve it, but unfortunately not. It still counts in Watts instead of KWh it seems. Thinking it may be some cache attached to the device, either in home assistant or Deconz? Even when I completely wipe the database, two hours later it's still reporting incorrect numbers... so frustrating.

Does anyone know how to clear everything related to the energy dashboard, including the devices, sensors and what not..?

popy2k14, 2022-11-28:
Hmm, all backlogs are in the database, so when you delete it, all old data is gone.
In my case the sensor works.

Maybe you got the wqrong one and there is one for other pulsed/kwh?

Qalculus, 2022-11-28:
OPs and others problem in this thread was that the value got converted to MWh, i.e a factor 1000 less than the expected kWh. If you are seeing it as Wh it means you have the opposite problem, that the value is a factor 1000 larger, now that it has been fixed. What does your sensor say the expected kWh per pulse is and what is the kWh per pulse for your electrical box?
It should be fixable by dividing the value by 1000 in the device configuration in the Deconz Interface, but I never figured out how to get the change to persist across restarts of the Deconz addon in Home Assistant.

jenseo, 2022-11-29:
> OPs and others problem in this thread was that the value got converted to MWh, i.e a factor 1000 less than the expected kWh. If you are seeing it as Wh it means you have the opposite problem, that [...]

Sorry, that was just me being bad at energy values, it is de facto 1000 less than expected, so Mwh then I guess

jenseo, 2022-11-29:
> Hmm, all backlogs are in the database, so when you delete it, all old data is gone.
> 
> In my case the sensor works.
> 
> 
> 
> Maybe you got the wqrong one and there is one for other pulsed/kwh?

The weird part is that all data is removed from the energy dashboard, but after two hours when new data is added, it reports 1000 less still.

I did fiddle with it in deconz when the issue first arised (removing the /1000 value) but everything there is restored as well, but it might be that I have missed some step there perhaps. I wish I could do a "factory reset" in the deconz gui to make sure

jenseo, 2022-11-29:
Alright, I was able to solve this finally. I had to call `set state` under developer tools, then it finally updated to the correct values 🙏 So for anyone struggling with the same issue, you can always try that

popy2k14, 2022-11-29:
nice that you solved the issue and thx for posting it.

Idaho947, 2023-04-05:
Hello, is it normal that I've no batterie return for this device ?

Smanar, 2023-04-05:
Lol same than for other, from that I m reading you have a lowbattery but not a battery level (but it seem I was from for the previous one)

Smanar, 2023-04-05:
Do you have access to the GUI ?
To check value ?
The DDF is looking on Endpoint 0x02, Cluster 0x0702, attribute 0x0200.

Idaho947, 2023-04-05:
[image]

Arf unknowed cluster

Smanar, 2023-04-06:
No, it's fine, can you check  on Endpoint 0x02, Cluster 0x0702, attribute 0x0200 if the value is the battery value ?
Reading it in deconz can force an update in the API too.

Smanar, 2023-04-06:
>How can I check the endpoint if it is unknow ?

There are visible on your capture, unknow cluster are on the endpoint 01, and usefull one on the 02, it's the big number on gray part.

Idaho947, 2023-04-06:
So in the log this come every 5s :
[image]

Smanar, 2023-04-07:
So for me :
- attribute report are normal report, it's from the device itsel, not aggresive, but yes every 5s is too mich for me for a battery device
- I don't remember where the original code is from, but it seem this device don't use this attribute, at least your device.

Can try as bind in the DDF

```
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 20,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0400",
          "dt": "0x2A",
          "min": 20,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
```

But for the lowbaterry return, I have checked again, Z2M are using the attribute 0x0200 too.
IDK why it's disabled for your device.

Idaho947, 2023-04-07:
Maybe a old or a new firmware. 
I make a pr for the manufacturer name, for me it's juste Develco so maybe an old version.

Smanar, 2023-04-08:
A yes, the actual DDF only use

```
  "manufacturername": ["Develco Products A/S", "frient A/S"],
  "modelid": ["ZHEMI101", "ZHEMI101"],
```

Can try with adding your manufacture name visible in the GUI (remember you need to add manufacturername AND modelid, even they are already in the list)

Idaho947, 2023-04-10:
Ok I try your ddf.
`{
  "schema": "devcap1.schema.json",
  "manufacturername": ["Develco Products A/S", "frient A/S", "Develco"],
  "modelid": ["ZHEMI101", "ZHEMI101", "ZHEMI101"],
  "vendor": "Develco products",
  "product": "External meter interface ZHEMI101",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0007",
        "endpoint": "0x02",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x8000",
            "cl": "0x0000",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          },
          "parse": {
            "at": "0x8000",
            "cl": "0x0000",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015",
            "script": "develco_firmware.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/interfacemode",
          "parse": {
            "at": "0x0302",
            "cl": "0x0702",
            "ep": 2,
            "eval": "let v = [0x0000, 0x0001, 0x0002, 0x0100, 0x0101, 0x0102, 0x0103, 0x0104].indexOf(Attr.val); Item.val = v != -1 ? v + 1 : 0;",
            "fn": "zcl",
            "mf": "0x1015"
          },
          "read": {
            "at": "0x0302",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          },
          "write": {
            "at": "0x0302",
            "cl": "0x0702",
            "dt": "0x31",
            "ep": 2,
            "eval": "[0x0000, 0x0001, 0x0002, 0x0100, 0x0101, 0x0102, 0x0103, 0x0104][Item.val - 1];",
            "fn": "zcl",
            "mf": "0x1015"
          },
          "values": [
            [1, "Pulse counting electricity - Unit kWh"],
            [2, "Pulse counting gas - Unit m3"],
            [3, "Pulse counting water - Unit m3"],
            [4, "Kamstrup KMP protocol"],
            [5, "Linky protocol"],
            [6, "DLMS-COSEM - IEC62056-21 mod A"],
            [7, "P1 dutch standard â€“ DSMR 2.3 version"],
            [8, "P1 dutch standard â€“ DSMR 4.0 version"]
          ]
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pulseconfiguration",
          "parse": {
            "at": "0x0300",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "mf": "0x1015"
          },
          "read": {
            "at": "0x0300",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          },
          "write": {
            "at": "0x0300",
            "cl": "0x0702",
            "dt": "0x21",
            "ep": 2,
            "eval": "Item.val;",
            "fn": "zcl",
            "mf": "0x1015"
          }
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "refresh.interval": 3660,
          "parse": {
            "at": "0x0200",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = ((Attr.val >> 1) & 0x01) ? true : false",
            "fn": "zcl",
            "mf": "0x1015"
          },
          "read": {
            "at": "0x0200",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          }
        },
        {
          "name": "state/power",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "read": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl"
          }
        }
      ]
    }
  ],
 "bindings": [
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 20,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0400",
          "dt": "0x2A",
          "min": 20,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}`

Smanar, 2023-04-10:
```
	"manufacturername": ["Develco Products A/S", "frient A/S", "Develco"],
	"modelid": ["ZHEMI101", "ZHEMI101", "ZHEMI101"],
```

I think this is enought, have you tried to delete and re-include the deive, to  test it managed by DDF core instead of the legacy ?

Idaho947, 2023-04-10:
Yes it's woks. But what your modification for the binding do?

Smanar, 2023-04-10:
>Yes it's woks. But what your modification for the binding do?

Oups, haven't see.
Your device don't use the original DDF because your device was not inside. Just adding your model is enought to make it using the DDF core.

The DDF modification increase the reporting time and disable the attribute 0x0200, but you can try without the modification on the binding part, in my mind your device was supported by the DDF, and I was wrong.

Idaho947, 2023-04-10:
I've made a PR last week to include mine in the official ddf.

I'll see in one mounth if your modification change the batterie drain. thanks for your help

Idaho947, 2023-04-30:
Hello non change, the (new) battery have to be changed again...

Smanar, 2023-04-30:
It's the DDF you are using ?
Your device is "Develco" / " ZHEMI101" ?

{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "Develco Products A/S",
    "frient A/S",
    "Develco"
  ],
  "modelid": [
    "ZHEMI101",
    "ZHEMI101",
    "ZHEMI101"
  ],
  "vendor": "Develco products",
  "product": "External meter interface ZHEMI101",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x02",
        "0x0702"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0007",
        "endpoint": "0x02",
        "in": [
          "0x0000",
          "0x0702"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x8000",
            "cl": "0x0000",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          },
          "parse": {
            "at": "0x8000",
            "cl": "0x0000",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015",
            "script": "develco_firmware.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/interfacemode",
          "parse": {
            "at": "0x0302",
            "cl": "0x0702",
            "ep": 2,
            "eval": "let v = [0x0000, 0x0001, 0x0002, 0x0100, 0x0101, 0x0102, 0x0103, 0x0104].indexOf(Attr.val); Item.val = v != -1 ? v + 1 : 0;",
            "fn": "zcl",
            "mf": "0x1015"
          },
          "read": {
            "at": "0x0302",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          },
          "write": {
            "at": "0x0302",
            "cl": "0x0702",
            "dt": "0x31",
            "ep": 2,
            "eval": "[0x0000, 0x0001, 0x0002, 0x0100, 0x0101, 0x0102, 0x0103, 0x0104][Item.val - 1];",
            "fn": "zcl",
            "mf": "0x1015"
          },
          "values": [
            [
              1,
              "Pulse counting electricity - Unit kWh"
            ],
            [
              2,
              "Pulse counting gas - Unit m3"
            ],
            [
              3,
              "Pulse counting water - Unit m3"
            ],
            [
              4,
              "Kamstrup KMP protocol"
            ],
            [
              5,
              "Linky protocol"
            ],
            [
              6,
              "DLMS-COSEM - IEC62056-21 mod A"
            ],
            [
              7,
              "P1 dutch standard â€“ DSMR 2.3 version"
            ],
            [
              8,
              "P1 dutch standard â€“ DSMR 4.0 version"
            ]
          ]
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pulseconfiguration",
          "parse": {
            "at": "0x0300",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "mf": "0x1015"
          },
          "read": {
            "at": "0x0300",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          },
          "write": {
            "at": "0x0300",
            "cl": "0x0702",
            "dt": "0x21",
            "ep": 2,
            "eval": "Item.val;",
            "fn": "zcl",
            "mf": "0x1015"
          }
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "refresh.interval": 3660,
          "parse": {
            "at": "0x0200",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = ((Attr.val >> 1) & 0x01) ? true : false",
            "fn": "zcl",
            "mf": "0x1015"
          },
          "read": {
            "at": "0x0200",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl",
            "mf": "0x1015"
          }
        },
        {
          "name": "state/power",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 2,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          },
          "read": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 2,
            "fn": "zcl"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 20,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0400",
          "dt": "0x2A",
          "min": 20,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

Idaho947, 2023-04-30:
Yes it's this one.
I don't know why it woks normaly last year (06/2022 - 12/2022)

Smanar, 2023-04-30:
Was before the DDF engine ? If yes can try with disabling this DDF.
For me reporting on the DDF are longuer than in legacy code, can I don't think it's from them.

The legacy code can handle this device as sleeper, can try to set the "sleeper" in the DDF to true.

Idaho947, 2023-04-30:
Ok I test that. Anseer in one month.