dresden-elektronik/deconz-rest-plugin #6371: Device nameNamron ZigBee RGB Remote 4512706

olemr, 2022-09-30:
## Device
[image]

- Product name: Namron ZigBee RGB Remote 4512706
https://www.elektroimportoren.no/namron-zigbee-rgb-fjernkontroll/4512706/Product.html
- Manufacturer: 
- Model identifier: 
[image]
- Device type : 
  - Remote

## Screenshots

  - Endpoints and clusters of the node 
[image]

  - Node Info panel
[image]

### Basic
[image]
Same for all 4 channels:

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics
[image]

### Other clusters that are not mentioned above
[image]
[image]

Mimiix, 2022-11-01:
Reoepend on request by the user.

steinis123, 2022-12-30:
Status?

Mimiix, 2023-05-31:
Reopened on request.

@Smanar or @BabaIsYou , are you able to help out?

Smanar, 2023-05-31:
Aouch, lol, yeah, not a basic device ^^
I hope all bind/report are not used ....

olemr, 2023-06-02:
Thanks for chiming in ....
Away for the weekend.
Will report back later next week with logs.

closed 2023-07-01 by github-actions[bot]